espressif/arduino-esp32 #2: pow function needs #include math.h

bbx10, 2016-10-06:
Uncommenting the #include allows the program to compile. But the #include is not needed when building for Arduino Uno or Espressif ESP8266.

```
//#include <math.h>

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("pow(2.0, 5.0)=");
  Serial.println(pow(2.0, 5.0));
}

void loop() {
}

```

Compile error
```
/home/me/arduino-esp32/portable/sketchbook/powtest/powtest.ino: In function 'void setup()':
powtest:7: error: 'pow' was not declared in this scope
   Serial.println(pow(2.0, 5.0));
                              ^
exit status 1
'pow' was not declared in this scope

```

me-no-dev, 2016-10-07:
wait... what are you reporting? :D are you saying I need to include math by default so behavior is consistent with the rest?

igrr, 2016-10-07:
Yep, math.h should be included from Arduino.h I think.

bbx10, 2016-10-07:
I confirmed ESP8266 and AVR Arduino.h include math.h so that is the solution.

bbx10, 2016-10-08:
I confirmed https://github.com/espressif/arduino-esp32/commit/7530db5f06bd4efaa2d66fee96560d31b325e937 fixes problem.

closed 2016-10-11 by me-no-dev