espressif/arduino-esp32 #4: analogWrite() not implemented

vshymanskyy, 2016-10-06:


me-no-dev, 2016-10-07:
will be when the driver is ready

me-no-dev, 2016-11-13:
two types of analogWrite are now implemented through SigmaDelta and LEDC.
Headers for the relevant drivers are [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-sd.h) and [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.h)

jandolina, 2017-06-05:
Hello,
Has there been any progress on adding alalogwrite to the Arduino IDE for the ESP32?

jonnor, 2017-07-24:
Will you provide an Arduino-compatible `analogWrite` based on one of these functions? Really convenient for compatibility with existing code examples

Mendes11, 2017-08-25:
Is there any news about this? It's labed as 'to be implemented' but what is the priority of this? It would really helpful to have this function so it can be compatible with some libs that use it.

vseven, 2017-09-10:
Just tried to move some code over from a ESP8266 to my shiny new ESP32 and had the same issue with analogWrite not being implemented.  Will this be fixed any time soon?

monteslu, 2017-12-05:
any update?  Just hit this problem as well

vseven, 2017-12-05:
There are work arounds using ledC.  I uploaded a example into the GitHub repo:

https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/ledcWrite_RGB/ledcWrite_RGB.ino

You attached a pin to ledc, set it up, then you can call commands with ledc to update the output.

Sod-Almighty, 2018-01-04:
Over two years, and no fix? Seriously?

stickbreaker, 2018-01-04:
@Sod-Almighty different hardware. a direct equivalent implementation is impossible. There are other methods to achieve the same functionality, but they are not exactly compatible. 

You have to adjust your code to use either SigmaDelta or LEDC.   Both work but they are not drop in.  You as the programmer have to decide how to handle the differences.

Chuck.

vseven, 2018-01-05:
As posted see my example.  If you are programming for both say both a ESP8266 and a ESP32 you can use some if then statements to determine which to use.  For example a RGB LED strip control program I have attaches the pin to use in the setup:
```

		#if defined(ARDUINO_ARCH_ESP32)
			ledcAttachPin(m_nPinR, m_nChannelR);
  			ledcSetup(m_nChannelR, 5000, 8);
		#else
			pinMode(m_nPinR, OUTPUT);
		#endif
```

Then when I need to write to the output:

```
		#if defined(ARDUINO_ARCH_ESP32)
			ledcWrite(m_nChannelR, subStringR);
		#else
			analogWrite(m_nPinR, subStringR);
		#endif

```

Although I will disagree with @stickbreaker  - nothing is impossible.  There has to be a way to add a wrapper for analogWrite that just does the LedC stuff for you but it doesn't look like that's going to happen.

stickbreaker, 2018-01-05:
@vseven so,
  If I use your defines, I can get a 490hz pwm signal on 4 different pins, each one with a different 8bit duration? (Arduino Uno 3,  9, 10, and 11) and a 980hz PWM signal on 2 different pins (Arduino Uno 5,6) Also with different duration's?

I do not see how the following Arduino Uno commands `analogWrite(3,25);` `analogWrite(5,26);` could be directly translated.  
  
To directly translate from `analogWrite()` to `ledcWrite()` every interaction of of the Arduino timers would have to be emulated:
```c++
TCCR2B = (TCCR2B & 0b11111000) | 6;
```
I have only used analog Write for one motor controller; a brushed 8 Amp gear motor.  I had to adjust the PWM freq way down before the motor could operate.  So I reconfigured the base PWM to 122hz with this TCCR2B value.  This allowed me to  use `analogWrite(3,x)` to successfully operate this motor. But, I cannot expect this same command to set the ESP32's pin 3 PWM freq to 122hz.

I believe that making a partial translation would be more of a hindrance than benefit.  

I know that the ESP32  is different, so to successfully set a PWM freq to 122hz I must Learn how to configure the ESP32 to achieve this goal.  
 
Chuck.

vseven, 2018-01-05:
Although I made the ledC example sketch for the ESP32 it was more of a learning thing then anything.  I've never programmed a Uno controller, I've never actually used the analogWrite command for that matter.  I've only used the ESP32 and then I modified my code to also support ESP8266 chips (using analogWrite) so I don't know the detailed ins and outs.  I just know that for a ESP32 you have to add in the ledC PWM "channel" and tie it to your output.  You can then modify the channel for your frequency and resolution.

Like I said I'm just using it for relatively simple analog RGB LED control and my code works for both the ESP32 and 8266 without having to modify anything.   Check out the sketch above.  It might help (or not).

stickbreaker, 2018-01-05:
@vseven Like you have experienced, the ESP32 hardware is not equivalent to the AtMega328p.  So to effectively use it you have to Learn the differences and exploit them. Your example is a good reference of how to achieve a similar result.  My harping that the programmer must discover/learn the ESP32 hardware to maximize his effectiveness is my hardware background showing through. 

Thank you for your assistance.  I have not needed `analogWrite()` yet on the ESP32 so I am ignorant on the work-a-rounds.

Chuck.

capedra, 2018-07-20:
Wow, it looks like this is the oldest opened issue on espressif/arduino-esp32.

hznupeter, 2018-08-16:
it's a very important function ,is there anyone can rerewrite the lib?

ERROPiX, 2018-10-28:
Hi @me-no-dev 

I have create a small library that provide the classic analogWrite for ESP32, please test it and give me your feedback

Repository https://github.com/ERROPiX/ESP32_AnalogWrite

ctnalves, 2019-01-29:
> Hi @me-no-dev
> 
> I have create a small library that provide the classic analogWrite for ESP32, please test it and give me your feedback
> 
> Repository https://github.com/ERROPiX/ESP32_AnalogWrite

It's working perfectly! Thank you very much!

Sod-Almighty, 2019-01-30:
> I have create a small library that provide the classic analogWrite for ESP32

Excellent. I will have to give it a try.

alexceltare2, 2019-04-30:
This has gone far too long. Thanks @ERROPiX for providing a good wrapper.

me-no-dev, 2019-04-30:
this will not go into master. It is not the only PWM (or analogWrite) on the ESP32. There are far more reasons, but I think no point going into them. If anything goes in the master that is analogWrite, that would be MWPWM which does not have Arduino hal yet.

skickar, 2019-05-21:
It seems like we have time to hear the reasons, because I'm here to say:

Please implement analogWrite so I can write cross-platform code for these microcontrollers.

(updated, 2019 request)

jandolina, 2019-05-21:
FastLed is awesome for everything. Analog write is nice for anyone
transitioning from the arduino world to esp32. Analog write would make a
lot more examples carry over. But for all it’s worth. There are plenty of
libraries that take car of PWM.

atanisoft, 2019-05-21:
@jandolina FastLED uses the RMT to send commands out on the esp32 and not analogWrite to create a PWM signal.

jandolina, 2019-05-21:
Damn, my bad

exocode, 2019-05-31:
There are a few implementation proposals within the list of issues.

Why is that not implemented for 3 years? Does nobody use AnalogWrite? As @skickar pointed out: we need that for cross-platform development.  :-/

stickbreaker, 2019-05-31:
@exocode It seems that you have an interest in it, why don't you implement one of the proposals that will be compatible with `analogWrite()`.  

Chuck.

exocode, 2019-05-31:
Hi @stickbreaker, I tried that, but without success. I wanna reuse the same codebase for different ESPs (ESP32 and ESP8266). It seems, that in my build process the `ifdef` statement is not working properly: You can see it here: (with example code) https://github.com/platformio/platformio-core/issues/2597

exocode, 2019-06-01:
Ok. It seems that the platform I use (platformio) had a problem within it's latest stable release. I installed the development version and itn (https://github.com/espressif/arduino-esp32/issues/4#issuecomment-433720705) worked.

The error in my case relied on #ifef which was not compiled properly

mrayy, 2019-06-22:
This is my solution for compatibility issue with this function. I hope it becomes a builtin function
////////////////////////////////////////////////////////////////////////////////////

int _adc = 0;
int ADC_MAP[50];

#define PWM_FREQ 32000
//change this to set PWM output resolution. 8--> 0~255 values
#define PWM_RESOLUTION 8 
//#define USE_SIGMADELTA

void setupPWM(int pin)
{
  pinMode(pin, OUTPUT);
#ifdef USE_SIGMADELTA
  sigmaDeltaSetup(_adc, PWM_FREQ);
  sigmaDeltaAttachPin(pin, _adc);
#else

  ledcSetup(_adc, PWM_FREQ, PWM_RESOLUTION);
  ledcAttachPin(pin, _adc);
#endif
  ADC_MAP[pin] = _adc;

  _adc = _adc + 1;
}

void analogWrite(int pin, uint8_t val)
{
#ifndef USE_SIGMADELTA
  ledcWrite(ADC_MAP[pin], val);
#else
  sigmaDeltaWrite(ADC_MAP[pin], val);
#endif
}

// the setup routine runs once when you press reset:
void setup() {
  // declare pin 9 to be an output:
  setupPWM(9); //Similar to pinMode (9,OUTPUT);
}

void loop()
{
   analogWrite(9,128);
}

ERROPiX, 2019-06-22:
@mrayy I think my solution is better, check this repository https://github.com/ERROPiX/ESP32_AnalogWrite

mrayy, 2019-07-01:
Unless this became internally supported by ESP32, it will be very difficult to manage the channels when using third party libraries. 
For example, I recently used ESP32_AnalogWrite library and ESP32_Servo library, which each library was using its own internal ledc channels management, resulting in a conflict that the same ledc channel being allocated by both libraries.

sdaitzman, 2019-09-04:
Hello! Just expressing continued interest in adding analogWrite.

extesy, 2019-09-04:
I had a same problem recently: I needed to control a servo and two motors and since the libraries were different (there's no single library to control both) they had their own conflicting implementations of analog write.

Harvie, 2019-10-18:
Can you please add this? It should be really easy to do.
It's a major inconvenience for lots of people.

skickar, 2019-10-19:
Whoever downvoted Harvie's comment should just go implement AnalogWrite.

alexceltare2, 2019-10-20:
> Whoever downvoted Harvie's comment should just go implement AnalogWrite.

@ERROPiX already did a brilliant job.

dsyleixa, 2019-11-21:
analogWrite() is actually a crucial Arduino command (like e.g.  digitalWrite) and thus should be available for ESP by default!

dsyleixa, 2019-11-22:
analogWrite() is actually a crucial Arduino command (like e.g.  digitalWrite) and thus should be available for ESP by default!

alternatively and to delimit  from "analogWrite DAC values"  it might be helpful to implement both a function
SoftPwmWrite() fpr pwm pins
additionally to
analogWrite() for DAC pins.

Harvie, 2019-12-01:
> @ERROPiX already did a brilliant job.

So what is the problem with merging it upstream?

WayneKeenan, 2019-12-09:
I ran into the same problem as @extesy and @mrayy did.
This is such a core Arduino feature I was more than surprised to find it not there when I first went to use it.

alexceltare2, 2019-12-09:
> So what is the problem with merging it upstream?
According to @me-no-dev ,  only MWPWM can be used for `analogWrite`, however I am about to object that and say: Implement it with whatever wrapper possible and worry about perfecting it later.

alexceltare2, 2019-12-09:
> > @ERROPiX already did a brilliant job.
> 
> So what is the problem with merging it upstream?

So what is the problem with merging it upstream?
According to @me-no-dev , only MWPWM can be used for analogWrite, however I am about to object that and say: Implement it with whatever wrapper possible and worry about perfecting it later.

alexceltare2, 2019-12-09:
> > @ERROPiX already did a brilliant job.
> 
> So what is the problem with merging it upstream?

According to @me-no-dev , only MWPWM can be used for analogWrite, however I am about to object that and say: Implement it with whatever wrapper possible and worry about perfecting it later.

Harvie, 2020-01-05:
@igrr @kedars @mahavirj @me-no-dev @projectgus @Spritetm 
Can you PLEASE push this bit harder? :-)

wouterds, 2020-01-12:
Oh wow 😲, didn't realise this was missing! I might look into another some other bluetooth enabled boards then.

jandolina, 2020-01-13:
The ESP32 is an amazing platform. I have been a big fan of this hardware for years. The analogWirte no native is not a dealbreaker. There are many PWM libraries to help you get you where you need to be. Yes it would be cool to have this function, but the platform offers so much that it is a non issue.

Start here:
https://randomnerdtutorials.com/esp32-pwm-arduino-ide/

ERROPiX, 2020-01-13:
@wouterds @jandolina I have created a good wrapper library to work with analogWrite
https://github.com/ERROPiX/ESP32_AnalogWrite you can install it on both Arduino IDE and PlatformIO

skickar, 2020-01-14:
If you love the esp32 so much, why don't you go write an analog write love
poem to it so everyone else can enjoy it too.

WayneKeenan, 2020-01-14:
Power is red
Radio is blue
I can’t dim the LED
Analogwrite, I miss you

wouterds, 2020-01-14:
> @wouterds @jandolina I have created a good wrapper library to work with analogWrite
> https://github.com/ERROPiX/ESP32_AnalogWrite you can install it on both Arduino IDE and PlatformIO

Thanks, that does seem to work! For a bit. I'm not sure but it looks like there's a memory leak? I'm writing to 4 pins an analog value every loop and I see free memory draining rapidly until it crashes and reboots again.

dsyleixa, 2020-01-14:
as stated, analogWrite() is crucial in the Arduino API world for cross-platform-compatibility, so please provide it for ESP32, too, ASAP.

sdaitzman, 2020-01-14:
Is there something unusually difficult/on hold about implementing this? Not being facetious, just trying to understand the issue.

skickar, 2020-01-14:
thank you for this, I've been receiving emails from this thread for the
last several months and this is the only one that's made me any less angry.
Also, if it wasn't clear, my vote is for implementing analog write.

wouterds, 2020-01-14:
I briefly looked into this to see if I could copy it from @ERROPiX his repository but I have too little C/C++ knowledge (and this repository looks quite complex). But there is an open PR (https://github.com/espressif/arduino-esp32/pull/3110. Would be cool if @magicblocks & @me-no-dev could finish this up or if someone else could pick it up.

me-no-dev, 2020-01-20:
it needs to work in the following way:
1. find available PWM peripheral (LEDC, MCPWM, SigmaDelta, RMT, Soft?)
2. Make sure that the clock requirements will not interfere with other used ports on the same peripheral (LEDC and others share some clocks)
3. Mark the selected port as used and save it's usage type
4. Do not forget to free the port if a call to some other pin function is made. Now this is the big issue here... means that any time you call pinMode and others, have to check if that pin is attached to port and so on.
5. All of that functionality must be tracked by each peripheral as well (so you can use LEDC or other and still use analogWrite)

The lib above does not take any of those considerations. It's just not that simple :) But I already have some code written towards that goal.

gauix4, 2020-02-08:
arduino code:

Serial.print(_ssid);
Serial.print(_pass);

tharangachaminda, 2020-04-04:
for Arduino IDE there is a library and it worked for me perfectly for motor controller project.
[https://www.arduinolibraries.info/libraries/esp32-analog-write](https://www.arduinolibraries.info/libraries/esp32-analog-write)

dsyleixa, 2020-04-05:
perhaps someone of the devs  could test it and make a fork to integrate it to the common ESP32 standard Arduino core libs?

bperrybap, 2020-08-01:
Ok, so I just ran into this issue today.
Not having an analogWrite() function breaks my LCD library. (hd44780)
The hd44780 library currently works on EVERY other Arduino platform/core except this one.
This includes the esp8266 library.
While the hd44780 library code is smart enough to use digitalPinHasPWM() to detect if PWM support exists on the pin,
the code still doesn't work (won't even compile) on this platform because of a call to analogWrite() which does not exist.
Users should not have to muck up their code with platform/core specific conditionals to work around a missing Arduino core library function.

Given that analogWrite() is a core API function, it should always exist in this platform.
Until analogWrite() is properly implemented, which I believe is possible, 
my suggestion would be to add an analogWrite() function but not support PWM.
The Arduino.cc analogWrite() function already handles this for pins that do not support PWM.

For the esp32 analogWrite() with PWM support disabled, it would mean
treating all pins as if they do not support PWM, which technically would be correct since they don't have PWM capability using analogWrite() since the underlying code for PWM is not yet implemented.
To do this:
- change the digitalPinHasPWM() in all variants pins_arduino.h header file to return false
- create a minimal analogWrite() function that mimicks the Arduino analogWrite() behavior when PWM is not supported on a pin.
When PWM is not supported on a pin, the official Arduino core code will set the pin low if the value is less than half way to max and high if the value is over half way to max. 

In each variant's pins_arduino.h header file change the digitalPinHasPWM() macro to something like this:

#define digitalPinHasPWM(p)         (0) // analogWrite() does not yet support PWM

And for the matching analogWrite() function:

```
void analogWrite(uint8_t pin, int val)
{
        if(val < 128)
                digitalWrite(pin, LOW);
        else
                digitalWrite(pin, HIGH);
}
```
Making these changes/updates will allow existing code that calls analogWrite() to function as it does on all the other platforms.
It also allows the code to use digitalPinHasPWM() to detect if PWM is supported on the specified pin.

IMO, this is a very small/simple change that resolves the main issue of analogWrite() not existing which causes any code that attempts to use analogWrite() to fail to compile.

At some point a proper analogWrite() that supports PWM when possible, should be added to the core

[93 more replies]

closed 2022-04-06 by VojtechBartoska