espressif/arduino-esp32 #10: RAM usage and the maximum RAM reported during compilation

krzychb, 2016-10-07:
Hello everybody,

I would like to clarify what I see when compiling WiFiMulti.ino example sketch using Arduino IDE 1.6.12:

1. When compiling [WiFiMulti.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiMulti/WiFiMulti.ino) for `ESP32 Dev Module` with latest git release of this repository, the following is shown:

  ```
Sketch uses 289,903 bytes (27%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 88,144 bytes (76%) of dynamic memory, leaving 26,544 bytes for local variables. Maximum is 114,688 bytes.
Low memory available, stability problems may occur.
  ```

2. Compilation of equivalent [WiFiMulti.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino) for `Generic ESP8266 Module` with 2.3.0 core provides:

  ```
Sketch uses 228,325 bytes (52%) of program storage space. Maximum is 434,160 bytes.
Global variables use 32,036 bytes (39%) of dynamic memory, leaving 49,884 bytes for local variables. Maximum is 81,920 bytes.
  ```

* Why RAM usage for ESP32, for seeming identical sketch, is almost 3x bigger than for ESP8266?
* Why only 114kB maximum RAM is reported for ESP32? I was expecting 520 kB.

Thank you for lunching this repository in parallel to esp-idf!
I am sure this requires a lot of effort to maintain extra branch for Arduino for fledgling ESP32.

Krzysztof

me-no-dev, 2016-10-07:
have not really gotten down to properly configure reported sizes ;)

me-no-dev, 2016-11-16:
this is mostly fixed ot done as much as it can be done currently. RAM should be properly calculated now

closed 2016-11-16 by me-no-dev

krzychb, 2016-11-16:
No more terrifying warnings "Low memory available, stability problems may occur." :wink: 
Thank you @me-no-dev!