espressif/arduino-esp32 #6: F macro compiler errors

bbx10, 2016-10-07:
This code compiles on Uno and ESP8266 but produces compiler errors on ESP32.

```
void setup() {
  Serial.begin(115200);
  Serial.println(F("hello world"));
}

void loop() {
}
```

```
In file included from /home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/Arduino.h:52:0,
                 from /tmp/buildad9bb1d844c0ed80cdda279063404da3.tmp/sketch/pgmstring.ino.cpp:1:
/home/me/arduino-esp32/portable/sketchbook/pgmstring/pgmstring.ino: In function 'void setup()':
/home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/WString.h:39:53: error: 'PSTR' was not declared in this scope
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                                     ^
/home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/WString.h:38:76: note: in definition of macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
/home/me/arduino-esp32/portable/sketchbook/pgmstring/pgmstring.ino:3:18: note: in expansion of macro 'F'
   Serial.println(F("hello world"));
                  ^
exit status 1
```

Including pgmspace.h (not necessary on Uno and ESP8266) produces the following.

```
"/home/me/arduino-esp32/hardware/expressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/config" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/bt" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/driver" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/esp32" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/freertos" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/log" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/newlib" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/nvs_flash" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/spi_flash" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/expat" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/json" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/mbedtls" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/nghttp" "-I/home/me/arduino-esp32/hardware/expressif/esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib -MMD -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -DF_CPU=160000000L -DARDUINO=10609 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-I/home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32" "-I/home/me/arduino-esp32/hardware/expressif/esp32/variants/esp32" "/tmp/buildad9bb1d844c0ed80cdda279063404da3.tmp/sketch/pgmstring.ino.cpp" -o "/tmp/buildad9bb1d844c0ed80cdda279063404da3.tmp/sketch/pgmstring.ino.cpp.o"
In file included from /home/me/arduino-esp32/portable/sketchbook/pgmstring/pgmstring.ino:1:0:
/home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/pgmspace.h:22:14: error: conflicting declaration 'typedef char __FlashStringHelper'
 typedef char __FlashStringHelper;
              ^
In file included from /home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/Arduino.h:52:0,
                 from /tmp/buildad9bb1d844c0ed80cdda279063404da3.tmp/sketch/pgmstring.ino.cpp:1:
/home/me/arduino-esp32/hardware/expressif/esp32/cores/esp32/WString.h:37:7: error: 'class __FlashStringHelper' has a previous declaration as 'class __FlashStringHelper'
 class __FlashStringHelper;
       ^
exit status 1
Error compiling for board ESP32 Dev Module.
```

me-no-dev, 2016-10-07:
not sure how to handle those yet... we will not store anything in flash the way we do on ESP8266.
Ivan is on vacation so we will look at it soon

bbx10, 2016-10-07:
Ok, no rush. I did something like this to get the code to compile without doing mass replacements.

```
#undef F
#define F(...) __VA_ARGS__
```

igrr, 2016-10-07:
Since we have DCACHE on the ESP32, we can define all pgmspace stuff to use direct memory access (similar to how it is done on Arduino Due).

bbx10, 2016-10-08:
I confirmed https://github.com/espressif/arduino-esp32/commit/2ba4c40acaa76f9cfa2e5c22069fbc17f1eeee48 fixes this problem.

closed 2016-10-11 by me-no-dev