espressif/arduino-esp32 #11: I2C not communicating with slave device

bbx10, 2016-10-08:
I tried an HTU21DF (temperature and humidity) and a BMP180 (barometric pressure) but the drivers do not get past the first I2C transmission. Both sensors work on an ESP8266.

The first thing the HTU21DF driver does is send a RESET command like this. Writing to this register forces the sensor to factory defaults.

```
  Wire.beginTransmission(HTU21DF_I2CADDR);  // addr 0x40
  Wire.write(HTU21DF_RESET);                // 0xFE
  Wire.endTransmission();
```
This is what it looks like on an Uno.

[image]

Here is what happens on my ESP32. If other people have I2C working, the problem must be in my setup or ESP32.

[image]

Since there is a NAK, the driver aborts the rest of the transmission.

I am trying various changes to esp32-hal-i2c.c but have not made any progress.

me-no-dev, 2016-10-08:
I have HTU21 here, no clue if it's DF though... will try the code above, but please share the rest of it so I can test both sensors here.

me-no-dev, 2016-10-08:
I'm looking deeper into it and can see other issues also
I have it working as is with TPM102 temp sensor and SSD1306 oled, but if I run scan... things go to hell because transmissions happen without delay in between

me-no-dev, 2016-10-08:
I fixed much of the code (not pushed yet) but what I see with my SHT21 is that it can not pull SDA enough to register the ACK

[image]

me-no-dev, 2016-10-08:
I've done what I can in my latest commit. From all of the devices that I tested, only that device and AM2320 do not respond. I tested IMUs, barometers, oled and 1602 screens, RTC and some more. All showed in scan.

comcat, 2016-10-09:
My testing result:

BMP085, BMP180, SSD1306 OLED can be found by scanning i2c
SHT20, TSL2561, PCF8563 can not be found

me-no-dev, 2016-10-10:
please try the latest commit! it should be all good now :)

sorscode, 2016-10-11:
I'm unable to find my PCF8563, but was able to find some old DS1307, even though it gave me a weird reading, and printed out it found a dozen devices when I was searching for the DS1307. I'm going to test other i2c devices this evening.

ladyada, 2016-11-22:
testing today: 
SSD1306, PCF8523, HT16K33, HDC100x, SHT31-D works
but others like PCA9685, DS3231, Si7021, BMP280 and TCS34725 are not seeming to work with drivers for classic arduino/esp8266

the sensors have 10K pullups on SDA/SCL

me-no-dev, 2016-11-23:
this get's more and more interesting :D 
@ladyada from the ones you listed as non-working I have only the DS and I'll give it a go, but is there any way that you can give some more info on what is failing?
Is it not getting ACK? Does the transmission fail in the middle? Can you look at a scope/logic analyser and tell what is going wrong?

ladyada, 2016-11-23:
yeah i will bring home a salae logic over the holiday and see if i can see what is up

me-no-dev, 2016-12-09:
try the latest commit please :)

me-no-dev, 2016-12-09:
@ladyada is there any way that you can prepare a pack of the most used i2c/spi devices/sensors that I can purchase from you? Having the most common devices to test could greatly help to improve libraries. I have PayPal that I can use to pay.

me-no-dev, 2016-12-15:
many many things have been fixed in I2C. I'm closing this issue now, if someone still has problems, feel free to reopen

closed 2016-12-15 by me-no-dev

ladyada, 2017-03-19:
oof apologies i didnt see your reply - i did just retest with a wide variety of i2c sensors and all are working now. once i have more than one ESP32 feather ill get you a kit!

me-no-dev, 2017-03-19:
Thanks :) That would be great!

sansillusion, 2017-08-28:
Can anyone give me some instruction on using a DS3231 with a nodeMCU-32S (pins and library to use) ?
I tried many things but nothing seems to work.