espressif/arduino-esp32 #12: 70% of example sketches fail to compile

krzychb, 2016-10-08:
Hi,

Current git release provides the following examples sketches:

[image]

Out of them the following compile out of box:

  1. WiFiClient
  2. WiFiClientBasic
  3. WiFiMilti

All other drop some errors. 
For example, compilation of HTTPSReqest.ino fails as follows:

```
C:\Users\Krzysztof\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\WiFi\examples\HTTPSRequest\HTTPSRequest.ino:15:30: fatal error: WiFiClientSecure.h: No such file or directory

 #include <WiFiClientSecure.h>

                              ^

compilation terminated.
```
Test was done using Arduino 1.6.12 portable.

If this is the case, then I would remove all the sketches that do not work. 
This is not to confuse people that are testing this repository.

Krzysztof

me-no-dev, 2016-10-08:
done! fixed a couple to work ;)

closed 2016-10-08 by me-no-dev

krzychb, 2016-10-08:
@me-no-dev,
Congratulations on first closure of issue report!
And what an amazing speed :+1: