espressif/arduino-esp32 #13: No "Serial" Module

samjulien, 2016-10-08:
Hi there. Windows 8.1 here. Followed the instructions on getting set up and receiving the following when trying to upload a blinking light sketch:

```
Traceback (most recent call last):

  File "C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/esptool.py", line 25, in <module>

    import serial

ImportError: No module named serial

exit status 1
Error compiling for board ESP32 Dev Module.
```

Any ideas?

MrAPierce, 2016-10-09:
Hi I had the same issue, you're missing pyserial. You can install pyserial via pip. Open CMD as Administrator and type: pip install pyserial

I believe that pip is installed by default when installing python, Pip should go and get pyserial and install it in the correct location. Restart Arduino IDE and Build, worked for me.

samjulien, 2016-10-09:
Ah, excellent, thanks. Now it compiles and gives me:

```
esptool.py v2.0-dev
Connecting...

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('f')
```

So that's something.

MrAPierce, 2016-10-09:
That error seems related to uploading, double check that you holding down boot button then press and  reset button and then release both, then click upload (arrow button).

Does it compile by clicking the tick (verify) button?

samjulien, 2016-10-09:
Yes, it compiles with the verify button, and yes, I'm doing the boot-and-reset press and upload. No dice. If I open the serial monitor at 115200 I do see some boot text and info about the wifi starting, so the device is alive. I've got flash frequency at 40MHz and flash speed at 115200, are those right? And does the programmer setting matter? I've got AVRISP mkII selected.

MrAPierce, 2016-10-09:
These are the setting that work for me:

[image]

samjulien, 2016-10-09:
Sorry for the delayed response - thanks for the screenshot. Still getting the same error for some reason. I'll keep tinkering.

MrAPierce, 2016-10-11:
Hi @samjulien, did you have any luck getting this to work?

sorscode, 2016-10-11:
Hi @samjulien, Did you by chance install the Silabs driver? I was able to get my Dev board working, once I go the driver install on Mac. I just finished testing PubSubClient on the ESP32, and it works good.

samjulien, 2016-10-11:
@MrAPierce - not yet, still not sure what the problem is. I'm just uploading super simple LED blink code with an LED in pin 12 and a resistor to ground.

@sorscode - are these same drivers by Silabs used for the ESP8266? Googling "silabs esp32" took me to the same page of "CP210x USB to UART Bridge VCP Drivers," which I do have installed from working with the 8266.

Once I get this working I'm happy to submit a PR to the readme with anything pertinent. Definitely not just trying to be a pain, hoping to help out the larger community by being a guinea pig. :)

sorscode, 2016-10-11:
@samjulien That would be the one. Which port do you have selected when uploading? I'm going to build a quick Windows 8.1 VM and test.

samjulien, 2016-10-11:
Okay, cool. A buddy of mine pointed out that it's been several months since I installed those drivers for the ESP8266 and perhaps an update to them is in order, so I'll also give that a go and report back.

samjulien, 2016-10-12:
Hmm, updated the drivers, took away the LED, did the boot mode dance, changed sketch to just empty `setup` and `void` loops - still no dice. Same invalid head of packet error. Settings are as you have it above. Weird.

AnalogLamb, 2016-10-14:
Hi You need to install python2.7 and pyserial,

Please see here: https://learn.adafruit.com/arduino-lesson-17-email-sending-movement-detector/installing-python-and-pyserial

tamberg, 2016-11-07:
+1 for AnalogLamb, see https://github.com/pyserial/pyserial, or just type `pip install pyserial`
then restart the Arduino IDE. Fixed it for me.

samjulien, 2016-11-07:
Yeah, above I had already installed pyserial and was able to compile but got a different error. That was on Windows 8.1. I just got a new laptop with Windows 10 on it and will give this another shot soon.

yashchitalia, 2016-11-13:
I too am seeing the serial error "ImportError: No module named serial" in spite of having both python and pyserial. When I opened up a python console on my Terminal and imported serial, that worked fine. Any suggestions?

me-no-dev, 2016-11-16:
Arduino compile runs the version of python that you have in the path :) so basically starting a shell without running .bash_profile and executing python. Where is your python installed? Why not use the default one (I use it daily and it's totally fine)?

yashchitalia, 2016-11-16:
No that was just a theory. Turns out, I am running the inbuilt version of python, but the Arduino compiler still returns an error importing "serial".  Importing serial in the terminal works fine. In fact, I printed the command that had failed in the compiler (using the verbose output functionality), and ran it in the terminal, and that executed fine.
So I'm not sure what's going on there.

me-no-dev, 2016-11-16:
your PATH variable is different if you run in terminal than what programs trying to execute command see. Usually that is limited to /usr/bin:/usr/sbin:usb/local/bin:/usr/local/sbin
run ```which python``` in terminal and see where python is. It might be different than python that might be in one of the locations by default

yashchitalia, 2016-11-16:
So `which python` returns `/usr/local/bin`, and that's why I think its running the inbuilt version of python.

me-no-dev, 2016-11-16:
ok here is an idea :)
prepend ```which "{compiler.python.cmd}" && ``` to [this](https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L75) command and run build. it should print the path it sees right before building the bin

me-no-dev, 2016-11-16:
```
recipe.objcopy.hex.pattern=which "{compiler.python.cmd}" && "{compiler.python.cmd}" "{runtime.platform.path}/tools/esptool.py" --chip esp32 elf2image --flash_mode "{build.flash_mode}" --flash_freq "{build.flash_freq}" -o "{build.path}/{build.project_name}.bin" "{build.path}/{build.project_name}.elf"
```

yashchitalia, 2016-11-16:
So that worked! Thanks a lot! Adding the text to platform.txt revealed that the path it was trying to use was at `/usr/bin/`. However, changing that path in platform.txt seems to throw a generic`exit status 1
Error compiling for board ESP32 Dev Module.` error. Any suggestions?

me-no-dev, 2016-11-16:
do not change the path :)
open terminal and do
```bash
sudo mv /usr/bin/python /usr/bin/python_other
sudo ln -s /usr/local/bin/python /usr/bin/python
```
that will link the correct python to /usr/bin and save the day :)

yashchitalia, 2016-11-16:
So both those commands are forbidden. But I get the general idea. I'll try some other things out (like creating an alias, etc.), and see if that works.

evanmj, 2016-11-29:
I had this same issue on windows with the sparkfun board... I installed pyserial, then got the next issue.  Then I went back to holding the reset button while it programmed, and it worked.  Thanks guys!

dgdosen, 2016-11-30:
Hi all;

I'm working on a mac with homebrew installed python.

Arduino seems to be ignoring my path - as I choose to use the homebrew python over the system python.  

Nevertheless, if I "brew unlink python" - leaving only the system python, I can:

- run the system python
- see that pyserial is not installed
- install it (using sudo)
- and compile the sketch.

I don't have my esp32 hooked up right now, but will test it later.

dgdosen, 2016-11-30:
FYI - the above solution works (for homebrewed python macs) I was able to upload sketches and get them working on my esp32 (ESP32 Thing from Sparkfun)

Now I want to find/use a toolchain to avoid the Arduino IDE.

Ideally, the IDE would recognize my path settings so one can avoid the work around.

me-no-dev, 2016-11-30:
@dgdosen if you want to use it without the IDE:
- Download and install [esp-idf](https://github.com/espressif/esp-idf)
- Create blank idf project (from one of the examples)
- in the project folder, create a folder called components
- git clone this repository inside components as name arduino (```git clone https://github.com/espressif/arduino-esp32.git arduino```)
- ```make menuconfig``` has some Arduino options (maybe enable arduino autostart)
- in your main.cpp, include Arduino.h and format it as any other sketch (if autostart is enabled)
- ```make flash monitor``` will build, upload and open serial monitor

blackketter, 2016-12-01:
I can recommend using platform.io to do ESP32 builds on the command line without an IDE.  See http://platformio.org/get-started/cli

me-no-dev, 2016-12-01:
@blackketter and how is ESP32 working on platformio?

blackketter, 2016-12-01:
Well, I haven't dug in deeply but a blink example compiled, downloaded and ran perfectly on my SparkFun ESP32 Thing.

anziclone, 2016-12-04:
So I figured out a workaround for this problem and got the code to work on a Mac. 

1) Install PySerial using "sudo -H pip install pyserial" in you terminal.

2) Then get the path of python that you are working on in the terminal using "which python" command. This will give you a path name like below:

`/Library/Frameworks/Python.framework/Versions/2.7/bin/python`

Copy this path.

3) Open up the platform.txt in the hardware/espressif/esp32 folder.
Change the value of `compiler.python.cmd=python` to `compiler.python.cmd=/Library/Frameworks/Python.framework/Versions/2.7/bin/python`.
Also change the value of `tools.esptool.cmd=python` to `tools.esptool.cmd=/Library/Frameworks/Python.framework/Versions/2.7/bin/python`. This worked just fine for me and code is compiling and uploaded to the esp32 successfully.

me-no-dev, 2017-02-08:
this is now all resolved because python is no longer required on Windows

closed 2017-02-08 by me-no-dev

harshithuchil, 2017-08-17:
here am working with esp32  and arduino  but when i connect my esp32  its not showing me the port can anyone solve this ????

tolo69, 2017-11-01:
@harshithuchil,  did you already  solve the issue?
I face similiar Problem with ESP32DevC under win10. I have installed the CP2102 VCP-Driver. But after connecting the board it is not appearing in device Manager as virtual COM-port.

T-vK, 2019-08-01:
On Fedora you follow these instructions: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/fedora.md

And in addition to that, you run `pip install pyserial --user`. (Not as root!)

psiphi75, 2019-09-29:
For anyone who comes here from a search engine.  On Ubuntu 19.04, my version of `pip` is for Python 3.7, you need to install `pip2`.

```
sudo apt install python-pip
```

Then install pyserial

```sh
pip2 install pyserial
```

JulianNicholls, 2020-01-31:
Thank you @psiphi75. That was exactly the solution on Ubuntu 19.10 as well.

lviss, 2020-07-06:
Thank you @psiphi75. That was exactly the solution on Debian 10 as well.

cyzen, 2020-07-10:
Thank you @psiphi75. That was right step to the solution on Ubuntu 20.04 as well.
After i installed pip2 for python2,7 
(https://linuxize.com/post/how-to-install-pip-on-ubuntu-20.04/)
i could finally do "pip2 install pyserial"
and ESP32 compiling works at Arduino-IDE 1.8.13 without an error

lbernstone, 2020-07-10:
python2 is past end of life.  Move forward to python3 instead of backwards.
https://github.com/espressif/esptool/issues/528#issuecomment-626400411

jrc, 2020-09-18:
On macOS 10.15, Python 2 doesn't include `pip`. Here's how to do a user install of serial, so as not to pollute `/usr` etc.

```
easy_install --user pip
python -m pip install --user pyserial

```

spragiot, 2020-10-09:
to solve this issue just follow these steps

1. install and update python
sudo apt-get install python3 python3-pip python3-setuptools

2. set default python version
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10

regards

vipinsinghal1234, 2021-02-09:
Issue: import serial module not found :
sudo apt install python-is-python3

for library clash :
sudo pip3 uninstall serial
sudo pip3 install pyserial

worked for me because in my case this error was due to python supported by IDE

zume2020, 2021-05-16:
on Linux systems just try this ```sudo apt install python-is -python3```

fire17, 2021-08-21:
# sudo apt install python-serial python3-serial 

solved it! :D

new-php, 2024-12-12:
I encountered a similar issue when working with Python-based tools on a different setup. The error usually happens because the required `pyserial` module isn’t installed. If you’re on Windows, you can fix this by installing it via `pip` in the Python environment you’re using with Arduino.  

For detailed steps on setting up Python and Pip (though the guide is for Ubuntu 24.04), you can check out Vultr's [Python and Pip Installation Guide for Ubuntu](https://docs.vultr.com/how-to-install-python-and-pip-on-ubuntu-24-04). While the guide is Linux-focused, the concepts translate well to troubleshooting Python dependencies on any OS.  

Once you’ve pip set up, just run:  
```bash
pip install pyserial
```  
That should resolve the error!