espressif/arduino-esp32 #16: Issuse: pp.c 2998

stephmack, 2016-10-12:
It appears that when I use the example "WiFiClientEvents" and after connecting to a router, anywhere from 10-30 seconds, a pp.c 2998 appears on the serial monitor and then it looks likes the system hangs.

Has anyone else seen this and knows what's going on?

I don't recall seeing this when I was using a different router at home vs. at work (i'll verify tonight)...  Could it have something to do with the router it's connected to?

Thanks,

Steve

me-no-dev, 2017-02-08:
this sisue should be now resolved. If you still have issues, feel free to reopen :)

closed 2017-02-08 by me-no-dev

blmhemu, 2017-07-07:
@me-no-dev . I think this issue still persists. When i used OTA example code with D-Link router i got this error and system hangs. When i run the same code this time using mobile hotspot, it is done instantaneously. Something guess something is wrong with libraries cause the router is working fine.

me-no-dev, 2017-08-17:
can you try this now again please :)