espressif/arduino-esp32 #17: MD5 Error,

AnalogLamb, 2016-10-14:
Hello There is a MD5 Error like this: https://github.com/espressif/esp-idf/issues/50

But have resolved it, Thank Espressif.

Solution:
esptool.py has bugs. Only copy new esptool.py from here https://github.com/themadinventor/esptool/tree/bugfix/write_fails_cmp_bit_set 

and replace the old one in directory: arduino-1.6.12\hardware\espressif\esp32\tools

It's OK

me-no-dev, 2016-10-17:
that is good to know even though I do not have this problem on OSX.
I would rather have this fixed in esp-idf and merged here then

me-no-dev, 2016-10-17:
let's keep this open until resolved for all

me-no-dev, 2017-02-08:
issue is now resolved :)

closed 2017-02-08 by me-no-dev

architmuchhal12, 2017-05-25:
Hey,

I am trying to access the following link to get the new _esptool.py_ because I am encountering the MD5 Flashing error.

[https://github.com/themadinventor/esptool/tree/bugfix/write_fails_cmp_bit_set](url)

Can anyone post the working link?
Thanks!