espressif/arduino-esp32 #20: WiFi.h collides with the internal Arduino WiFi.h

Sweet-Peas, 2016-10-17:
This is what I get when trying to include "WiFi.h"

Multiple libraries were found for "WiFi.h"
 Used: /home/pontus/Programs/arduino-1.6.11/libraries/WiFi
 Not used: /home/pontus/Programs/arduino-1.6.11/hardware/espressif/arduino-esp32/libraries/WiFi
Using library WiFi at version 1.2.7 in folder: /home/CORPUSERS/23061151/Programs/arduino-1.6.11/libraries/WiFi 

See how it picks the wrong library. Someone should probably rename it back to EspWiFi.h or something like that.

me-no-dev, 2016-10-17:
hmm... the other way around here... I was so hoping to finally get rid of the ESP prefix.
I'll investigate further. Note that I have the core installed in my [sketchbook]/hardware/espressif...
```text
Multiple libraries were found for "WiFi.h"
 Used: /Users/ficeto/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino-1.6.12.app/Contents/Java/libraries/WiFi
Using library WiFi at version 1.0 in folder: /Users/ficeto/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi 
```

me-no-dev, 2016-10-17:
same thing if core is in Arduino install folder
```text
Multiple libraries were found for "WiFi.h"
 Used: /Applications/Arduino-1.6.12.app/Contents/Java/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino-1.6.12.app/Contents/Java/libraries/WiFi
Using library WiFi at version 1.0 in folder: /Applications/Arduino-1.6.12.app/Contents/Java/hardware/espressif/esp32/libraries/WiFi 
```

Sweet-Peas, 2016-10-19:
I tried installing Arduino IDE 1.6.12 and moved the esp32 environment to Arduino/hardware but with the same result. I have no idea why the behavior is different between our two machines.

clowrey, 2016-12-01:
I have had this sort of behavior aswell.. where it should pick the version in Arduino/hardware folder but does not, I suspect it is a bug in Arduino IDE.

smjaberl, 2017-01-27:
Me too! Is there any solution for the problem?

me-no-dev, 2017-01-27:
Solution would be to find what is common between you and include workaround in the instructions. I personally can not come up with a reason to why this happens. Maybe it's something in the settings that Arduino stores for each user of a computer.

igrr, 2017-01-27:
Personally I think using "ESP32WiFi.h" would be perfectly fine; we're not aiming to be 100% compatible with the original WiFi.h library anyway. Even the official Arduino WiFi library for Arduino 101 is called "WiFi101.h", so i don't think that naming it with a chip/platform-specific prefix explicitly is a bad thing.

smjaberl, 2017-01-28:
hey guys,  it's working fine now! 

I read the instructions and found the option "recursively clone submodules too" was checked in the git GUI. I use TortoiseGit, and there I checked the option "recursive" for a new checkout. Then I followed the instructions and now Its working.

@me-no-dev: Please write it to the instructions that the "recursive" option is needed!

@igrr:  Yes I think it's generally a good way to name it "ESP32WiFi.h" or "WiFiESP32.h".

per1234, 2017-07-11:
The Arduino IDE uses the architectures value in library.properties as one of the criteria for deciding library include priorities. The architecture of the board is determined by the architecture folder of the hardware package. As you can see, @Sweet-Peas did not correctly follow the installation instructions, causing the architecture folder to be named `arduino-esp32` rather than `esp32`:
```
Not used: /home/pontus/Programs/arduino-1.6.11/hardware/espressif/arduino-esp32/libraries/WiFi
```
library.properties for the WiFi library [sets ](https://github.com/espressif/arduino-esp32/blob/13bb22de301de79f600ff6b8bd5fafad2ccb2d95/libraries/WiFi/library.properties#L9)the `architectures` value to `esp32`:
```
architectures=esp32
```
The WiFi library bundled with the Arduino IDE sets the `architectures` value like so:
```
architectures=*
```
Since the wildcard match was better than the architecture mismatch and the WiFi library bundled with the Arduino IDE was given include priority.

In the case of @me-no-dev the installation instructions were followed correctly, causing the architecture folder to be named `esp32`:
```
 Used: /Users/ficeto/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
```
and the Arduino IDE chose the correct library.

So I don't think it is necessary to rename the library. As soon as you can add Boards Manager installation support the likelihood of the package being incorrectly installed will be greatly reduced.

cmaxmedia, 2017-12-28:
Hi, not sure if i am right here, but i have the same issue:

Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\cmaxmedia\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Fehler beim Kompilieren für das Board WEMOS LOLIN32.

(Multiple libraries were found for "WiFi.h")

is there any solution where i not have to modify any files?
thnx Markus

yekai1030, 2018-03-12:
per1234 is right. Try not to modify any files, that will cause other unknow problemes, just modify your directory name to "esp32". For me it was "arduino-esp32-master" and then i have modify it to "esp32", and that's work!

WayneKeenan, 2018-04-17:
I have installed into the correct location (``/Users/wayne/Documents/Arduino/hardware/espressif/esp32/`) and am still getting the error for Adafruit ESP32 Feather:

```
Multiple libraries were found for "WiFi.h"
 Used: /Users/wayne/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino_1.8.4.app/Contents/Java/libraries/WiFi
```

Why not just follow the ESP8266 explicit naming convention and not surprise people or rely on IDE features that seem brittle?

per1234, 2018-04-17:
@WayneKeenan that's no error. It's just some helpful information the Arduino IDE provides. Read what it says carefully. We would expect the ESP32 version of the WiFi library to be used and sure enough, what does it say?:

>    Used: /Users/wayne/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi

That would only be a problem if you actually wanted to use the Arduino WiFi library. Considering you're compiling for an ESP32, I find that highly unlikely.

WayneKeenan, 2018-04-17:
yes, sorry, I realised it was another 'legacy' name related error, had already deleted my comment.

ChrisNZ2016, 2018-04-28:
I'm getting the same result, but with a different cause:
`In file included from /Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/stdexcept:39:0,
Multiple libraries were found for "WiFi.h"
 Used: /Users/username/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/array:38,
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/tuple:39,
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:55,
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/memory:79,
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi/src/WiFiClient.h:28,
                 from /Users/username/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi/src/WiFi.h:37,
                 from /Users/username/Documents/Arduino/libraries/CayenneMQTT/src/CayenneMQTTESP32.h:25,
                 from /Users/username/Documents/Arduino/libraries/CayenneMQTT/examples/Connections/ESP32/ESP32.ino:17:
/Users/username/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/string:53:34: fatal error: bits/basic_string.tcc: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.
`
As noted in the log, compilation fails due to this, so can't be uploaded to the board. Arduino 1.8.5

per1234, 2018-04-28:
Extracted from your nightmare blob of output (use [code fencing](https://guides.github.com/features/mastering-markdown/)!!!):
```
Multiple libraries were found for "WiFi.h" 
Used: /Users/username/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi 
Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
```
So the correct WiFi library was used and therefore whatever problem you're having is completely unrelated to this issue.

ChrisNZ2016, 2018-04-28:
I think I might have found the issue. I've got storage optimisation turned on for my mac and it's marked a whole load of files as not needing to be held locally. This file was one of them. Error was being caused by the file not being available locally, even though it existed.

pandadom, 2018-05-05:
i had the same problem, but solved it by not including #include <Arduino.h> in the sketch

shehrazjazib, 2018-05-09:
I was facing the same problem and now I resolved it. You include more than one library for a specific task, like in my case, I had included 2 wifi libraries and I was facing the problem. So you need to delete unrelated libraries and there would be only one library( only 1 wifi library).  it will work fine.

tw-lottery, 2018-05-23:
I got the same issue, and it's been fixed by a `git pull --rebase`  in folter `/Documents/Arduino/hardware/espressif/esp32`.

Not sure if you have run `git submodule update --init --recursive` before you run `git pull` inside this folder.

embeddednerd, 2018-06-12:
I solved it by using #include <Arduino.h> in the sketch, not sure what difference it makes but it is working.

micjm, 2019-02-05:
> I think I might have found the issue. I've got storage optimisation turned on for my mac and it's marked a whole load of files as not needing to be held locally. This file was one of them. Error was being caused by the file not being available locally, even though it existed.

This solved the problem for me.

ozcanserhat, 2019-02-24:
Chose different board from Tools->Board after then build your code. That forces Arduino IDE to rebuilt. Then select your actual target board. When you build it should be solved. At least it worked for me.

majikthys, 2019-08-23:
> Personally I think using "ESP32WiFi.h" would be perfectly fine; we're not aiming to be 100% compatible with the original WiFi.h library anyway. Even the official Arduino WiFi library for Arduino 101 [...]

The confusion that this name collision causes, even when working nominally, is widespread.

1amchris, 2019-11-08:
I kept getting this error where the two libraries would collide so the compilation would fail. I've decided explicitly tell the Arduino IDE that I wanted to use the ESP32's WiFi library. Works flawlessly now!

If you want to use explicitly the ESP32 <WiFi.h> library I've created a <WiFi32.h> library. You can get it from my repo. link : github.com/1amchris/WiFi32.h

Shahzainali, 2020-01-10:
Fixed this issue after updating to 1.0.4, Thanks

vogesdg, 2020-04-08:
Hi I get these errors after new install of Arduino 1.8.12. Any davice will be appreciated.
Arduino: 1.8.12 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'void _handle_async_event(lwip_event_packet_t*)':

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:183:68: error: no matching function for call to 'AsyncClient::_s_dns_found(const char*&, ip_addr_t*, void*&)'

         AsyncClient::_s_dns_found(e->dns.name, &e->dns.addr, e->arg);

                                                                    ^

In file included from C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:24:0:

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.h:137:17: note: candidate: static void AsyncClient::_s_dns_found(const char*, ip_addr*, void*)

     static void _s_dns_found(const char *name, struct ip_addr *ipaddr, void *arg);

                 ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.h:137:17: note:   no known conversion for argument 2 from 'ip_addr_t* {aka _ip_addr*}' to 'ip_addr*'

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'bool _start_async_task()':

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:221:141: error: 'xTaskCreateUniversal' was not declared in this scope

         xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);

                                                                                                                                             ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'void _tcp_dns_found(const char*, ip_addr*, void*)':

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:322:59: error: invalid application of 'sizeof' to incomplete type 'ip_addr'

         memcpy(&e->dns.addr, ipaddr, sizeof(struct ip_addr));

                                                           ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: At global scope:

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:350:32: error: field 'call' has incomplete type 'tcpip_api_call_data'

     struct tcpip_api_call_data call;

                                ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:350:12: note: forward declaration of 'struct tcpip_api_call_data'

     struct tcpip_api_call_data call;

            ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_output(tcp_pcb*, int8_t)':

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:390:70: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_output_api, (struct tcpip_api_call_data*)&msg);

                                                                      ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:390:70: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:350:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_write(tcp_pcb*, int8_t, const char*, size_t, uint8_t)':

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:413:69: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]
[... 123 lines cut ...]
        ^

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:982:33: error: invalid use of incomplete type 'struct ip_addr'

         connect(IPAddress(ipaddr->u_addr.ip4.addr), _connect_port);

                                 ^

In file included from C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:24:0:

C:\Users\Daniel\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.h:53:8: note: forward declaration of 'struct ip_addr'

 struct ip_addr;

        ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Daniel\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

lbernstone, 2020-04-08:
You have hijacked a thread which has nothing to do with your issue.  Man up, open a new issue, and format it properly.

yilianziyu, 2020-05-18:
Mark,meet the same issue

PhilippMolitor, 2020-09-03:
Is there still no real solution to this? I'm currently unable to get any basic sketch to work because of this.

per1234, 2020-09-03:
>Is there still no real solution to this?

@PhilippMolitor the only problem is user error. The solution is to follow the instructions and use the platform correctly. Please read the explanation I provided earlier in the thread:

https://github.com/espressif/arduino-esp32/issues/20#issuecomment-314582178

tonyarkles, 2020-09-22:
> > Is there still no real solution to this?
> 
> @PhilippMolitor the only problem is user error. The solution is to follow the instructions and use the platform correctly. Please read the explanation I provided earlier in the thread:

I followed the instructions to install stable via Board Manager and am running into this on Catalina.

```
Multiple libraries were found for "WiFi.h"
 Used: /Users/tonyarkles/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /private/var/folders/hr/3xpw78hd2zz_h46055xfjfr00000gn/T/AppTranslocation/C04144EC-16D8-43EB-B13B-E48A50A1C68A/d/Arduino.app/Contents/Java/libraries/WiFi
```

per1234, 2020-09-22:
It's working perfectly.

The "Used" library is the one that should be used.

The "Not used" library is the one that shouldn't be used.

Case solved.

majikthys, 2020-09-26:
Sometimes functioning is far from "working perfectly". When we are talking about working within the Arduino IDE, an explicitly entry level programming environment, our goals should prioritize reducing confusion for novice developers. 

It is likely that changing the api now would to address this would be too onerous because it would break backwards compatibility, and that's a fine position to take. 

A 'solution' to this 'bug' might simply be some generous and accessible documentation.

But I would suggest when this ticket is closed, it is done with recognition that it's confusing for many and therefore not ideal.

per1234, 2020-09-26:
I think it would be helpful to make the documentation very clear that only platform developers and beta testers should use the "Using Arduino IDE with the development repository" installation method. Novices should always use Boards Manager.

It might be helpful to make it more clear the installation instructions are an OR. I occasionally see people posting on the Arduino forum who follow Boards Manager installation instructions, and then keep going down the list to also do the manual `git clone` installation! I think the format of ESP32's instructions makes this less likely to happen though.

As for the 'It says: Multiple libraries were found for "WiFi.h", what is wrong?' issue, that could be added to an FAQ.

npsantini, 2021-02-05:
Is there a solution to this problem? I just installed the Arduino IDE on Mac OS big sur and I get the same issues that other people were getting. When I tried to use the absolute path to the file it still fails because the paths to the dependencies are relative as well.

```
Multiple libraries were found for "WiFi.h"
 Used: /Users/nicksantini/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
fork/exec /Users/nicksantini/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py: permission denied
```

Any help with this would be greatly appreciated.

sansillusion, 2021-02-05:
Your problem is NOT due to multiple libraries. You can see in your log that
it chose the right one. But failed to access esptool.py with a permission
denied. Now I can't help here as I dont use mac products.

surendrakumarmali, 2021-02-18:
Updating the esp2 to 1.0.5.rc7(recent) works for me

npsantini, 2021-02-25:
> Updating the esp2 to `1.0.5.rc7` (recent) in Board Manager works for me (MacOS BIG SUR 11.2.1)

THANK YOU!

I have spent so much time trying to figure this out and this was the only solution that has actually worked. This solves the multiple libraries problem on Mac OS Big Sur.

hoc-anms, 2021-04-18:
I also have a sample case :(

lbernstone, 2021-04-18:
I think the original issue is very much solved, as it is now possible to install through board manager, which will place files in the correct location.  If you have a problem, please open a new issue.

PhilippMolitor, 2021-04-19:
Also, moving from Arduino IDE to PlatformIO solved this problem, while improving my code quality a lot and also adding all the good IDE features the VS Code has. Really a step forward.

VojtechBartoska, 2021-04-22:
how @lbernstone mentioned this issue is mostly solved. Please open a new one if you face another problem.

closed 2021-04-22 by VojtechBartoska