espressif/arduino-esp32 #23: Blink.ino only works when Serial Monitor is open

hexagon5un, 2016-10-24:
Steps to reproduce: 
1) flash Blink,ino with an LED on GPIO13
2) Hit reset.
3) No blinky.
4) Open serial terminal of your choice.
5) Hit reset.
6) Blinks.

Multiple resets fail to get blinking, and the first one after a serial terminal is connected works every time. The same effect occurs regardless of which pin is used for GPIO.

Fun bug!  That baffled me for a bit.  

(Using Arduino 1.6.8 and fresh pulls from esp-idf and arduino-esp32 from today. Board is a blue WROOM dev board.)

me-no-dev, 2016-10-24:
this has to do with the dev board hardware design rather than code issue :)
should be fixed in the next hardware revision of the dev module

hexagon5un, 2016-10-25:
Yup.  Sorry about that.  [Here](https://github.com/espressif/esp-idf/issues/59) is the link to the discussion about the hardware.  I'll close this one.

It was such a strange bug, I figured it had to be software.  :)

closed 2016-10-25 by hexagon5un

jjmonteiro, 2019-07-16:
Hi guys, sorry for reviving an old issue but is there any hw or sw fix for this?