espressif/arduino-esp32 #27: DNSServer & WebServer

moelski, 2016-10-27:
Hi !

Do you have an idea when DNSServer & WebServer will be embedded? 

regards
  Dominik

me-no-dev, 2016-10-27:
when the issues we currently have with the network are fixed upstream

moelski, 2016-11-11:
Hi !

Did you knw when the network issues are done? 
Or do you have to wait for the ESP-IDF 1.0 release to finish this stuff? 

regards
   Dominik

igrr, 2016-11-11:
That's currently in backlog. With some luck, we will get these issues fixed in 1.0.

josmunpav, 2016-12-25:
Any updates on WiFiServer? Really need this

Jorgen-VikingGod, 2017-02-21:
@josmunpav did you try this example? [SimpleWiFiServer.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino)

josmunpav, 2017-02-21:
Hi Jorgen, in the end I moved to mongoose as both http server and client. More versatile, well documented and working like a charm on the ESP32. Learning curve is harder than the standard Arduino WiFi classes but I think is worth it if your application is complex

Thanks anyway

sansillusion, 2017-07-13:
DNSServer can be copied from esp8266 to esp32 and it works (change esp8266 to esp32 in definition for no warning)

sansillusion, 2017-07-13:
for WebServer I currently use https://github.com/bbx10/WebServer_tng while waiting for an official version

CelliesProjects, 2018-03-31:
I am a bit late to this party, but I use https://github.com/me-no-dev/ESPAsyncWebServer with good results.