espressif/arduino-esp32 #29: configTime() method not implemented

forthlightning, 2016-11-03:
Porting over code from ESP8266 and hoping for an analog to [time.c in the 8266 core].(https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/time.c)

I was digging around in [time.h] (https://github.com/espressif/arduino-esp32/blob/03f5d62323f238552de57e91b48cff41a7a7009c/tools/sdk/include/newlib/time.h) but the types and pointers were out of my comfort zone. 

Specifically configTime() was a very easy way to get the system clock synced with internet time.

lbernstone, 2016-11-12:
This depends on sntp.c, which in turn depends on lwip components that are not working yet.

igrr, 2016-11-12:
There is an sntp example in esp-idf, you can it in Arduino as well.

closed 2017-01-24 by me-no-dev

forthlightning, 2017-01-24:
Hi, has anyone confirmed this works? 

I'm using this function, as called from my wifi_event_handler()'s SYSTEM_EVENT_STA_GOT_IP event. These calls are unsuccessful in getting time. 

```
void init_time()
{
    long int gmt_offset_sec = -25200;
    int dst_offset_sec = 0;
    configTime( gmt_offset_sec, dst_offset_sec , "pool.ntp.org" );
    struct tm timeinfo;
    if( !getLocalTime( &timeinfo, 2000 ) )
    {
    	printf( "failed to obtain time\n" );
    	return;
    }
    time_t now;
    printf( "%ld\n",time( &now ) );
}

void wifi_event_handler( WiFiEvent_t event )
{
    printf("[WiFi-event] (%d)\n", event);

    switch(event)
    {
    case SYSTEM_EVENT_STA_GOT_IP:
    	wifi_got_ip_addr = true;
        break;

    case SYSTEM_EVENT_STA_DISCONNECTED:
    	wifi_got_ip_addr = false;
        break;

    default:
    	break;
    }
}
```

me-no-dev, 2017-01-24:
I init time on SYSTEM_EVENT_STA_GOT_IP and have no problems. 
almost that same code above ;)

me-no-dev, 2017-01-24:
I do not see you calling init_time anywhere above

forthlightning, 2017-01-26:
Not sure what I was doing wrong but it works now! Thanks.