espressif/arduino-esp32 #30: Ticker not implemented

forthlightning, 2016-11-03:
My sketch is in large part based on [Ticker.h](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ticker/Ticker.h) in the Arduino core for the esp8266. Any idea when this library will be implemented on the 32?

iAchyut, 2017-07-28:
How can I implement Ticker.h for my ESP32 ?

forthlightning, 2017-08-01:
I use timers: http://www.freertos.org/FreeRTOS-Software-Timer-API-Functions.html

There is a bit more code for configuration but they work nicely.

lonerzzz, 2017-08-01:
I ported my code over to using FreeRTOS as well and it works quite reasonably.

sansillusion, 2017-09-03:
Any news on that ?
It would make it easier to code rather than using if conditions to detect esp8266 or esp32 etc...

bertmelis, 2017-11-05:
If you want backward (of is it forward?) compatibility you can use a quick and dirty library: https://github.com/bertmelis/Ticker-esp32

copercini, 2018-06-02:
Closed due it was implemented here: https://github.com/espressif/arduino-esp32/pull/1057

closed 2018-06-02 by copercini