espressif/arduino-esp32 #33: WiFiClient.ino, [E][connect():62] error: 23

tamberg, 2016-11-07:
WiFiClient.ino example works fine for about 10 POSTs, then fails with
`[E][connect():62] error: 23`

me-no-dev, 2016-11-08:
there are many things that still need to be improved in the network stack upstream in IDF. I can promise you that things will get better. I am also working on a different implementation that is achieving much higher speeds even now, but still fail because of the issues I mentioned.

tamberg, 2016-11-08:
No worries. It works good enough to run our (Arduino WiFi.h based) library https://yaler.net/esp32

keith204, 2016-11-13:
I'm running into this as well.

me-no-dev, 2016-12-30:
Can you confirm this is still the case?

keith204, 2016-12-30:
I pulled the latest code just now, but have not updated the firmware on the ESP32.  Should I update the firmware? 

Here's the result:  a few dozen success stories, and then:

HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET,POST,DELETE
Access-Control-Allow-Headers: X-Requested-With, Phant-Private-Key
Content-Type: text/plain
X-Rate-Limit-Limit: 300
X-Rate-Limit-Remaining: 266
X-Rate-Limit-Reset: 1483074738.762
Date: Fri, 30 Dec 2016 05:04:56 GMT
Connection: close
Transfer-Encoding: chunked
Set-Cookie: SERVERID=; Expires=Thu, 01-Jan-1970 00:00:01 GMT; path=/
Cache-control: private

a
1 success

0

closing connection
connecting to data.sparkfun.com
Requesting URL: /input/g6R85OZ9glhlE6EXjbWw?private_key=qznmqPG2VpF0nJn7XqwV&humidity=67&temp=3.7
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET,POST,DELETE
Access-Control-Allow-Headers: X-Requested-With, Phant-Private-Key
Content-Type: text/plain
**X-Rate-Limit-Limit: 300Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.**
Register dump:
PC      : 0x400847f7  PS      : 0x00060033  A0      : 0x80083dbe  A1      : 0x3ffdbf00  
A2      : 0x0000001d  A3      : 0x3ffdbf84  A4      : 0x00000004  A5      : 0x3ffd59fc  
A6      : 0x3ffdbfb0  A7      : 0x00000000  A8      : 0x800828a2  A9      : 0x3ffdbef0  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3ffc083c  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400847f7:0x3ffdbf00 0x40083dbe:0x3ffdbf20 0x40082b91:0x3ffdbf40 0x400e1c7d:0x3ffdbf80 0x400db23d:0x3ffdbfb0 0x400e1ed8:0x3ffdbfe0 0x400e21e4:0x3ffdc000 0x400daac0:0x3ffdc040 0x400daee5:0x3ffdc0c0 0x400daf3d:0x3ffdc0f0 0x400d0fe0:0x3ffdc110 0x401061c5:0x3ffdc130 0x400d1e30:0x3ffdc160 0x400d1e6c:0x3ffdc180 0x400d0e72:0x3ffdc1a0 0x400fefc5:0x3ffdc200

**CPU halted.**

me-no-dev, 2016-12-30:
yes you should update/rebuild/reflash the firmware on your ESP32 else updates will not get to it :)

keith204, 2017-01-03:
me-no-dev,

To be clear, I used the Arduino IDE to re-upload the _sketch_ to the ESP32 prior to my testing.

Since then, I've been trying to use the ESP32 Download Tool (similar to what I have used to update the firmware of ESP8266 before) and have been unsuccessful with so far.

Before trying to resolve these ESP32 Download Tool firmware update troubles, can you confirm this is what I ought to be doing?  Or, was re-uploading the sketch enough?

me-no-dev, 2017-01-04:
I recommend you upload through the IDE as that is proven working. Also to decode the above backtrace, you can use https://github.com/me-no-dev/EspExceptionDecoder and we can see better what is going on :)

keith204, 2017-01-04:
Ok, I did upload through the IDE, so I will debug the exception using the tool you mentioned.

me-no-dev, 2017-02-08:
do you guys still have this problem? If so please reopen the issue!

closed 2017-02-08 by me-no-dev

keith204, 2017-02-21:
I'm newish to GitHub, and am glad to see there's a close/reopen process.  I think closing the issue is best for now.  It may be a month or two before I resume this type of work, and if nobody else is running into this, it is probably a user error.