espressif/arduino-esp32 #40: I2C and digitalWirte problematic with Arduino 1.6.12

peterotte, 2016-11-10:
Currently, I2C (e.g. the simple example from https://www.arduino.cc/en/Tutorial/MasterReader) does not work under Arduino 1.6.12. Tested with macOS and Ubuntu 64bit and the lastest version from arduino-esp32.
Looking at the I2C signals with a scope shows a constant HIGH on SCL (actively driven) and an SDA in high impedance mode.
At the same time no digitalWrite to change an output does work as long as I2C is somewhere used in the code.
None of these errors show up with Arduino 1.6.9 (and same code).

me-no-dev, 2016-11-10:
What you describe is really strange as esp32-arduino has little to do with the IDE. When you say digitalWrite, on which pins do you want to write? SCL is currently set as OUTPUT and SDA as OUTPUT_OPEN_DRAIN as you have noticed, but SCL should have been running. I really do not see how the IDE could influence the result.

me-no-dev, 2016-12-15:
All your troubles should be now fixed :)

closed 2016-12-15 by me-no-dev