espressif/arduino-esp32 #42: I2C apparently not working on builds as of this post date

nkolban, 2016-11-11:
Date - 2016-11-11.  I am testing an I2C Scanner that I believe worked a calendar week ago.  I have re-downloaded the latest ESP-IDF and the latest Arduino and re-built all.  Now the results are not properly coming back.  I attached a logic analyzer and there are the results

[image]

The code I was running was:

```
#include "Wire.h"
#include "esp_log.h"
#include "driver/gpio.h"

#define SDAPIN (GPIO_NUM_21)
#define SCLPIN (GPIO_NUM_22)

static char tag[]="i2cscanner";

extern "C" {
	void I2CScanner();
}

void I2CScanner() {
	ESP_LOGI(tag, "I2C scanning with SDA=%d, CLK=%d", SDAPIN, SCLPIN);
	Wire.begin(SDAPIN, SCLPIN);
	int address;
	int foundCount = 0;
	for (address=10; address<127; address++) {
		Wire.beginTransmission(address);
		uint8_t error = Wire.endTransmission();
		if (error == 0) {
			foundCount++;
			ESP_LOGI(tag, "Found device at 0x%.2x", address);
		}
	}
	ESP_LOGI(tag, "Found %d I2C devices by scanning.", foundCount);
}

```

ladyada, 2016-11-12:
downloaded today & using just "Wire.begin()", have i2c working with a few different chipsets: PCF8523, HT16K33, SSD1306 so far. doesn't mean all is perfect but i've got general functionality working

me-no-dev, 2016-11-12:
@nkolban here is what I used for scanning:
```cpp
void scanI2C(){
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 0x01; address < 0x7f; address++){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      Serial.print("I2C device found at address 0x");
      if (address<16) Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
      nDevices++;
    } else if (error == 4){
      Serial.print("Unknow error at address 0x");
      if (address<16) Serial.print("0");
      Serial.println(address,HEX);
    }
    //delay(1);
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done");
}
```

nkolban, 2016-11-12:
Thank you sir ... One of the core differences between your app and mytest is that I am using alternate Pins for I2C than the defaults.  Let me try with the defaults and see if it makes any differences.

me-no-dev, 2016-11-12:
it should not, because the default pins are default only because I and Ivan have picked those particular pins. Underneath in reality there are no default pins. Every pin (almost) is the same in that regard. Default pins are software defined only, the hardware itself does not have dedicated I2C pins.
SPI and UARTs have some hardware pin definitions but even there I have defined them purely in software and any other pin will work just as well! Pin Matrix FTW!!!

nkolban, 2016-11-12:
Thank you all for the posts.  Based on the high confidence that the functions worked I went back and re-checked my circuit assembly and ... sure enough, I had wired incorrectly with commensurate unpredictable results.  After re-wiring, I no longer had issues.  The error was 110% at my end and apologize for the distraction.

closed 2016-11-12 by nkolban