espressif/arduino-esp32 #44: shiftIn/shiftOut

ladyada, 2016-11-12:
not a super common set of Arduinoese but they popup here and there, just toss in the standard wiring_shift.c from your fav arduino dist

https://github.com/adafruit/arduino-esp32/commit/5decd2c1e223b6b0f20957e942298ed1a38001f7

i had to comment out a random() prototype that the compiler was complaining about @ https://github.com/adafruit/arduino-esp32/commit/871f28ed9b594bf4b4e82bc124293024b878a552

didn't actually *test* shiftin/out since just needed the reference for compiling an old project

me-no-dev, 2016-11-12:
Will take notice of the random issue! Thanks!
As for ShiftIn and ShiftOut, I left those out intentionally, because there is a counter hardware in the esp32 that will do the job much much better than the usual implementation :) Even though ShiftOut might end up the same code as now.
Maybe PR the changes except shiftIn? Does that make sense?

ladyada, 2016-11-12:
well, there's a lot of code that uses shiftIn() - replacing it with non-digitalWrite code is probably a good idea down the road. 
For compatibility with existing ESP8266 code, i'd recommend just putting it in as is and then having a todo-optimize list to speed it up later :)
that said, the random() prototype complaint needs fixin' either way

me-no-dev, 2016-11-13:
random fix [here](https://github.com/espressif/arduino-esp32/commit/2515a3cfed79488ee68effa9ef2dc11c071b9124)
shiftIn/Out to go next

me-no-dev, 2016-11-13:
shiftIn/shiftOut added [here](https://github.com/espressif/arduino-esp32/commit/2cc1a6a1dabfd3359af80321dc53cf953383685f)

ladyada, 2016-11-15:
yay!

closed 2016-11-15 by ladyada