espressif/arduino-esp32 #45: Cannot flash Nano32

luc-github, 2016-11-12:
I have a Nano32 received today and I am using today git version with ide 1.6.9 under Windows 10 using default settings.
When I try to flash any sample like WiFiClientBasic on my Nano32, it failed :
```
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Erasing flash...
Compressed 4192 bytes to 2755...

Writing at 0x00001000... (100 %) 
Wrote 4192 bytes (2755 compressed) at 0x00001000 in 0.1 seconds (effective 632.8 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
```

Any tweak is necessary to flash ?

peterotte, 2016-11-12:
Try push and hold pin 0 down before flashing. Release it not before flashing successfully starts. Helped for my ESP32 Thing from Sparkfun under Linux.

luc-github, 2016-11-12:
thanks @peterotte I did it several times without success then it suddently works - and now no need even to change push the flash button

weird - I guess it is hardware problem but as I have only one unit I cannot compare

Thank you for your help

closed 2016-11-12 by luc-github