espressif/arduino-esp32 #47: simplify creation of Serial1

ladyada, 2016-11-13:
since Serial is hardcoded for debug, its nice to use one of the extra uarts for talkin' to a GPS

this diff:
https://github.com/adafruit/arduino-esp32/commit/1bf6318a0735921e0a9779a78a0977892c4a6623
makes it easier, just add  HardwareSerial Serial1(2); and in setup: Serial1.begin(115200); 

note that uart #1 is not available because it is shared with SPI flash, so it was not tested. uart #2 works great on gpio #16 + #17. no other gpio's are available

me-no-dev, 2016-11-13:
I'm totally cool with this :) You want to PR the change, or do you want me to add it together with shiftIn/Out stuff? Also if flash is DIO, UART1 should be available.

me-no-dev, 2016-11-13:
added [here](https://github.com/espressif/arduino-esp32/commit/f0b7959425623e3229aadadd4145560b4064aa79) thanks!

closed 2016-11-13 by ladyada