espressif/arduino-esp32 #51: deepSleep available?

peterotte, 2016-11-15:
Are different sleep modi already supported? E.g. the deepSleep from ESP8266.

me-no-dev, 2016-11-16:
soon :) there is some API exposed but not to switch the sleep mode yet

GiraffeGiraffe, 2016-12-01:
Any timeline for this?

torntrousers, 2016-12-16:
Using the system call esp_deep_sleep(5000000); seems to work with the latest code and sleeps for that many microseconds.

igrr, 2016-12-16:
Also some other wake modes are now implemented (in the ESP-IDF): http://esp-idf.readthedocs.io/en/latest/api/deep_sleep.html

torntrousers, 2016-12-16:
Is it possible to use those from an Arduino sketch somehow?

igrr, 2016-12-16:
You can `#include "esp_deep_sleep.h"` and call these functions. This does require latest ESP-IDF, with Arduino core added as a component.

GiraffeGiraffe, 2016-12-18:
Now it works:  esp_deep_sleep(5000000); Power current is 11uA. 

Any way to get hibernation mode deep sleep (2.5uA) ?

igrr, 2016-12-18:
Possibly [by powering down RTC fast memory](http://esp-idf.readthedocs.io/en/latest/api/deep_sleep.html#_CPPv224esp_deep_sleep_pd_config26esp_deep_sleep_pd_domain_t26esp_deep_sleep_pd_option_t). I haven't done current measurements myself yet, so can not say what the resulting current will be.
Also there might be something related to clock selection — right now we are running from 150kHz RC oscillator. I need to check with hardware folks whether running from 32kHz crystal will reduce current (it probably should).

GiraffeGiraffe, 2016-12-19:
Thanks, I will try this.

GiraffeGiraffe, 2016-12-21:
How do I get #include esp_deep_sleep.h  with arduino?
I don't know how to call esp_deep_sleep_pd_config from arduino?
Can anybody help me?

closed 2017-02-08 by me-no-dev

Ashok07, 2017-03-24:
esp_deep_sleep(15000000); is work for me
but the problem is, when ever the system is wake up it starts running from main(setup), but i just want to run loop only. which means i just want to run my code from where i left in sleep mode;
pls help me to solve this.

matmat94, 2017-03-24:
I don't know if static or global variables are kept in memory after a wake-up, but you could still store data in flash such as variables you want to save and maybe a function pointer to the resume fonction to be executed just after the deep sleep.

moujix, 2017-10-25:
hy
someone help me to make esp sleep for many hour or wake up in specific hour
have you any example with arduino

AndreKR, 2017-12-15:
@moujix https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/ExternalWakeUp/ExternalWakeUp.ino#L26

Rita-GE, 2020-02-27:
Hi,everybody.
How to get hibernate?
Is there any example?

martinius96, 2020-07-31:
Nothing new about hibernation in Arduino core?