espressif/arduino-esp32 #52: using SPI causes IllegalInstruction exception

elFallino, 2016-11-15:
I try to get Adafruits ILI9341 TFT library working.
I found that SPI seems not to work and could track it down to the call of _pinMatrixOutAttach_ in esp32_hal_spi.c

Arduino  sketch:
```

#include "SPI.h"

#define TFT_DC 9
#define TFT_CS 11
#define TFT_MOSI 8
#define TFT_CLK 6
#define TFT_RST 10
#define TFT_MISO 7

void setup() {
  Serial.begin(115200);
  
  Serial.println("creating object");
  SPIClass myspi = SPIClass();    //Does not matter if using HSPI, VSPI or FSPI

  Serial.println("calling begin()");
  myspi.begin(TFT_CLK, TFT_MISO, TFT_MOSI, TFT_CS);

  Serial.println("end successful");
}

void loop() {
}

```

Error on the ESP32:
```
[...]
load:0x40080000,len:260
entry 0x40080034
rtc v118 Oct 19 2016 15:22:11
XTAL 40M
creating object
calling begin()
Guru Meditation Error of type IllegalInstruction occurred on core   1. Exception was unhandled.
Register dump:
PC      :  400d05a2  PS      :  00060630  A0      :  800d053f  A1      :  3ffb68d0  
A2      :  00000002  A3      :  00000006  A4      :  3ff640bc  A5      :  00000000  
A6      :  3ff64000  A7      :  00000000  A8      :  800d0594  A9      :  3ffb68b0  
A10     :  00000026  A11     :  00000008  A12     :  00000000  A13     :  00000000  
A14     :  3ffb3540  A15     :  3ffb3508  SAR     :  0000001a  EXCCAUSE:  00000000  
EXCVADDR:  00000000  LBEG    :  400d404d  LEND    :  400d405d  LCOUNT  :  ffffffff  
CPU halted.
```

Chain of method calls as far as I tracked it:
1: In sketch: myspi.begin() 
2: SPi.cpp (method SPIClass::begin): spiAttachSCK() 
3: esp32-hal-spi.c (method spiAttachSCK): pinMatrixOutAttach()

I am using Arduino 1.6.12 and the esp32-arduino plugin from today, Nov 15th.

Let me know if i can provide more information.

ladyada, 2016-11-16:
btw the library totally works with default SPI device - was commited and tested with esp32 feather + ili9341 featherwing - slow but works :)

https://github.com/adafruit/Adafruit_ILI9341/commit/51934bde72db1434e0edce11fdf077c3a467ddcc

elFallino, 2016-11-16:
The same changes I did, too. It compiles well but fails for the same reason. So I tried to track down the point that seems to cause that.

me-no-dev, 2016-11-16:
well :) you are using the SPI flash pins to attach your LCD ;) that will surely mess things up :)

elFallino, 2016-11-16:
Oh damn, my fault. ;)
Had the opinion that I could do so as long as CS selects the right one.
Selecting other pins than the FSPI ones does fine.
Thank you.

closed 2016-11-16 by elFallino