espressif/arduino-esp32 #58: ESP32 RTCWDG reset

kriswiner, 2016-11-19:
I put together my first ESP32, downloaded the Arduino core and ran the python script and was able to flash the device successfully several times. But when I press reset after flashing the device contnually resets itself with the following serial output:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:0
load:0xfc00080f,len:355391
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:0
load:0xfc00080f,len:355391
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57
`

Any idea what could be wrong?

I used a simple blink program and my device has a 4 MB flash even though all of the boards seem to have a 1 Mbyte max. Any help would be appreciated.

kriswiner, 2016-11-20:
From the ESP32 errata I found this which sounds like the problem:

3.1. When ESP32 is powered up or wakes up from Deep-sleep, a
spurious watchdog reset occurs.
Description:
The spurious power-up watchdog reset cannot be bypassed with software.
However, when ESP32 wakes up from Deep-sleep, the watchdog reset can be bypassed
with software.
Workarounds:
After waking up from Deep-sleep, CPU will read an instruction from the RTC fast memory,
and will then execute boot program from this memory. This program needs to clear the
illegal access flag in the cache MMU by:
• setting the PRO_CACHE_MMU_IA_CLR bit in DPORT_PRO_CACHE_CTRL1_REG to
1, and then,
• clearing the bit.

Is there some plan to fix this in the Arduino core?

igrr, 2016-11-20:
A workaround for the bug is present in ESP-IDF, so this Arduino core also has it. From the output you have posted, only the first RTCWDT reset is the manifestation of this bug. The other ones are due to the first stage bootloader failing to load program from flash.

You may need to do some troubleshooting of the upload step: check if the program was written correctly to flash, check if the board is adequately powered, check if flash chip voltage is set correctly by bootstrapping pin, check if anything else is connected to flash chip lines.

kriswiner, 2016-11-20:
Thank you for the reply.

The board uses the same MIC5528 500 mA 3v3 LDO and FT230XQ USB-to-Serial for flashing as the ESP8266 and ESP8285 Dev Boards I have designed and work well. Unless the ESP32 has much greater power requirements the power should be adequate.

I did connect the flash VDD with the board 3V3 and left the ESP32 FLASH_VDD unconnected. Does the ESP32 FLASH_VDD need to be connected to 3V3?

The 4 MByte flash N25Q032A I am using I have used before with the ESP8266 and in this case I could only get the program to run using DOUT and never DIO or QIO even though the flashing seemed to work normally. I notice in the serial output the flash mode seems to be stuck in DIO. I tried to modify it to be DOUT in the board.txt file but it doesn't seem to be reflected in the serial output. Could this be the reason the bootloader is failing to load the program from flash?

When I flash the chip I get this in the terminal:

esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Compressed 5264 bytes to 3363...

Writing at 0x00001000... (100 %) 
Wrote 5264 bytes (3363 compressed) at 0x00001000 in 0.1 seconds (effective 668.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 78...

Writing at 0x00008000... (100 %) 
Wrote 3072 bytes (78 compressed) at 0x00008000 in 0.0 seconds (effective 768.0 kbit/s)...
Hash of data verified.
Compressed 225808 bytes to 67939...

Writing at 0x00010000... (20 %) 
Writing at 0x00014000... (40 %) 
Writing at 0x00018000... (60 %) 
Writing at 0x0001c000... (80 %) 
Writing at 0x00020000... (100 %) 
Wrote 225808 bytes (67939 compressed) at 0x00010000 in 3.9 seconds (effective 466.2 kbit/s)...
Hash of data verified.

Leaving...

Does this seem normal to you?

Also, what does this mean?

" check if flash chip voltage is set correctly by bootstrapping pin"

Do you mean whether the chip is put into bootloader mode by pulling GPIO 0 to GND or something else?

I very much appreciate your help!

igrr, 2016-11-20:
Please check the voltage at flash_vdd pin. Depending on the state of MTDI pin at reset, ESP32 will use either 3.3v or 1.8v output of the internal LDO for the SDIO power domain (this is mentioned in the pin list document and in the TRM). If ESP32 is using 1.8V logic levels to talk to the flash chip which is powered from 3.3V, this may cause issues at startup.

kriswiner, 2016-11-20:
ESP32 FLASH_VDD is at 3.29 V, MTDI (GPIO12) is at GND. Should I try to pull MTDI up at reset (power on)?

igrr, 2016-11-20:
Okay, probably some other issue then. I'll ask colleagues if they have any idea...

kriswiner, 2016-11-20:
Thank you. Could this be simply the same DIO vs. DOUT issue we have seen before with the ESP8266 and ESP8285?

kriswiner, 2016-12-07:
OK, tried Igor's custom bootloader with the DOUT flash mode and it is working, thank you. Here is the output just for completeness.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:3868
load:0x40078000,len:5028
load:0x40080000,len:260
entry 0x40080034
[0;32mI (19) boot: Espressif ESP32 2nd stage bootloader v. V0.1[0m
[0;32mI (24) boot: compile time 18:15:32[0m
D (28) esp_image: reading image header @ 0x1000[0m
D (32) boot: magic e9[0m
D (35) boot: segments 04[0m
D (37) boot: spi_mode 03[0m
D (40) boot: spi_speed 00[0m
D (42) boot: spi_size 01[0m
[0;32mI (45) boot: SPI Speed      : 40MHz[0m
[0;32mI (49) boot: SPI Mode       : DOUT[0m
[0;32mI (53) boot: SPI Flash Size : 2MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
D (68) bootloader_flash: mmu set paddr=00000000 count=1[0m
D (73) boot: mapped partition table 0x8000 at 0x3f408000[0m
D (79) boot: load partition table entry 0x3f408000[0m
D (84) boot: type=1 subtype=2[0m
[0;32mI (87) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
D (94) boot: load partition table entry 0x3f408020[0m
D (99) boot: type=1 subtype=1[0m
[0;32mI (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
D (110) boot: load partition table entry 0x3f408040[0m
D (114) boot: type=0 subtype=0[0m
[0;32mI (118) boot:  2 factory          factory app      00 00 00010000 00100000[0m
D (125) boot: load partition table entry 0x3f408060[0m
D (130) boot: type=ff subtype=ff[0m
[0;32mI (133) boot: End of partition table[0m
[0;32mI (138) boot: Loading app partition at offset 00010000[0m
D (143) esp_image: reading image header @ 0x10000[0m
D (148) esp_image: reading 9 image segments[0m
D (152) bootloader_flash: mmu set paddr=00010000 count=2[0m
D (228) bootloader_flash: mmu set paddr=00020000 count=1[0m
D (243) bootloader_flash: mmu set paddr=00020000 count=1[0m
D (254) bootloader_flash: mmu set paddr=00020000 count=1[0m
D (256) bootloader_flash: mmu set paddr=00020000 count=2[0m
D (337) bootloader_flash: mmu set paddr=00030000 count=1[0m
D (339) bootloader_flash: mmu set paddr=00030000 count=2[0m
D (374) bootloader_flash: mmu set paddr=00040000 count=1[0m
D (434) bootloader_flash: mmu set paddr=00040000 count=1[0m
D (434) esp_image: reading checksum block at 0x4d5e0[0m
D (435) esp_image: reading image header @ 0x10000[0m
D (438) boot: bin_header: 233 9 2 0 40080c18[0m
[0;32mI (442) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) [0m
D (450) boot: found drom segment, map from 00020010 to 3f400010[0m
[0;32mI (456) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x031f4 ( 12788) map[0m
[0;32mI (465) boot: segment 2: paddr=0x00023204 vaddr=0x3ffc2c40 size=0x02528 (  9512) load[0m
D (473) bootloader_flash: mmu set paddr=00020000 count=1[0m
[0;32mI (483) boot: segment 3: paddr=0x00025734 vaddr=0x40080000 size=0x00400 (  1024) load[0m
D (487) bootloader_flash: mmu set paddr=00020000 count=1[0m
[0;32mI (493) boot: segment 4: paddr=0x00025b3c vaddr=0x40080400 size=0x126c8 ( 75464) load[0m
D (501) bootloader_flash: mmu set paddr=00020000 count=2[0m
[0;32mI (546) boot: segment 5: paddr=0x0003820c vaddr=0x400c0000 size=0x00000 (     0) load[0m
D (546) bootloader_flash: mmu set paddr=00030000 count=1[0m
[0;32mI (549) boot: segment 6: paddr=0x00038214 vaddr=0x00000000 size=0x07df4 ( 32244) [0m
D (557) boot: found irom segment, map from 00040018 to 400d0018[0m
[0;32mI (562) boot: segment 7: paddr=0x00040010 vaddr=0x400d0018 size=0x0d5c0 ( 54720) map[0m
[0;32mI (571) boot: segment 8: paddr=0x0004d5d8 vaddr=0x50000000 size=0x00008 (     8) load[0m
D (579) bootloader_flash: mmu set paddr=00040000 count=1[0m
D (585) boot: configure drom and irom and start[0m
D (589) boot: start: 0x40080c18[0m
ÿtest blue
test blue
test blue
test blue
test blue
test blue
`

closed 2016-12-07 by kriswiner

fungineers-youtube, 2019-09-04:
Hi @kris @igr where can i get the custom bootloader and could you please advise on how to install it. Thank you!