espressif/arduino-esp32 #59: Guru Meditation Error of type InstrFetchProhibited occurred on core   0. Exception was unhandled.

mihalski, 2016-11-19:
I've compiled and uploaded all the Arduino ESP32 Wifi examples (those are the only examples that exist) and I've had almost identical results with them all..

I am using the latest arduino-esp32 from github.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0000,len:0
load:0x3ffc0000,len:640
load:0x40078000,len:3180
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
rtc v118 Oct 19 2016 15:22:11
XTAL 40M
tcpip_task_hdlxxx : 3ffc5c20, prio:18,stack:2048
frc2_timer_task_hdl:3ffc76ec, prio:22, stack:2048
Guru Meditation Error of type InstrFetchProhibited occurred on core   0. Exception was unhandled.
Register dump:
PC      :  d4d7a4be  PS      :  00060230  A0      :  8008a0c1  A1      :  3ffc3620  
A2      :  d4d7a4be  A3      :  3ffbee12  A4      :  3ffc3640  A5      :  000000fa  
A6      :  00000005  A7      :  3ffae0c0  A8      :  3f4004f8  A9      :  3ffc3600  
A10     :  00000000  A11     :  00000001  A12     :  00000000  A13     :  00000006  
A14     :  00000006  A15     :  00000000  SAR     :  0000001a  EXCCAUSE:  00000014  
EXCVADDR:  d4d7a4bc  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
CPU halted.
```
Are these examples supposed to be working? For what it's worth here is the compilation and upload:
```
C:\Users\michal\Downloads\arduino-1.6.12\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\michal\Downloads\arduino-1.6.12\hardware -hardware C:\Users\michal\AppData\Local\Arduino15\packages -tools C:\Users\michal\Downloads\arduino-1.6.12\tools-builder -tools C:\Users\michal\Downloads\arduino-1.6.12\hardware\tools\avr -tools C:\Users\michal\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\michal\Downloads\arduino-1.6.12\libraries -libraries C:\Users\michal\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10612 -build-path C:\Users\michal\AppData\Local\Temp\arduino_build_775382 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\michal\AppData\Local\Temp\arduino_modified_sketch_578225\WiFiClientBasic.ino
C:\Users\michal\Downloads\arduino-1.6.12\arduino-builder -compile -logger=machine -hardware C:\Users\michal\Downloads\arduino-1.6.12\hardware -hardware C:\Users\michal\AppData\Local\Arduino15\packages -tools C:\Users\michal\Downloads\arduino-1.6.12\tools-builder -tools C:\Users\michal\Downloads\arduino-1.6.12\hardware\tools\avr -tools C:\Users\michal\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\michal\Downloads\arduino-1.6.12\libraries -libraries C:\Users\michal\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10612 -build-path C:\Users\michal\AppData\Local\Temp\arduino_build_775382 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\michal\AppData\Local\Temp\arduino_modified_sketch_578225\WiFiClientBasic.ino
Using board 'esp32' from platform in folder: C:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32\cores\esp32" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32\variants\esp32" "C:\Users\michal\AppData\Local\Temp\arduino_build_775382\sketch\WiFiClientBasic.ino.cpp" -o "nul"
"C:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32\cores\esp32" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32\variants\esp32" "-IC:\Users\michal\Downloads\arduino-1.6.12\hardware\espressif\esp32\libraries\WiFi\src" "C:\Users\michal\AppData\Local\Temp\arduino_build_775382\ske [... cut ...]
Writing at 0x00018000... (21 %) 
Writing at 0x0001c000... (28 %) 
Writing at 0x00020000... (35 %) 
Writing at 0x00024000... (42 %) 
Writing at 0x00028000... (50 %) 
Writing at 0x0002c000... (57 %) 
Writing at 0x00030000... (64 %) 
Writing at 0x00034000... (71 %) 
Writing at 0x00038000... (78 %) 
Writing at 0x0003c000... (85 %) 
Writing at 0x00040000... (92 %) 
Writing at 0x00044000... (100 %) 
Wrote 392256 bytes (226316 compressed) at 0x00010000 in 3.7 seconds (effective 853.4 kbit/s)...
Hash of data verified.

Leaving...
```

me-no-dev, 2016-11-19:
does that happen when you use WiFiClient or just anything? Try better power source because I have had many similar problems because of bad usb cable or power source. When booting WiFi, there is a certain spike in consumption that could drop the voltage over thin USB cables enough to trip the chip

mihalski, 2016-11-19:
Thank you!

That was absolutely PART of the problem. I was relying on just an FTDI board to supply power. It still wasn't able to connect to my Wifi network when I used a proper power supply.. Just kept searching with ............. until I stopped it.

I tried WifiClientEvents and got this:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0000,len:0
load:0x3ffc0000,len:640
load:0x40078000,len:3180
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
rtc v118 Oct 19 2016 15:22:11
XTAL 40M
tcpip_task_hdlxxx : 3ffc5a2c, prio:18,stack:2048
frc2_timer_task_hdl:3ffc74f8, prio:22, stack:2048
phy_version: 187, Oct 10 2016, 19:23:46, 0
pp_task_hdl : 3ffc9d58, prio:23, stack:8192
mode : softAP(24:0a:c4:01:dd:21)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
Guru Meditation Error of type LoadProhibited occurred on core   0. Exception was unhandled.
Register dump:
PC      :  400f9ac4  PS      :  00060b33  A0      :  800f9b20  A1      :  3ffc7430  
A2      :  3ffc0f54  A3      :  3ffbe058  A4      :  3ff47024  A5      :  ffffffff  
A6      :  3ffb0ca0  A7      :  3ffbf48c  A8      :  800f9ab0  A9      :  00000000  
A10     :  00060b20  A11     :  3ffc7450  A12     :  40083e63  A13     :  3ffc3be0  
A14     :  00000001  A15     :  00000015  SAR     :  00000011  EXCCAUSE:  0000001c  
EXCVADDR:  800f9b20  LBEG    :  400fa418  LEND    :  400fa42e  LCOUNT  :  ffffffff  
CPU halted.
```
And her's an example from WifiScan:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0000,len:0
load:0x3ffc0000,len:640
load:0x40078000,len:3180
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
rtc v118 Oct 19 2016 15:22:11
XTAL 40M
tcpip_task_hdlxxx : 3ffc5a20, prio:18,stack:2048
frc2_timer_task_hdl:3ffc74ec, prio:22, stack:2048
phy_version: 187, Oct 10 2016, 19:23:46, 0
pp_task_hdl : 3ffc9d4c, prio:23, stack:8192
mode : sta(24:0a:c4:01:dd:20)
[E][WiFiGeneric.cpp:316] startWiFi(): esp_wifi_connect: 1
Setup done
scan start
scan done
no networks found

scan start
scan done
no networks found

scan start
Guru Meditation Error of type IllegalInstruction occurred on core   0. Exception was unhandled.
Register dump:
PC      :  400ca744  PS      :  00060730  A0      :  800d64bf  A1      :  3ffc5950  
A2      :  3ffb05b8  A3      :  3ffb0b14  A4      :  400db6a8  A5      :  3ffb0b14  
A6      :  00000000  A7      :  ffffffff  A8      :  800db740  A9      :  3ffc5930  
A10     :  3ffc26bc  A11     :  400815cc  A12     :  00000000  A13     :  ffffffff  
A14     :  b33fffff  A15     :  00060523  SAR     :  00000015  EXCCAUSE:  00000000  
EXCVADDR:  00000000  LBEG    :  00000000  LEND    :  00000000  LCOUNT  :  00000000  
CPU halted.
```
It's progress but I still haven't had any luck with any of the examples.

me-no-dev, 2017-02-08:
I'm pretty sure you have everything sorted by now :) if not, reopen the issue so we can have a fresh look.

closed 2017-02-08 by me-no-dev