espressif/arduino-esp32 #60: signal: segmentation fault

zhivko, 2016-11-19:
While compiling simple blink example:
```
int ledPin = 5;

void setup()
{
    pinMode(ledPin, OUTPUT);
    Serial.begin(115200);
}

void loop()
{
    Serial.println("Hello, world!");
    digitalWrite(ledPin, HIGH);
    delay(500);
    digitalWrite(ledPin, LOW);
    delay(500);
}
```

I get this error while compiling (click on verify) in arduino ide. What could be the reason?
I removed all esp8266 variables from ~/.bashrc that I used for compiling for esp8266.

```
/home/klemen/arduino-1.6.12/arduino-builder -dump-prefs -logger=machine -hardware /home/klemen/arduino-1.6.12/hardware -tools /home/klemen/arduino-1.6.12/tools-builder -tools /home/klemen/arduino-1.6.12/hardware/tools/avr -built-in-libraries /home/klemen/arduino-1.6.12/libraries -libraries /home/klemen/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=115200 -ide-version=10612 -build-path /tmp/arduino_build_557210 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /home/klemen/Arduino/led_blink/led_blink.ino
/home/klemen/arduino-1.6.12/arduino-builder -compile -logger=machine -hardware /home/klemen/arduino-1.6.12/hardware -tools /home/klemen/arduino-1.6.12/tools-builder -tools /home/klemen/arduino-1.6.12/hardware/tools/avr -built-in-libraries /home/klemen/arduino-1.6.12/libraries -libraries /home/klemen/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=115200 -ide-version=10612 -build-path /tmp/arduino_build_557210 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /home/klemen/Arduino/led_blink/led_blink.ino
Using board 'esp32' from platform in folder: /home/klemen/arduino-1.6.12/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/klemen/arduino-1.6.12/hardware/espressif/esp32
Detecting libraries used...
"/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/cores/esp32" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/variants/esp32" "/tmp/arduino_build_557210/sketch/led_blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/cores/esp32" "-I/home/klemen/arduino-1.6.12/hardware/espressif/esp32/variants/esp32" "/tmp/arduino_build_557210/sketch/led_blink.ino.cpp" -o "/tmp/arduino_build_557210/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/klemen/arduino-1.6.12/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_557210/preproc/ctags_target_for_gcc_minus_e.cpp"
signal: segmentation fault
Error compiling for board ESP32 Dev Module.
```

zhivko, 2016-11-19:
Forgot to mention - I experience issue on ubuntu linux.
Now I tried on winfows and I get to:
```
Sketch uses 102,069 bytes (9%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,160 bytes (3%) of dynamic memory, leaving 285,752 bytes for local variables. Maximum is 294,912 bytes.
python.exe C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/esptool.py --chip esp32 --port COM3 --baud 115200 write_flash -z --flash_freq 80m --flash_mode dio 0x1000 C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader.bin 0x8000 C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/bin/partitions_singleapp.bin 0x10000 C:\Users\klemen\AppData\Local\Temp\arduino_build_403827/blink.ino.bin 
esptool.py v2.0-dev
Connecting...

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

me-no-dev, 2016-11-20:
@zhivko try holding the boot button (GPIO0 low) while trying to flash. Some boards have issues resetting properly.

zhivko, 2016-11-20:
I also try gpio0 low, but according to scope trace this is what happens - io0 firmly low, but EN line doesn't make full switch from 3.3 to 0.
It seems reset should happen while io0 is low to make device go in flash mode. It seems EN line is problem on my board but cannot figure it why.
Schematics is taken from devC espressif, more info here:
[http://esp32.com/viewtopic.php?f=2&t=430&p=2176#p2176](url)

zhivko, 2016-11-20:
Ah finally. I had fault ftdi adapter. With new one <i could upload sketch ;).

On same board but another usb adapter (this time CP2102) reset en and gpio lines are like this.
[image]

Seems to work OK on arduino ide with following in boards.txt:
```

esp32.serial.disableDTR=true
esp32.serial.disableRTS=false
```

```
Sketch uses 102,069 bytes (9%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,160 bytes (3%) of dynamic memory, leaving 285,752 bytes for local variables. Maximum is 294,912 bytes.
python.exe C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/esptool.py --chip esp32 --port COM3 --baud 115200 write_flash -z --flash_freq 80m --flash_mode dio 0x1000 C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader.bin 0x8000 C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/bin/partitions_singleapp.bin 0x10000 C:\Users\klemen\AppData\Local\Temp\arduino_build_629099/blink.ino.bin 
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Compressed 5264 bytes to 3363...

Writing at 0x00001000... (100 %) 
Wrote 5264 bytes (3363 compressed) at 0x00001000 in 0.3 seconds (effective 141.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 78...

Writing at 0x00008000... (100 %) 
Wrote 3072 bytes (78 compressed) at 0x00008000 in 0.0 seconds (effective 1445.7 kbit/s)...
Hash of data verified.
Compressed 225792 bytes to 67923...

Writing at 0x00010000... (20 %) 
Writing at 0x00014000... (40 %) 
Writing at 0x00018000... (60 %) 
Writing at 0x0001c000... (80 %) 
Writing at 0x00020000... (100 %) 
Wrote 225792 bytes (67923 compressed) at 0x00010000 in 6.0 seconds (effective 302.8 kbit/s)...
Hash of data verified.

Leaving...
```
Great! thanks

zhivko, 2016-11-20:
Another question - I added led to my board to pin no 11.
I uploaded sketch, and I barely see led light instead of bright flashes - When I measured voltage on gpio11 - it shows 1.57V.

So what's wrong with using:
[code]
int ledPin = 11;
[/code]

in blink example?

penfold42, 2016-11-20:
Looks like gpio11 is used for flash
http://www.esp32.com/viewtopic.php?t=400

zhivko, 2016-11-23:
Why they mark GPIO11 when it is flash reserved? Ughh...

closed 2016-11-23 by zhivko

penfold42, 2016-11-23:
Assuming the esp32 is similar to the esp8266, under some circumstances they can be used.

There a different flash chip IO modes - DIO, QIO that use all or some subset of the pins to communicate.