espressif/arduino-esp32 #61: Unable to set hostname

sorscode, 2016-11-20:
First off great work on the Arduino Core.

On the ESP8266, I'm able to set the hostname of the of ESP via:
WiFi.hostname(myhostname);

However this does not work on the ESP32.
I do have the ArduinoJSON library and the PubSubClient library working, to do basic PWM on Meanwell LDD LED drivers.

me-no-dev, 2016-11-20:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/tcpip_adapter/tcpip_adapter.h#L386
API was added in the last commit of IDF changes, but has not made it into the Arduino classes yet

me-no-dev, 2017-02-08:
see https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino#L78 for setting hostname

closed 2017-02-08 by me-no-dev

jeroenst, 2018-11-24:
wifi.setHostname("hostname");