espressif/arduino-esp32 #62: libm.a problem: missing __ieee754_remainder  function?

minyk, 2016-11-21:
Like https://github.com/esp8266/Arduino/issues/612, When I using Adafruit's Ultimate GPS library, I met below during complie:
```
/Users/minyk/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'
/Users/minyk/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib/libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':
/home/ivan/freertos/smpemu/newlib/newlib_xtensa-2.2.0/xtensa-esp108-elf/newlib/libm/math/../../../.././newlib/libm/math/e_fmod.c:38: undefined reference to `__ieee754_remainder'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

When I comment out `fmod` from the GPS library code, it successfully compile. 

Thanks.

nkolban, 2016-11-21:
It is likely that this issue is not directly related to the Arduino functions built on top of ESP-IDF but rather with ESP-IDF itself.  See:

https://github.com/espressif/esp-idf/issues/83

minyk, 2016-11-21:
@nkolban Thank you!

closed 2016-11-21 by minyk