espressif/arduino-esp32 #67: 'WiFiClient' does not name a type

zhivko, 2016-11-23:
Trying:
```
c:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\WiFi\examples\WiFiClientBasic\WiFiClientBasic.ino
```
I get:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\klemen\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=115200 -ide-version=10612 -build-path C:\Users\klemen\AppData\Local\Temp\arduino_build_304887 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\klemen\Documents\Arduino\wifi\wifi.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\klemen\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=115200 -ide-version=10612 -build-path C:\Users\klemen\AppData\Local\Temp\arduino_build_304887 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\klemen\Documents\Arduino\wifi\wifi.ino
Using board 'esp32' from platform in folder: C:\Program Files (x86)\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Program Files (x86)\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\variants\esp32" "C:\Users\klemen\AppData\Local\Temp\arduino_build_304887\sketch\wifi.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exceptions -fstrict-volatile-bitfields -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10612 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\variants\esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\WiFi\src" "C:\Users\klemen\AppData\Local\Temp\arduino_build_304887\sketch\wifi.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -c -w -Os -g3 -Wpointer-arith -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -fno-rtti -ffunction-sections -fdata-sections -mlongcalls -nostdlib  -std=gnu++11 -fno-exception [... cut ...]
      ^

wifi:30: error: 'void loop()' previously defined here

 void loop()

      ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Using library WiFi at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\WiFi 
exit status 1
'WiFiClient' does not name a type
```

Blink example works in my windows arduino esp32 installation though...

me-no-dev, 2016-11-24:
you have 2 sketches in the same folder
wifi.ino and WiFiClientBasic.ino

zhivko, 2016-11-24:
So each ino must be in its own folder?
This  should be mentioned in readme.md.
Does this limitation comes from arduino itselve?

me-no-dev, 2016-11-24:
should not be mentioned in the readme because it's Arduino thing. But having two compilable files that implement the same functions (loop and setup) will throw error with any builder.
Sketches are kept in folders with names the same as the sketches

zhivko, 2016-11-24:
OK, I put each ino in its own folder and trying WiFiScan example, I get this error in serial terminal and then wroom32 constantly reset itself - then I need to pull io0 to low so I can program it again.:
[image]

Sorry for not copy pasting error, but seems copy/paste in Serial monitor doesnt work...

me-no-dev, 2016-11-24:
not enough power to your ESP ;) maybe better cable?

zhivko, 2016-11-24:
Yeah - you are correct! I change power of my board from 3.3V pin of usb to ttl adapter to 5V (I have LD1117V33 on board) and example now works! Yes Thank you!

closed 2016-11-24 by zhivko

zhivko, 2016-11-25:
Just added txt so search engines could find this page....

```
ets Jun 8 2016 00:22:57 rst:0xl (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) ets Jun 8 2016 00:22:57
rst:0xl0 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0x00
clk_drv:0x00, q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:l
load:0x3ffc0008, len:0
load:0x3ffc0008,len:1388
load:0x40078000,len:3560
load:0x40080000,len:256
entry 0x40080034
rtc vll8 Oct 19 2016 15:22:11
XTAL 40M
tcpip_task_hdlxxx : 3ffc5a8c, prio:18,stack:2048 frc2_timer_task_hdl:3ffc7558, prio:22, stack:2048
```

lufebadeca, 2021-04-09:
I got the same error and it was only because I left a bracket open. Once I deleted the additional bracket, the issue solved immediately... just for the record