espressif/arduino-esp32 #69: I2C Problem - Clock Stretch Not Functioning

RodgerGr, 2016-11-25:
A problem was  encountered with the I2C Wire Library for ESP32

An ESP8266 program that sent a long I2C Write string to a slave device no longer worked when ported to ESP32.

An oscilloscope trace showed SCL was only going to mid level when the slave device was trying to pull it low. This seemed to be caused by both the ESP32 and slave device trying to drive the clock line at the same time.

[image]

Further investigation showed that Wire.begin called i2cAttachSCL() in esp32-hal-i2c.c and this used pinMode to set SCL as OUTPUT, rather than OUTPUT_OPEN_DRAIN.

I understand that the I2C standard calls for external pullup resistors on both SCL and SDA lines. For SCL, this is needed along with Open Drain outputs, so that the slave device can pull the SCL line low to temporarily suspend bus activity, known as "clock stretching".

To test the theory, I changed line 70 in esp32-hal-i2c.c from pinMode(scl, OUTPUT); to pinMode(scl, OUTPUT_OPEN_DRAIN); as shown below.
    
```
i2c_err_t i2cAttachSCL(i2c_t * i2c, int8_t scl)
{
    if(i2c == NULL){
        return I2C_ERROR_DEV;
    }
	pinMode(scl, OUTPUT_OPEN_DRAIN);
//	pinMode(scl, OUTPUT);
    pinMatrixOutAttach(scl, I2C_SCL_IDX(i2c->num), false, false);
    pinMatrixInAttach(scl, I2C_SCL_IDX(i2c->num), false);
    return I2C_ERROR_OK;
}
```
The code started working again and the ESP32 communicated correctly with the I2C slave device.

I am  a newbie so please forgive me if this should have been posted as a Pull Request.

me-no-dev, 2016-11-25:
no problem :) either way it's one line :) Thanks for the scope pictures! Those really help and explain many things. BTW What is the device you are connecting to?

me-no-dev, 2016-11-25:
done :)

closed 2016-11-25 by me-no-dev

RodgerGr, 2016-11-25:
Thank you kindly:)

RodgerGr, 2016-11-26:
Sorry, I am still getting used to this process:-(

The device I am communicating with is a Microchip MCP39F521 Analog Front-End, which requires that a rather long write string be sent to the device prior to reading the registers. The need for clock stretching may not be a problem with all devices, but anything that is slow enough to need a bit of time to think before responding may have faced the same issue.

p.s. I am seriously impressed by your lightning fast response to my issue and thank you for making the code changes in the master so promptly:-)