espressif/arduino-esp32 #70: WiFiClient fails after 3 loops

espuser, 2016-11-27:
Hi,
please insert a port disconnect into examples code

...
Serial.println("closing connection");
client.stop();
}

BR

me-no-dev, 2016-11-27:
Since ```client`` is a local variable for the function, at function end, the destruct method for client will be called, which will close the connection (call client.stop())

espuser, 2016-11-28:
Hi,
from point of view loop() you are right but the channel will be kept open
so after 3 loops (> MAXSOCKETS?) I get an connect fail in my case
BR

me-no-dev, 2016-11-28:
that is rather strange... it behaves fine when I try it here. What is the web server?

espuser, 2016-11-28:
Apache Server with simple Index.html "It works" content
connecting to 192.168.10.22
Requesting URL: /index.html....................?private_key=....................&value=2
HTTP/1.1 200 OK
Date: Mon, 28 Nov 2016 12:36:47 GMT
Server: Apache/2.4.10 (Win32) OpenSSL/1.0.1i mod_jk/1.2.40 PHP/5.5.15
Last-Modified: Sat, 30 Mar 2013 11:29:00T
....
closing connection

sorscode, 2016-11-29:
Post your code? I'll try and duplicate.

me-no-dev, 2016-11-29:
i see openSSL there, you are not connecting to HTTPS correct?

me-no-dev, 2016-11-29:
@espuser you know what? You might be correct! Destruct was not closing the client. Could you please try the latest commit?

espuser, 2016-11-29:
Hi,
the client runs without client.stop() now, but since You use the latest libs for WiFi from espressif I have the same problem as with the espressif/esp-idf toolchain, my esp32 NodeMCU cyclic reboots (about every 30 sec) or hangs
BR

me-no-dev, 2016-11-29:
There are lots of thing being worked on the network stack on IDF. Soon it will all be working :)

me-no-dev, 2016-12-01:
BTW pinMode time depends on which pin you apply it too. If it's an RTC pin, it's 2.6 us, if not it's 0.744us

closed 2016-12-07 by espuser