espressif/arduino-esp32 #71: How to use esp_wifi_internal_tx

Goldk, 2016-11-28:
Hey can I use esp_wifi_internal_tx to send my own packets (beacon for example)? How do I use it and what packets are supported?

igrr, 2016-11-28:
This function is in a header which is marked internal. So you can try calling that function, but then you are on your own.

closed 2016-11-28 by igrr

Goldk, 2016-11-29:
I would like to know which packets are supported and how to use it exectly. I can only see the header file, so I have no Idea what background actions taking place. That's why I'm asking.

(As it is written in the file:)
>  If someone really want to use specified APIs declared in here, please contact
>  espressif AE/developer to make sure you know the limitations or risk of 
>  the API, otherwise you may get unexpected behavior!!!

With the ESP8266 there was a function called wifi_send_pkt_freedom. With this you could send your own packets, but only unencrypted and no managment packets. 
And if you wanted to send beacon packets you had to change the buffer

> If you really want to send beacon, please start from "0x80, 0x00 ... " which means that the first 26 bytes need to be deleted. If you start from "0x00,0x00", wifi_send_pkt_freedom will detect the 80211 header and find out that it's a management packet, and send fail.

http://bbs.espressif.com/viewtopic.php?f=7&t=1357&p=10185&hilit=wifi_send_pkt_freedom#p10185

So... are here also limitations or something else to consider?

chrismerck, 2018-06-08:
@spacehuhn you've probably figured it out already, but for the record, you can see that `lwip`'s ESP32 port uses `low_level_output` in `components/lwip/port/netif/wlanif.c`, which in turn uses `esp_wifi_internal_tx`.

And, while the `esp_wifi_internal_tx` is undocumented (AFAICT), the `low_level_output` has a helpful comment:

```
/**
 * This function should do the actual transmission of the packet. The packet is
 * contained in the pbuf that is passed to the function. This pbuf
 * might be chained.
 *
 * @param netif the lwip network interface structure for this ethernetif
 * @param p the MAC packet to send (e.g. IP packet including MAC addresses and type)
 * @return ERR_OK if the packet could be sent
 *         an err_t value if the packet couldn't be sent
 *
 * @note Returning ERR_MEM here if a DMA queue of your MAC is full can lead to
 *       strange results. You might consider waiting for space in the DMA queue
 *       to become availale since the stack doesn't retry to send a packet
 *       dropped because of memory failure (except for the TCP timers).
 */
static err_t ESP_IRAM_ATTR
low_level_output(struct netif *netif, struct pbuf *p)
```

Generally, you can learn a lot reading the `lwip` implementation if you are interested in doing low-level things with wifi/ethernet on ESP32...