espressif/arduino-esp32 #72: Latest version draws too much energy - seems that WiFi is enabled by default

elFallino, 2016-11-29:
I updated today and now the esp32 fails to boot. It seems that the FTDI adapter can not provide enough energy.
I already had that problem before (with previous versions of this plugin) as I tried to run a sketch that uses WIFI without additional power source.

So I guess that the latest commits enable WiFi by default. In that case I think that WiFi should explicit be enabled by user code.

me-no-dev, 2016-11-29:
the stack is not started, just the radio. You should not power the esp from FTDI though. Get a proper LDO to 3.3V

elFallino, 2016-11-29:
I added a 3.3V buck converter, but radio was not started automagically in previous version, was it?
If so, I still think that it would be better to keep radio disabled until user code activates WiFi / BT.

igrr, 2016-11-29:
The difference from the previous version is in RF calibration process. Now it takes slightly longer and needs more current.

clowrey, 2016-12-01:
The latest version is also running at 240mhz vs 160mhz previously which means it draws more power overall too, I know that will be configurable in the future. 

Do you have plans/is it possible to have to RF calibration and all radio functions not start up until they are called/initialized from the IDF?

igrr, 2016-12-01:
There seems to be a regression in the ESP-IDF that wifi initialization only works correctly when it is called pretty early on. Someone (myself or one of my colleagues) will look into this next week. Once we fix this, WiFi initialization can be moved to a later stage, when e.g. WiFi.begin is called. @me-no-dev has already done all the work on Arduino side, we just need to figure out what goes wrong with WiFi init. Once WiFi init is moved from startup code, we can postpone RF init as well.

me-no-dev, 2016-12-01:
The force has been restored!
Do not include WiFi.h and phy will not start.

closed 2016-12-01 by me-no-dev

clowrey, 2016-12-01:
Now the dev board only draws 80ma at 240mhz from 150ma  with WiFi on :)

me-no-dev, 2016-12-01:
Oh this is really good! I can tell I need to optimize pinMode pulls to drop it under 1us :)
What I would really like to see in the table are some STM32 MCUs :) Other than that ESP32 seem to top them all :P

clowrey, 2016-12-01:
I'm guessing something changed in the pinMode function since earlier Nov as my previous test shows it 0.526 us even at the lower clock speed.. I was having trouble getting the 32Duino core working with some STM32 dev boards.. But I did just add a comparison to the Teensy 3.6 board with NXP K66 Cortex M4 that someone else ran these same tests on. The ESP32 compares pretty well against that chip but some areas are a little slower.

me-no-dev, 2016-12-01:
yes pinMode has changed because some pins have pull up/down settings elsewhere. I'll try to improve speed there, but some pins will always pinMode slower than others. If you pinMode pins 16 or 17 you will see a difference (closer to what you had before). I can probably run this against STM32F1 board

elFallino, 2016-12-01:
Thank you for solving this issue.
Where can I buy you a beer? :)

me-no-dev, 2016-12-01:
I live in Bulgaria. You coming by? :D

me-no-dev, 2016-12-02:
@clowrey here are my measurements with the latest commits (note I pinMode pin 16 which is not RTC pin) Added a couple STMs
[image]

clowrey, 2016-12-02:
Great! Can you share the google docs file so I can copy some of the results into my spreadsheet for comparison? It might be neat to create a timeline of improvements from current to future versions of the Arduino ESP32 core

me-no-dev, 2016-12-02:
it's not a google doc :) https://www.icloud.com/numbers/0wwqzBkq_vjYwtaYOmhyF8FqA#Untitled

sinelabore, 2017-02-20:
Hello ,

is there a way to keep WIFI switched off for EPS8266? I did not find any kind of documentation so far. I want to start WIFI in the Arduino sketch later on once needed.

Thanks,
Peter

me-no-dev, 2017-02-20:
ESP8266?

sinelabore, 2017-02-20:
Sorry yes, typo.

me-no-dev, 2017-02-20:
Why ask here :) this is the ESP32 repository. ESP8266 Arduino is [here](https://github.com/esp8266/Arduino)

pctj101, 2017-05-06:
Minor detail

@me-no-dev In regards to your Numbers spreadsheet, I believe the NRF52832 is 64Mhz.  (Not sure if your original clock speed was in reference to something else though...)

http://infocenter.nordicsemi.com/pdf/nRF52832_PS_v1.3.pdf
7.2.1 CPU performance
The CPU clock speed is 64 MHz.

me-no-dev, 2017-05-06:
@pctj101 It was mostly already made spreadsheet. I added some of the boards, but not the NRF :) do not have one

Ajioz, 2019-09-24:
I am having similar bug. This time I lost wifi connection and the router rebooted and got on track. My esp32 displayed correctly and also updated me through the serial monitor that I am connected, however, my esp itself just got stock without actually doing anything.  Below is the message print_out:

Connecting to MQTT... [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.0.142, MASK: 255.255.255.0, GW: 192.168.0.1