espressif/arduino-esp32 #74: ESP-WROOM32

thl-mot, 2016-12-02:
Hi,
I tried to run blink on an ESP-WROOM32. Download seams to work. RxD and TxD flashes while download. But no reaction from the ESP-Board. 
Another try with Serial: I expected a flashing TxD-LED but here as well: no reaction.
What is wrong? Where can I debug to get more information.
Thomas

lbernstone, 2016-12-03:
I have no idea if LED_BUILTIN is set to the correct pin right now.  me_no_dev is probably very busy trying to integrate the new IDF, so maybe try setting up a little breadboard with an led, resistor and your esp32, and set things up to write to a numbered pin instead of LED_BUILTIN. (https://www.arduino.cc/en/tutorial/blink)

clowrey, 2016-12-03:
There is no built in LED on the ESP-WROOM32 module itself but it is defined in the library you can easily find the pin by searching for LED_BUILTIN in this repository like so:
https://github.com/espressif/arduino-esp32/search?utf8=%E2%9C%93&q=LED_BUILTIN

thl-mot, 2016-12-03:
I checked several output pins with the LED, no result with the basic blink example.
Even a small Serial-Sender like the following one did not work:
```
void setup() {
    Serial.begin(9600)
}
void loop() {
    Serial.print("Hello");
  delay(1000);                      
}
```
I don't get any output at TxD.

Thomas

thl-mot, 2016-12-03:
I got it working, you have to press the button under the usb connector. So main problem solved, it is running in arduino-ide. Now I have to try this on sloeber-ide. This seams to be more work. Because the package-json does not yet contain all information to get it started.

me-no-dev, 2016-12-04:
no clue what package.json you refer to, but I have it running fine inside sloeber.

thl-mot, 2016-12-04:
with the package.json I ment the definition file for the package manager. As I included your esp32-package as custom hardware and libraries it works in sloeber-ide. I think the better way would be to include it by the package manager. At least its working now. There are some small things, because the ide shows errors even though its perfectly compiling.
The following lines of the WiFiScan example show errors:
Serial.print(WiFi.SSID(i));
Serial.print(WiFi.RSSI(i));
Do you have an idea where these errors come from and how I can fix it.
Thomas

me-no-dev, 2016-12-04:
not errors, eclipse tripping because it does not understand the methods correctly. Nothing is wrong with anything.
Package manager will come once we have some sort of stable Arduino. There are many many things still that need work.

thl-mot, 2016-12-05:
Do you have timeline/backlog/"list of things to do". Maybe I can help. Will you also include Bluetooth support or is this functionality outside the scope of Arduino?

me-no-dev, 2016-12-05:
Bluetooth will be included. Things are done as they come available in IDF, so there isn't a backlog really.

closed 2016-12-06 by thl-mot

zhivko, 2016-12-06:
Maybe a raletd topic to this one - the other way I tried GPIO34, and I was not able to make it work.
However https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide#arduino-example-blink ledpin=5 worked as it supposed.
@me-no-dev maybe you know what could be the reason GPIO34 does not work?

me-no-dev, 2016-12-06:
pins 34 and up can be only inputs

zhivko, 2016-12-06:
Oh... They should be then naped GPI and not GPIO ... Do you have any other idea how to visually represent this information on [https://github.com/espressif/arduino-esp32/blob/master/doc/esp32_pinmap.png].
I saw that on this link there are diferrent colors for each pin - can you put up a color legend with meaning?

me-no-dev, 2016-12-06:
you might find this useful: http://espressif.com/sites/default/files/documentation/esp32_chip_pin_list_en.pdf

aspiringguru, 2018-07-02:
link above was broken for me. this link worked for me > http://wiki.ai-thinker.com/_media/esp32/docs/esp32_chip_pin_list_en.pdf