espressif/arduino-esp32 #75: Sparkfun ESP32 Thing: Board gets hot just by including WiFi.h

chrgro, 2016-12-03:
I'm planning on contributing a bit to this repo, since I've just aquired a couple of these ESP32 based boards. If you can point me to some documentation then I'll help out as best I can. 

The following code keeps the ESP32 running cool (even if I add some simple led blinking in the loop).
`void setup() {
  // put your setup code here, to run once:
  pinMode(5, OUTPUT);
  digitalWrite(5, HIGH);
}

void loop() {
}`

While the exact same code just with included WiFi.h runs very hot! Why is this? 
`#include <WiFi.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(5, OUTPUT);
  digitalWrite(5, HIGH);
}

void loop() {
}`

clowrey, 2016-12-03:
Wifi core and radio uses more power.. 160+ ma

blackketter, 2016-12-03:
Seems to me that a side effect of including an extraneous .h file should not be turning on the radios.  

This should be deferred to wifi.begin(), no?

lbernstone, 2016-12-04:
I believe wifi currently has to be initialized at boot due to upstream issues. Devs are working on it.
https://github.com/espressif/arduino-esp32/commit/1d1aeecde26b9de7379641e9e9ed51eb6a0249b5

clowrey, 2016-12-04:
Wifi starting on boot issue was resolved as documented on this closed issue https://github.com/espressif/arduino-esp32/issues/72

But you must not include WiFi.h

chrgro, 2016-12-04:
I agree with @blackketter, wifi should not be enabled until you actually do something with it (for example WiFi.begin()). Probably not great to have sideeffects som including header files either. 

A common use case would be to enable the wifi, connect, send some data, then disconnect again. The WiFi.disconnect() function supports a wifioff switch, but it has no effect at the moment. 

By "upstream issues" you still mean issues in this arduino-esp32 repository? Or are there external dependencies?

me-no-dev, 2016-12-04:
By "upstream" he means issues in [esp-idf](https://github.com/espressif/esp-idf). Everything is built on top of that.
Currently the radio must be started on boot or it will not start at all, so I have added an overwrite-able method that get's triggered if WiFi.h is included. This will surely be fixed soon, but until then, we have to live with the current situation.

blackketter, 2016-12-04:
This might be a dumb questions, but is it currently possible to turn off
the radio after boot and then turn it on again?

me-no-dev, 2016-12-04:
no :)

me-no-dev, 2016-12-04:
For example if you call WiFi.mode(WIFI_OFF), it will turn off WiFi but the radio will stay on

jlcook83, 2016-12-18:
This all makes sense, but it sure does get hot!

santanapablo1975, 2017-02-16:
Hi All, any measurements for Bluetooth? I got also around 150mA with the same board even with BLE and Beacon... any advice around it?

me-no-dev, 2017-02-16:
how much do you get with just delay(1000); in the loop and nothing else (empty sketch)?

me-no-dev, 2017-02-16:
BTW, I got my Thing today and it's not hotter than it should be.
Rocking 2x240MHz 32bit cores ;) that uses power and generates heat. I think you are expecting miracles if you think that you can go really low on consumption while the cores are running at full speed and you have the radio on. BLE or Not. It's like sticking BT dongle instead of WiFi and expecting your laptop to last longer :) Well it will a bit because BT uses a bit less power, but at the same time you will get much less content through it also (YouTube ha-ha). The laptop analogy is a bit overkill but I think you get the idea. You can use less power overall if you wake up from time to time and check stuff, not even start the radio if not needed. Other than that, the computer is on and rocking (3 times more clock cycles than ESP8266 at 160 and 30 times more than Arduino UNO at 16MHz).

santanapablo1975, 2017-02-16:
you are right, the ESP32 is power full, I just was expecting in Bluetooth BLE mode less power consumption... on the other hand there is not yet any spec or data sheet with BT power consumption... but for wifi yes! maybe is as you said in any case the radio is on and it cover all the tx modes... in any case, thanks for the hints! regards

me-no-dev, 2017-02-16:
since they use the same radio... :)
Anyway, I just did a quick test. With just delay(1000) in the loop, my "Thing" got cold :) consumption dropped to under 70mA on the USB rail and that is on 240MHz. If I drop to 80 it will go even lower, but then the thing is not doing much, so in that case you really waste all of the power instead of crunching something useful :)

chrgro, 2017-02-17:
As I mentioned before, it is ok that it gets hot while the radio is running. 

What we are asking for is to be able to turn the radio on and off without reset. That way you can power up the wifi radio, send some data, then power down for a few hours, and then power back up again. That is something that should work on an IoT chip.

igrr, 2017-02-17:
Disabling WiFi/BT is being fixed now in the ESP-IDF. Once the change is merged, this can be implemented in Arduino.

me-no-dev, 2017-02-22:
ok so WiFi and BT can now be switched at will. See https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiBlueToothSwitch/WiFiBlueToothSwitch.ino for more info

closed 2017-03-10 by me-no-dev