espressif/arduino-esp32 #76: Arduino word function used in typedef

NevynUK, 2016-12-04:
The Arduino web site defines _word()_ as a function for casting bytes to a word.  

This port has word defined as _typedef unsigned int word_ on line 120 of Arduino.h.

This is breaking the Sparkfun TSL2561 libraray code.

Regards,
Mark

me-no-dev, 2016-12-05:
hmm... thanks for noticing this! I really wonder how that define got there, especially since the correct methods are implemented :)

closed 2016-12-05 by me-no-dev

hcs-svn, 2016-12-09:
removing "typedef unsigned int word;" breaks a lot of my code.
The avr core Arduino.h has both:

Line 121:
typedef unsigned int word;

and

Line 238:
uint16_t makeWord(uint16_t w);
uint16_t makeWord(byte h, byte l);
#define word(...) makeWord(__VA_ARGS__)

if you have both you can do:
word w = word(1, 2);