espressif/arduino-esp32 #77: get.py file

johankarem, 2016-12-04:
Hi everyone , I tried to install esp32 on arduino , after i type in the command promote : python get.py i get this errors :  
Platform: i686-mingw32
Traceback (most recent call last):
  File "get.py", line 122, in <module>
    mkdir_p(dist_dir)
  File "get.py", line 35, in mkdir_p
    os.makedirs(path)
  File "C:\Python27\Lib\os.py", line 157, in makedirs
    mkdir(name, mode)
WindowsError: [Error 5] Access is denied: 'dist/'

can you please help me ?

me-no-dev, 2016-12-04:
You do not have permissions to create a folder where get.py is. See to make sure you are the owner of the folder.

zhivko, 2016-12-05:
it could be you are using ordinary user instead of administrator user since
Arduino easy start in administrator folders.

johankarem, 2016-12-05:
@zhivko @me-no-dev  hello guys the problem solved but now im faceing new issue , when i verfiy the code in arduino : 
Sketch uses 150,319 bytes (14%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 20,760 bytes (7%) of dynamic memory, leaving 274,152 bytes for local variables. Maximum is 294,912 bytes.

i proceed to upload the code to my esp32 borad i got this message 

esptool.py v2.0-dev
Connecting...

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

what is the problem now ?

me-no-dev, 2016-12-05:
hold the boot button while trying to start upload. Your board (that you did not specify what it is) probably can not properly reset into download mode.

johankarem, 2016-12-05:
@me-no-dev  i pressed i got this message " 
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Compressed 5296 bytes to 3383...

Writing at 0x00001000... (100 %) 
Wrote 5296 bytes (3383 compressed) at 0x00001000 in 0.1 seconds (effective 730.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...

Writing at 0x00008000... (100 %) 
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 4915.1 kbit/s)...
Hash of data verified.
Compressed 251360 bytes to 102675...

Writing at 0x00010000... (14 %) 
Writing at 0x00014000... (28 %) 
Writing at 0x00018000... (42 %) 
Writing at 0x0001c000... (57 %) 
Writing at 0x00020000... (71 %) 
Writing at 0x00024000... (85 %) 
Writing at 0x00028000... (100 %) 
Wrote 251360 bytes (102675 compressed) at 0x00010000 in 2.5 seconds (effective 807.6 kbit/s)...
Hash of data verified.

Leaving...

is it mean , the code already uploaded ? but i did not get any output , i used blink example , this is my code : 

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(11, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
   Serial.println("Hello, world!");
  digitalWrite(11,HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(500);                       // wait for a second
  digitalWrite(11, LOW);    // turn the LED off by making the voltage LOW
  delay(500);                       // wait for a second
}

me-no-dev, 2016-12-05:
cmon man :D press the reset button and it will boot. And you are still not saying what is that board you are trying to upload to...

me-no-dev, 2016-12-05:
oh pin 11... :D good luck but this will not work. Pin 11 is connected to the onboard flash

johankarem, 2016-12-05:
oh man sorry xd , its wokring now , im using node mcu-32s , do u know any better development port ?? 
Thanks alot bro

me-no-dev, 2016-12-05:
no idea, but they all should be fine. I'm personally using Espressif modules, but I am not sure if you can get them from anywhere. I believe NodeMCU32S is based on that, but why it does not reboot I do not know.

closed 2016-12-05 by me-no-dev

johankarem, 2016-12-05:
yeah even me this is the first bored should i reboot manula