espressif/arduino-esp32 #78: Programming over the air

SirRibeiro, 2016-12-05:
Hello,
Is it possible to program the Esp32 wireless?

Or is there any idea when it will be possible?

Thanks

me-no-dev, 2016-12-05:
It will be possible yes, but it's not yet. Will be available when all requirements are fulfilled.

git-tiger, 2017-01-26:
for commercial product, will the firmware be secure? Encrypt bin file.

me-no-dev, 2017-02-08:
@git-tiger that would depend only on your implementation. ArduinoOTA would never make sense to be oversecured ;) it's meant for development, but there will be other options available (SSL server, so on) that would let you transfer updates securely post-production

arcao, 2017-02-13:
@git-tiger Both [flash encryption](http://esp-idf.readthedocs.io/en/latest/security/flash-encryption.html) and [secure boot](http://esp-idf.readthedocs.io/en/latest/security/secure-boot.html) are supported in [Espressif IDF for ESP32](https://github.com/espressif/esp-idf/). Arduino-ESP32 framework is built upon it,  so it will be somehow supported in future _(I guess)_, but it's too early to speak about it. There is lot of other work to do first.

me-no-dev, 2017-03-11:
implemented ;)

closed 2017-03-11 by me-no-dev

inforaudio, 2017-07-11:
Have you got ready ESP32 OTA in arduino ide??

mimansamaheshwari, 2018-08-17:
Is it possible to upload code multiple times via OTA on esp32 in Arduino Ide? If yes then how?