espressif/arduino-esp32 #79: SPI (or something associated with it) seems to be broken

elFallino, 2016-12-05:
Upgraded to the latest version (December, 12th) and ILI9341 display driver does not work. 
Using an older Version (November, 11th) it does work.

It is neither working using Hardware or Software SPI nor with enabled or disabled mutexes.

I will provide further information if wanted.

me-no-dev, 2016-12-05:
SPI driver has not changed and it's it's own separate thing that has nothing to do with IDF. It's quite impossible for it not to be working. And when you say that SoftSPI also does not work, that would mean that digitalWrite/digitalRead does not work and that could not be the case too.
Try and look for something else to be the problem.

elFallino, 2016-12-06:
I worked myself through the timeline of commits.
The last version that is working for me was the committ made on November, 27th.

The next version that I tested is the committ made on December, 1th (the ones between those two dates could not be tested because of the high energy draw - I have no access to a proper power source atm, so they do not boot).

I am going to try to investigate the commits made within this time span to find out more.

Maybe you have additional ideas.

elFallino, 2016-12-06:
Found out that it is caused by committ "latest IDF, 240MHz and BLE enabled libs" 
( https://github.com/espressif/arduino-esp32/commit/b82d0e166aa364ef246ef83bf6114008c09dfe8a )

When I replace the new IDF libs with those from November, 27th (or earlier) the SPI display is working again.

me-no-dev, 2016-12-06:
try adding
```cpp
#define CONFIG_DISABLE_HAL_LOCKS 1
```
to [this](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h) file and see if that changes anything

me-no-dev, 2016-12-06:
btw I'm running the latest IDF here with SPI screen attached and has been working the whole time, so it's really strange to figure out what is going on on your end.

elFallino, 2016-12-06:
I already tried that but downloaded the latest version of arduino-esp, again - just to make sure.
It still does not work.

Could it be a timing issue caused by the change from 160 to 240MHz? I already tried to change anything that looked like a 240 back to 160 in boards.txt and sdkconfig.h without luck.

I am using
- Arduino 1.6.12
- a TFT with ILI9341 controller via Hardware SPI (4-Wire)
- Adafruit's library
- an ESP32 (Wroom-32)

elFallino, 2016-12-09:
Using the latest esp32-arduino (including latest IDF) the display is working again.
Great - so far :)

Unfortunately, it is not working with disabled HAL locks.

me-no-dev, 2016-12-09:
if it fails with hal locks off, that prolly means that you are trying to write to SPI from different threads/interrupts? is that the case?

elFallino, 2016-12-09:
No, there is no threading at all. Also there is no exception or something displayed over serial.
Same code worked with older version of esp32-arduino (see above).

me-no-dev, 2016-12-09:
please send me your exact sketch and I'll run it here. I have ili9341 handy

elFallino, 2016-12-09:
You can simply use Adafruits ILI9341 library and test code.
https://github.com/adafruit/Adafruit_ILI9341

I connected the display that way:
```
#define TFT_DC 27
#define TFT_CS 32
#define TFT_RST 33
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_MISO 12
```

And calling the constructor this way:
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);`

It is working  (but slow)  when HAL locks are NOT disabled and is not working when they are disabled.
Same code works with older versions and disabled HAL locks (very fast, then) like mentioned above.

Blagodarya!

elFallino, 2016-12-11:
Seems that the problem is caused by my own changes within the ILI9341 library. But I do not see the reason (but that is nothing that you need to care about).

I altered Adafruits library (cpp file, line 95) and replaced `SPI.transfer(byte); ` with `SPI.write(byte); `  (and forgot to mention it) because _write()_ just writes the byte while _transfer()_ is writing the byte and then reads it to return it to caller.

So beside of the reading, both do the same but the latter is nearly twice as slow.
However, it is working with transfer(); but not with write();

So I guess this is just a timing problem that was not there before.

Thanks a lot, anyway.

me-no-dev, 2016-12-13:
this could be a bug also ;) i'll have a look

me-no-dev, 2016-12-13:
I look at the code and it looks totally fine. SPI does not write until the previous command has finished. Could I bother you to test something since you have reproducible fail case?

elFallino, 2016-12-13:
Sure. What shall I do?

me-no-dev, 2016-12-14:
add after the while loop [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L490) the following line:
```cpp
void old = spi->dev->data_buf[0];
```
and then try with just write :)

elFallino, 2016-12-14:
As already talked about:

- `void old = spi->dev->data_buf[0];` won't do it because the compiler complains about declaring a variable as void

- `uint8_t old` is working but causes a panic quickly
- putting `uint8_t old = spi->dev->data_buf[0];` at the end of that method right before HAL_MUTEX_UNLOCK(); seems to work - but I had some panics (different than the ones above) at the beginning but as I can not reproduce them they might be caused by something else within my code.

Testing this change with an dedicated testing sketch does work fine and fast using hardware SPI at up to 40MHz (if setting this value actualy is working) without issues.

Thanks a lot so far. I would be glad if I can help testing more.

me-no-dev, 2017-02-08:
closing this @elFallino ;)

closed 2017-02-08 by me-no-dev