espressif/arduino-esp32 #80: python certification error on Windows 10?

kriswiner, 2016-12-06:
I am trying to install the esp compile tools for ESP32 and I am encountering what looks like a certificate error on Windows 10. I was able to do this successfully on Windows XP but I have heard Windows 10 has "harder" certificate management. Anyone kjnow how to get around this? Here is the error log:

```
C:\Users\kris\Documents\Arduino\hardware\espressif\esp32\tools>python get.py
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-59.zip
Traceback (most recent call last):
  File "get.py", line 124, in <module>
    get_tool(tool)
  File "get.py", line 83, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "C:\Python27\lib\urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "C:\Python27\lib\urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "C:\Python27\lib\urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "C:\Python27\lib\urllib.py", line 443, in open_https
    h.endheaders(data)
  File "C:\Python27\lib\httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "C:\Python27\lib\httplib.py", line 897, in _send_output
    self.send(msg)
  File "C:\Python27\lib\httplib.py", line 859, in send
    self.connect()
  File "C:\Python27\lib\httplib.py", line 1278, in connect
    server_hostname=server_hostname)
  File "C:\Python27\lib\ssl.py", line 353, in wrap_socket
    _context=self)
  File "C:\Python27\lib\ssl.py", line 601, in __init__
    self.do_handshake()
  File "C:\Python27\lib\ssl.py", line 830, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)
```

C:\Users\kris\Documents\Arduino\hardware\espressif\esp32\tools>

me-no-dev, 2016-12-06:
I followed [this](https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md) instructions and had no such problem

kriswiner, 2016-12-07:
I followed the instructionas and I am still having the same error.

me-no-dev, 2016-12-07:
And how is that happening then? I spent time yesterday to fresh install 32 and 64 bit Windows 10 and run the instructions I linked and all went exactly as it should. So how come that is not the case for you?
Maybe uninstall Python and start fresh?

me-no-dev, 2016-12-07:
Maybe you installed 32bit Python on 64 bit machine?

kriswiner, 2016-12-07:
OK, I uninstalled everything and tried from scratch, still the same problem. However, I just downloaded the  xtensa-esp32-elf-win32-1.22.0-59.zip from this site and got it to work. Not sure why python didn't but no worries now.

closed 2016-12-07 by kriswiner