espressif/arduino-esp32 #82: Conflict with WiFi.h and SD.h

copercini, 2016-12-06:
If just include WiFi or SD alone they works fine, but if include both:

`#include <WiFi.h>
#include <SPI.h>
#include <SD.h>`

it gives the errors:

`
In file included from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/fcntl.h:3:0,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/lwip/lwipopts.h:38,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/lwip/lwip/opt.h:45,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/lwip/lwip/ip_addr.h:35,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:47,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/esp32/esp_event.h:23,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/esp32/esp_event_loop.h:22,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.h:28,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.h:28,

                 from C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\WiFi\src/WiFi.h:31,

                 from C:\Users\user\Desktop\ESP32\esp32_teste_\esp32_teste_.ino\esp32_teste_.ino.ino:17:

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:34:18: error: expected unqualified-id before numeric constant

 #define O_RDONLY 0  /* +1 == FREAD */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:55:15: note: in expansion of macro 'O_RDONLY'

 uint8_t const O_RDONLY = O_READ;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:35:18: error: expected unqualified-id before numeric constant

 #define O_WRONLY 1  /* +1 == FWRITE */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:59:15: note: in expansion of macro 'O_WRONLY'

 uint8_t const O_WRONLY = O_WRITE;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:36:17: error: expected unqualified-id before numeric constant

 #define O_RDWR  2  /* +1 == FREAD|FWRITE */

                 ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:61:15: note: in expansion of macro 'O_RDWR'

 uint8_t const O_RDWR = (O_READ | O_WRITE);

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:34:18: error: expected unqualified-id before numeric constant

 #define O_RDONLY 0  /* +1 == FREAD */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:27:20: note: in expansion of macro 'O_RDONLY'

 #define O_ACCMODE (O_RDONLY|O_WRONLY|O_RDWR)

                    ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:63:15: note: in expansion of macro 'O_ACCMODE'

 uint8_t const O_ACCMODE = (O_READ | O_WRITE);

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:34:18: error: expected ')' before numeric constant

 #define O_RDONLY 0  /* +1 == FREAD */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:27:20: note: in expansion of macro 'O_RDONLY'

 #define O_ACCMODE (O_RDONLY|O_WRONLY|O_RDWR)

                    ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:63:15: note: in expansion of macro 'O_ACCMODE'

 uint8_t const O_ACCMODE = (O_READ | O_WRITE);

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:12:18: error: expected unqualified-id before numeric constant

 #define _FAPPEND 0x0008 /* append (writes guaranteed at the end) */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:37:18: note: in expansion of macro '_FAPPEND'

 #define O_APPEND _FAPPEND

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:65:15: note: in expansion of macro 'O_APPEND'

 uint8_t const O_APPEND = 0X04;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:22:17: error: expected unqualified-id before numeric constant

 #define _FSYNC  0x2000 /* do all writes synchronously */

                 ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:41:17: note: in expansion of macro '_FSYNC'

 #define O_SYNC  _FSYNC

                 ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:67:15: note: in expansion of macro 'O_SYNC'

 uint8_t const O_SYNC = 0X08;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:18:18: error: expected unqualified-id before numeric constant

 #define _FCREAT  0x0200 /* open with file create */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:38:18: note: in expansion of macro '_FCREAT'

 #define O_CREAT  _FCREAT

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:69:15: note: in expansion of macro 'O_CREAT'

 uint8_t const O_CREAT = 0X10;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:20:17: error: expected unqualified-id before numeric constant

 #define _FEXCL  0x0800 /* error on open if file exists */

                 ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:40:17: note: in expansion of macro '_FEXCL'

 #define O_EXCL  _FEXCL

                 ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:71:15: note: in expansion of macro 'O_EXCL'

 uint8_t const O_EXCL = 0X20;

               ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:19:18: error: expected unqualified-id before numeric constant

 #define _FTRUNC  0x0400 /* open with truncation */

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32/tools/sdk/include/newlib/sys/_default_fcntl.h:39:18: note: in expansion of macro '_FTRUNC'

 #define O_TRUNC  _FTRUNC

                  ^

C:\Users\user\Desktop\arduino-1.6.12\hardware\espressif\esp32\libraries\SD\src/utility/SdFat.h:73:15: note: in expansion of macro 'O_TRUNC'

 uint8_t const O_TRUNC = 0X40;

               ^
`

Note: I am using adafruit SD library for ESP32:
https://github.com/adafruit/arduino-esp32/tree/master/libraries/SD

me-no-dev, 2016-12-06:
Sorry, but we are not supporting that library :) Those O_ things are POSIX flags and will not be removed from IDF. Maybe check if the regular flags will work and remove the definitions in SD library.

copercini, 2016-12-07:
Just renaming the flags in SD library it seems OK now.

closed 2016-12-07 by copercini