espressif/arduino-esp32 #83: Wifi Examples Building, Hanging

forthlightning, 2016-12-06:
I am trying to run the wifi examples, they build and upload fine but the program hangs after initial boot. I've tried WiFiClient, WiFiClientBasic, WiFiEvents, and WiFiScan and they all result in similar output as below:

```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1964
load:0x40078000,len:3668
load:0x40080000,len:260
entry 0x40080034
I (571) heap_alloc_caps: Initializing heap allocator:
I (571) heap_alloc_caps: Region 19: 3FFC1268 len 0001ED98 tag 0
I (573) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (583) cpu_start: Pro cpu up.
I (588) cpu_start: Single core mode
I (595) cpu_start: Pro cpu start user code
I (619) rtc: rtc v160 Nov 22 2016 19:00:05
I (630) rtc: XTAL 40M
W (706) cpu_start: failed to load RF calibration data, falling back to full calibration
I (2149) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 2
I (2304) cpu_start: Starting scheduler on PRO CPU.
tcpip_task_hdlxxx : 3ffc4ccc, prio:18,stack:2048
I (2312) wifi: frc2_timer_task_hdl:3ffc67dc, prio:22, stack:2048
I (2321) wifi: pp_task_hdl : 3ffc904c, prio:23, stack:8192
```

This may be unrelated but I have also been seeing this line recently

`
W (706) cpu_start: failed to load RF calibration data, falling back to full calibration
`

For reference I have freshly pulled esp-idf and arduino-esp32

me-no-dev, 2016-12-06:
So are you using esp32-arduino as idf component? Or are you using it through Arduino IDE?
If you are using it as component and have disabled arduino autostart, then you need to call initArduino(); in app_main

me-no-dev, 2016-12-06:
That will initialize the hardware, but not make you run through setup/loop.

forthlightning, 2016-12-06:
I followed your instructions here https://github.com/espressif/arduino-esp32/issues/13#issuecomment-264027363 to run it as an idf component. 

I've been using the esp-idf in part to get around the necessity of using setup and loop, so I will try an example using initArduino();

Are the "run arduino as idf component" and "use initArduino(); to start" in the documentation anywhere? I imagine that many people are using Arduino with esp32 simply for the library compatibility!

forthlightning, 2016-12-06:
To clarify: I set autostart to true previously.

If I disable autostart, can I omit setup and loop completely?

forthlightning, 2016-12-07:
I did some digging and it looks like when you set `config_autostart_arduino=N` in `make menuconfig` it is overwritten and changed to `#config_autostart_arduino is not set`

So when esp32-hal.h is included config_autostart_arduino is not defined, and is defined as 1.

I think this is causing the default app_main() in the ESP32 core to be defined instead of the users.

me-no-dev, 2016-12-07:
Oh you are so correct! What was I thinking when I wrote that :D Pull and try now

me-no-dev, 2016-12-07:
so if you do not autostart arduino, you need to call  initArduino(); in app_main to start WiFi/BLE or anything else that might be needed by the Arduino framework to function, except to run the setup/loop routine. So no thread is started for Arduino and you can do whatever you want, while using all Arduino API to access peripherals and so on.

forthlightning, 2016-12-07:
Ok so I pulled esp-idf, and cloned arduino into component folder in project folder.

Now I'm getting these messages when trying to build:

```
CXX main.o
AR libmain.a
LD arduino_as_component.elf
/Users/maxfieldmorrison/esp/arduino_as_component/build/esp32/libesp32.a(cpu_start.o):(.literal.main_task+0x0): undefined reference to 'app_main'
/Users/maxfieldmorrison/esp/arduino_as_component/build/esp32/libesp32.a(cpu_start.o): In function 'main_task':
/Users/maxfieldmorrison/esp/esp-idf/components/esp32/./cpu_start.c:235: undefined reference to 'app_main'
collect2: error: ld returned 1 exit status
make: *** [/Users/maxfieldmorrison/esp/arduino_as_component/build/arduino_as_component.elf] Error 1
```

I am not sure if this is between the component and my example or the esp-idf and my example.

My directory looks like this:
[image]

and my code looks like this:

```
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include "Arduino.h"

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"

void hello_task(void *pvParameter)
{
    printf("Hello world!\n");
    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

void app_main()
{
    initArduino();
    nvs_flash_init();
    xTaskCreate(&hello_task, "hello_task", 2048, NULL, 5, NULL);
}
```

Not sure what's going on here, thanks much for your help so far!

me-no-dev, 2016-12-07:
this is probably cpp file so you need to add ```extern "C"``` before ```void app_main()```

bengchet, 2016-12-08:
@forthlightning A few days before, I was doing the same thing as you, you can git clone this link [https://github.com/bengchet/ESP32-FreeRTOS-with-Arduino](url) as starting project template. I was using Eclipse IDE but it is not necessary. You might need to run make menuconfig before you start building it. The esp-idf setup I was simply following the setup guide by Espressif.

forthlightning, 2016-12-08:
@me-no-dev Yes! That was the problem. Thanks so much. 

Please consider adding some documentation about working with the arduino core as a component, I think many others will share my desire for the library compatibility of arduino and the flexibility of the esp-idf and freeRTOS.

@bengchet thanks for the tip! this looks super useful, but I think I'll stick with the official repo for now.

me-no-dev, 2016-12-08:
Great news :) I will add some info about using it as IDF component also.

closed 2016-12-08 by me-no-dev