espressif/arduino-esp32 #85: WifiClient.available() fails with errno 88

randomswdev, 2016-12-08:
I'm trying to connect to an MQTT broker via WiFi. I'm able to execute the connection (I see it logged on the server), but when my client tries to read the answer from the server it fails, with the following line:

[E][WiFiClient.cpp:201] available(): 88

(the line number is different than the one in the current source code, because of some additional traces I added).

If I understand correctly, errno 88 means NOTIMPLEMENTED.

me-no-dev, 2016-12-08:
Are you using it through Arduino IDE or through IDF as component?

randomswdev, 2016-12-08:
Trough the Arduino IDE

me-no-dev, 2016-12-08:
Maybe regression in IDF... I'll have a look. I'm working on new faster client anyway :)

randomswdev, 2016-12-08:
Do you have an outlook for the release date?

me-no-dev, 2016-12-08:
I have a few drivers to write first and there are things that need fixing in IDF before networking is stable anyway :)

me-no-dev, 2016-12-08:
interesting... I do not see any change that would break it and nobody else have reported it. Are you sure you are connected to the remote side?

ccfoo242, 2016-12-09:
I was just about to ask about this same error. Tried two different examples, one here https://developer.ibm.com/recipes/tutorials/connect-an-esp32-to-the-watson-iot-platform/ and one here https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide (changed the include to use just wifi.h

Similar error:
Reconnecting client to quickstart.messaging.internetofthings.ibmcloud.com
[E][WiFiClient.cpp:196] available(): 88

and
connecting to wifitest.adafruit.com
Requesting URL: /testwifi/index.html
[E][WiFiClient.cpp:196] available(): 88

Any help would be appreciated.

sorscode, 2016-12-09:
I run MQTT on mine. Let me download the latest and reflash.

sorscode, 2016-12-09:
Okay, so I can confirm, I'm seeing the same thing now on the latest.

[E][WiFiClient.cpp:196] available(): 88

I have been running mine using MQTT for a few weeks, with no issues. I'll do so digging as well.

sorscode, 2016-12-09:
Okay, so I did some playing.

I was running on this commit:
commit 58ac6fbf46f4657945f980c53c3d9bcaabfb50cd

Going to the latest it breaks. So I reverted back to the commit, that worked, and slowly moved forwards through the commits. The following commit, is the one that breaks it:
commit ee36407b43227674ba99039ea3a472e6a3b05f6a

So I now have moved mine to this commit (1 prior to the one above that breaks it):
commit e77ec634a9d1b54f410eb6a46150874e6dc5c464

I'm going to take a look around, and see if anything sticks out.

sorscode, 2016-12-09:
I haven't found anything yet. However, I do have a question. Is there a way to get more advanced/detailed debug output.

SirRibeiro, 2016-12-12:
Is there any solution to this problem?

If we use the commit 58ac6fb it is possible to read the response from the server, but at the end of some connections, it is no longer possible to communicate getting:

[E][WiFiClient.cpp:62] connect(): socket: 23

sorscode, 2016-12-12:
I was running on 58ac6fb since release, with no issues. I'm now on e77ec63, and running strong. I am using dedicated power, versus USB.

me-no-dev, 2016-12-13:
The one major change that I see in the commit that you report as breaking things is the addition of "ethernet". Will ask around

forthlightning, 2016-12-13:
I was also seeing this last night. 
`[E][WiFiClient.cpp:201] available(): 88`

Running Arduino as a component, trying to connect to MQTT server via WiFi. I originally was using PubSubClient, and was going to change to mongoose today, but if It's a problem with Client, sounds like I should busy myself elsewhere.

vshymanskyy, 2016-12-16:
I have the same problem

suddenlyimaged, 2016-12-17:
I'm having same trouble it is connecting but it's not, I can see the esp32 on a wifi sniffer and arduino serial monitor shows it outputting.

jlcook83, 2016-12-18:
Same issue here
[E][WiFiClient.cpp:196] available(): 88
I am using the Arduino IDE Wifi example modified to connect to data.sparkfun.com.

I will try manually merging the aforementioned commits onto my machine, that were reported 'working' - as a fix for now..

sorscode, 2016-12-18:
So I've been focused on this issue this weekend. The issue is in the TCP stack. I'm narrowing down where. The error, is on initial TCP connection to the MQTT server, and ever 15 seconds (Keep Alive timer). I did some deep packet, packet captures late last night. Will review today, and see what info I get from that, then go back into looking through the code. 

I'm also going to try any reproduce, by trying something other than MQTT also.

I also have noticed, that on the last code that does work, it seems if you are doing too much communication over MQTT, it will lock the ESP32 up.

sorscode, 2016-12-18:
Just an update. I can recreate this on just a standard pull of data from a website. I used SparkFuns test example, for their ESP32 Thing device. It's when the data comes back to the ESP32, is when it throws the error 88. What I've noticed inside ESP32, is Error 88 means, Function not implemented. 

I'm hoping to get to spend a little more time today on this.

jlcook83, 2016-12-18:
@sorscode I appreciate you deep-diving into this, as I am sure others will as well.  It certainty appears to be a problem even at the simplest implementations (such as the sparkfun's example) as you have mentioned.

*Even though I consider myself a seasoned c /c++ /etc troubleshooter, I am new to Espressif's line, and to the ESP-IDF, and is taking me some time to navigate through all the source - this gives me an appreciation for the progress you have made already!

I did perform a very simple test, by ignoring the error in WiFiClient.available(), and forcing it to still return a 'count' (it of course, is returning an invalid 'count') - to see if I could parse out a response from the server anyhow, and yes  the server response is available (see below).  Which, I guess means there is something going wrong much deeper past " int res = ioctl(sockfd, FIONREAD, &count);" .. this you already knew, I know, but just thought I'd mention the fact that parsing a response is *somewhat* possible - despite the garbage the error is also giving us..

```
int WiFiClient::available()
{
    if(!_connected) {
        return 0;
    }
    int count;
    int res = ioctl(sockfd, FIONREAD, &count);
    if(res < 0) {
        log_e("%d", errno);
       // stop();           <- Temporarily Ignoring
       // return 0;         <- Temporarily Ignoring
    }
    return count;
}
```

WiFi connected
IP address: 
192.168.1.110
connecting to data.sparkfun.com
Requesting URL: /input/DJlxjN3VQmUxVY5m8Xqm?private_key=[MY PRIVATE KEY HERE]&data0=1

> [E][WiFiClient.cpp:196] available(): 88  (+762 more occurrences..)

HTTP/1.1 200 OK
> [E][WiFiClient.cpp:196] available(): 88  (+32 more occurrences..)

Access-Control-Allow-Origin: *
> [E][WiFiClient.cpp:196] available(): 88 (+47 more occurrences..)

Access-Control-Allow-Methods: GET,POST,DELETE
> [E][WiFiClient.cpp:196] available(): 88 (+67 more occurrences..)

Access-Control-Allow-Headers: X-Requested-With, Phant-Private-Key
> [E][WiFiClient.cpp:196] available(): 88 (+26 more occurrences..)

Content-Type: text/plain
> [E][WiFiClient.cpp:196] available(): 88 (+25 more occurrences..)

X-Rate-Limit-Limit: 300
> [E][WiFiClient.cpp:196] available(): 88 (+29 more occurrences..)

X-Rate-Limit-Remaining: 286
> [E][WiFiClient.cpp:196] available(): 88 (+36 more occurrences..)

X-Rate-Limit-Reset: 1482094059.683
> [E][WiFiClient.cpp:196] available(): 88 (+37 more occurrences..)

Date: Sun, 18 Dec 2016 20:43:46 GMT
> [E][WiFiClient.cpp:196] available(): 88 (+19 more occurrences..)

Connection: close
> [E][WiFiClient.cpp:196] available(): 88 (+28 more occurrences..)

Transfer-Encoding: chunked
> [E][WiFiClient.cpp:196] available(): 88 (+70 more occurrences..)

Set-Cookie: SERVERID=; Expires=Thu, 01-Jan-1970 00:00:01 GMT; path=/
> [E][WiFiClient.cpp:196] available(): 88 (+24 more occurrences..)

Cache-control: private
> [E][WiFiClient.cpp:196] available(): 88 (+7more occurrences..)

a
> [E][WiFiClient.cpp:196] available(): 88 (+12more occurrences..)

**1 success**
> [E][WiFiClient.cpp:196] available(): 88 (+3 more occurrences..)

0
> [E][WiFiClient.cpp:196] available(): 88 (+ infinite more...)

sorscode, 2016-12-19:
Okay so lots of progress today. I'm now able to make commit 48abb79 work. The issue doesn't seem to be code related at all, but an actual library. I haven't attempted yet to determine which one, but I might take a swing at that, after I post this.

So this is how I got it to work, after going through the entire Arduino Core.

Do the following: (Sorry in advance I'm a Mac user, so you will have to do the equivalent in other OS's.)
Go to the your esp32 folder, where you cloned too
git reset --hard e77ec63
copy the content of "tools/sdk/lib" to another folder (desktop is what I did)
return back to the root of your esp32 folder
git reset --hard 48abb79
copy the files that you just previously saved, back into "tools/sdk/lib". Don't remove the files that were there, just replace (count is different). 

That is it. It will compile, and TCP functionality is restored. Mine has been running now for about 2 hours, and I've been sending it random MQTT commands, to do stupid stuff (ex. turn leds on/off, PWM on LEDs, and turn on relays). 

Now time to see if I can determine which file(s), is causing the problem.

jlcook83, 2016-12-19:
unfortunately for me I am not having success with commit "48abb79" proceeding after wifi connect (never proceeds to getting an ip address).. just spits out some 'jibberish' followed by a TG0WDT_SYS_RESET.  Another attempt spits out a bit more legible results, but just hangs there with '0  0  0  0  0  0' being the last bit of output.. (again, never obtains an IP).  

I do get a successful IP assignment with later commits, but of course the method that @sorscode has provided above has some incompatibilities with those later commits, so for now I am left trying to understand why I cannot get an IP assignment with commit "48abb79" (with or without the inserted /tools/sdk/lib from commit "e77ec63".... 

```
I (2978) wifi: connected with Messier, channel 9
..........   0ÿÿÿß€@   0   0   0   0   0ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

```
I (4566) wifi: connected with Messier, channel 9
........  0  0  0  0  0  0
```

MrSinan, 2016-12-19:
İ confirm that the ersion 58ac6fb is running without problems on sparkfun ESP32 dev board. What is the big difference between this and the actual version???

sorscode, 2016-12-20:
MrSinan - what version of Arduino IDE are you running, and did you do this with or without my mod?

jlcook83, 2016-12-20:
58ac6fbf46f4657945f980c53c3d9bcaabfb50cd is working here too, not sure how I overlooked it, I'm certain I tried it at least once..  
sorscode - I am having success without the mod...

sorscode, 2016-12-20:
Interesting. 

I'm going to blow my esp32 folder away, and try again, without modification. Then start working up the commit level again.

sorscode, 2016-12-20:
Okay, So I messed with this before I leave for work. I had to only apply the fix for only one of the commits, after that everything works fine up to e77ec63 (December 2nd, 2016). After that all produce the 88 error. If you are getting a CPU Panic or Halt, make sure you are not running off USB Power. I'm using a stable 5v Power Supply with 2A of available current to this board (ESP32 Dev). 

So everyone is aware, I pulled down the IDF late last night, and going to work with that, to find this issue. I will focus on it tonight for a couple hours, when I get back from work.

MrSinan, 2016-12-24:
@sorscode: i use the last version arduino ide.

sorscode, 2016-12-24:
So I wanted to give an update. @me-no-dev and I have messed with this, and think a solution has been found. I'm running in the latest code, and working with no issues. I'm going to continue to test, and keep @me-no-dev updated, and hopefully have the fix in his next release.

palomarcoder, 2016-12-27:
Appreciate all your work on this - how is it going?

palomarcoder, 2016-12-28:
If anyone would like help testing, let me know.  I'm using the Arduino 1.8.0 IDE with the espressif esp32 core.  I'm using wifi.begin, connect, status, localIP and also PubSubClient subscribe, publish, and callback.

I can also test udp begin, read, write functions.

me-no-dev, 2016-12-29:
currently working on the build process (something broke) and I'll test quickly the server/udp libs I have prepped. If they work, I'll merge them.

me-no-dev, 2016-12-30:
Updates are pushed, client is fixed, Server and UDP are added. I'm finally closing this

closed 2016-12-30 by me-no-dev

palomarcoder, 2016-12-30:
I can confirm that it is working with the mqtt based IBM IoT Platform.  Thanks!

jlcook83, 2016-12-30:
Works great!  Thanks for your efforts on this one @me-no-dev @sorscode

suddenlyimaged, 2016-12-31:
Working great on my Widora/eBox esp32 and running Blynk! Great work guys!

carldunham, 2017-01-27:
Still seeing this in Platform.io. Does that need an update?