espressif/arduino-esp32 #87: Analog Attenuation API?

torntrousers, 2016-12-10:
Thanks for the recently committed analog functions. 
Is there a problem with analog attenuation or some reason why its not exposed as an API to Arduino sketches? Or is it coming soon perhaps?

me-no-dev, 2016-12-10:
Tried it and it did not work :)

torntrousers, 2016-12-10:
Heh,  yes me too fiddling about in esp32-hal-adc and it didn't work for me either. Could I raise this as a bug with Espressif somewhere - (I don't properly understand how all the components fit together ) - is it  hardware thing or the sdk or idf ...?

me-no-dev, 2016-12-10:
it's either hardware or we are missing some setting in the software that I do not know about. You can raise an issue in esp-idf as lower level guys visit that repo :)

torntrousers, 2016-12-13:
Over [here](http://www.esp32.com/viewtopic.php?f=13&t=693&p=2940) it was suggested to use adc1_config_channel_atten, thats seems to work ok. 

Its in #include <driver/adc.h> and can do:
adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_11db);
adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_6db);
adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_2_5db);
adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_0db);

That seems to give a max input voltage of:
0db   = 1.1v
2.5db = 1.4v
6db   = 1.9v
11db  = 3.2v

me-no-dev, 2016-12-13:
no wonder... I was writing to the wrong register

me-no-dev, 2016-12-14:
please try the latest commit
```cpp
    analogSetAttenuation(ADC_0db);
    analogSetPinAttenuation(A18, ADC_0db);
    analogSetPinAttenuation(A19, ADC_2_5db);
    analogSetPinAttenuation(A6, ADC_6db);
    analogSetPinAttenuation(A7, ADC_11db);
```

closed 2016-12-30 by me-no-dev