espressif/arduino-esp32 #90: I2C still not working

kriswiner, 2016-12-11:
I can read the I2C addresses of the I2C sensors with pins 21/22 but I cannot read the WHO_AM_I byte.
This works on the Teensy and STM32L4:

`  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, 1);            // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}`

me-no-dev, 2016-12-12:
what is that slave device that you are trying to access?

kriswiner, 2016-12-12:
MPU9250 and MS5637.

kriswiner, 2016-12-13:
OK, I confirmed I can read the MS5637, which essentially uses one byte at a time reads and writes. Here is the sketch that works just fine:

```
#include "Wire.h"   

// See MS5637-02BA03 Low Voltage Barometric Pressure Sensor Data Sheet
#define MS5637_RESET      0x1E
#define MS5637_CONVERT_D1 0x40
#define MS5637_CONVERT_D2 0x50
#define MS5637_ADC_READ   0x00

#define MS5637_ADDRESS 0x76   // Address of altimeter

#define SerialDebug true  // set to true to get Serial output for debugging

#define ADC_256  0x00 // define pressure and temperature conversion rates
#define ADC_512  0x02
#define ADC_1024 0x04
#define ADC_2048 0x06
#define ADC_4096 0x08
#define ADC_8192 0x0A
#define ADC_D1   0x40
#define ADC_D2   0x50

// Specify sensor full scale
uint8_t OSR = ADC_8192;     // set pressure amd temperature oversample rate
  
// Pin definitions
int intPin = 14;  

int myLed = 5;

uint16_t Pcal[8];         // calibration constants from MS5637 PROM registers
unsigned char nCRC;       // calculated check sum to ensure PROM integrity
uint32_t D1 = 0, D2 = 0;  // raw MS5637 pressure and temperature data
double dT, OFFSET, SENS, TT2, OFFSET2, SENS2;  // First order and second order corrections for raw S5637 temperature and pressure data
    
int16_t tempCount;            // temperature raw count output
float   temperature;          // Stores the MPU9250 gyro internal chip temperature in degrees Celsius
double Temperature, Pressure; // stores MS5637 pressures sensor pressure and temperature

// global constants for 9 DoF fusion and AHRS (Attitude and Heading Reference System)
float pi = 3.141592653589793238462643383279502884f;

void setup()
{
  Serial.begin(115200);
  delay(4000);
  
  Wire.begin(21,22); // 21/22 are default on ESP32
  Wire.setClock(400000); // choose 400 kHz I2C rate
//  Wire.setClockStretchLimit(1000000);
  delay(1000);
  
  // Set up the interrupt pin, its set as active high, push-pull
  pinMode(intPin, INPUT);
  pinMode(myLed, OUTPUT);
  digitalWrite(myLed, LOW);
  
  I2Cscan();// look for I2C devices on the bus
    
  // Reset the MS5637 pressure sensor
  MS5637Reset();
  delay(100);
  Serial.println("MS5637 pressure sensor reset...");
  // Read PROM data from MS5637 pressure sensor
  MS5637PromRead(Pcal);
  Serial.println("PROM dta read:");
  Serial.print("C0 = "); Serial.println(Pcal[0]);
  unsigned char refCRC = Pcal[0] >> 12;
[... 174 lines cut ...]
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
}
```

But if I try to read the MPU9250 which requires write of a device address, then a register address (more than one byte), then I can not read any daya from it. Here is the multi-byte I2C function:

```
  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, 1);            // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

Some how the second byte is causing the I2C bus to get hing up. Not sure why but this is the reason most I2C devices cannot be read by the ESP32 except for the very simple (command and response types) ones.
```

Nourfdss, 2016-12-13:
@kriswiner  hey man can you please explain how did you connect the MS5637 to the esp32 im trying to conect the mpu6050 motion sensor but i failed

kriswiner, 2016-12-13:
I connected pin 21 to SDA, pin 22 to SCL and then 3V3 and GND. But this won't help you since i can't get anything but the I2C address out of the companion MPU9250. There is something different about the Invensense I2C bus/controller that is causing the I2C bus to latch up. Might be a clock stretching issue. I have seen this before on the ESP8266/85. I am using 4K7 pullips on the I2C lines, which usually works for a variety of controllers including the ESP8266/85. But somehow the ESP32 is having trouble with it.

Nourfdss, 2016-12-13:
So now what is the solution if i wanna get the motion sensor connected to esp32 i tried alot to find in the internet i did not fine anything

kriswiner, 2016-12-13:
Don't know, I'd like to get the other I2C sensors working as well. I am going to test more but I have not been able to get the MPU9250 to work yet.

Lokks like we will both have to just wait until this is fixed. Or go backt o using the ESP8266/85!

Nourfdss, 2016-12-13:
Okay man if i got anything i will post it here and let u know ,good luck

me-no-dev, 2016-12-13:
guys calm down :) i'm working on it and so far I can read MPU fine. Update will be up soon.

kriswiner, 2016-12-13:
No disresepect intended, you guys are doing great work!

Just trying to do my part ;>

I am looking forward to your fix.

me-no-dev, 2016-12-13:
issue with MPU was directly connected to the fact that before you read from it, you do not send stop (Wire.endTransfer(false)). Now that I have that fixed, I can not write more than 9 bytes to I2C... cat and mouse with this bus

me-no-dev, 2016-12-14:
please try the latest commit

kriswiner, 2016-12-14:
I just tried the new core and I am still having the same problem. I can see the I2C devices on the bus but cannot read from the MPU9250 WHO_AM_I register. This is the output I get:

```
Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x76  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF
```

I verified I can still get data from the MS5637 alone. That is, if I only ask for MS5637 data, read and write from/to MS5637 only then everything works, but when I try to talk with the MPU9250 the bus gets stuck and I can't talk to either.

kriswiner, 2016-12-14:
I also verified that I can get data from the VEML6040 I2C RGBW sensor. So some work and some don't. Odd...

kriswiner, 2016-12-14:
But cannot read the BMX055:

```
Scanning...
I2C device found at address 0x10  !
I2C device found at address 0x18  !
I2C device found at address 0x1E  !
I2C device found at address 0x59  !
I2C device found at address 0x68  !
I2C device found at address 0x69  !
I2C device found at address 0x6A  !
done

BMX055 accelerometer...
BMX055 ACC I AM 0xFF I should be 0xFA
BMX055 gyroscope...
BMX055 GYRO I AM 0xFF I should be 0xF
BMX055 magnetometer...
BMX055 MAG I AM 0xFF I should be 0x32
Could not connect to BMX055: 0xFF

```

tolson2000, 2016-12-14:
@kriswiner Can you try adding another Wire.beginTransmission to get the result instead of using endTransmission(false)? Just curious, I haven't received my ESP32 yet to help test. If it works.. might be a hint.

> But if I try to read the MPU9250 which requires write of a device address, then a register address (more than one byte), then I can not read any data from it. Here is the multi-byte I2C function:

```

  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);   // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);          // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, 1);        // Read one byte from slave register address 
  data = Wire.read();                          // Fill Rx buffer with result
  return data;                                     // Return data read from slave register
}

```

Change to...

```

  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission();                 //  <<------
  Wire.beginTransmission(address);  // <<------
  Wire.requestFrom(address, 1);       // Read one byte from slave register address 
  data = Wire.read();                         // Fill Rx buffer with result
  return data;                                    // Return data read from slave register
}

```

kriswiner, 2016-12-14:
I tried this:

```
  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data = 0;                        // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
//  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.endTransmission();                 //  <<------
  Wire.beginTransmission(address);  // <<------ 
  Wire.requestFrom(address, 1);            // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

and I get the same behavior. MS5637 alone works, but MPU9250 does not.
```

me-no-dev, 2016-12-14:
Here is what I have found:
- MPU does not like to be read one byte at a time.
- When you read only one byte, the MPU holds SDA low which trips the I2C bus and it reaches timeout.
- If you instead read two bytes, you will get the expected result

Here is an image when you read only one byte:
[image]

Here is when you read two bytes:
[image]

me-no-dev, 2016-12-14:
```cpp

const uint8_t MPU_addr=0x68; // I2C address of the MPU-6050
#define WHO_AM_I_MPU9250 0x75 // Should return 0x71

uint8_t mpuWhoAmI()
{
    uint8_t data; // `data` will store the register data
    Wire.beginTransmission(MPU_addr);         // Initialize the Tx buffer
    Wire.write(WHO_AM_I_MPU9250);                  // Put slave register address in Tx buffer
    Wire.endTransmission(true);             // Send the Tx buffer, but send a restart to keep connection alive
    Wire.requestFrom(MPU_addr, (uint8_t) 2);  // Read one byte from slave register address
    data = Wire.read();                      // Fill Rx buffer with result
    Wire.read();
    return data;                             // Return data read from slave register
}

```

me-no-dev, 2016-12-14:
Just tested manually and the MPU does not release SDA until it gets another 8 clocks.
I know I wrote code to add more clocks for ESP8266, but there I2C is software and here is hardware. We do not have another way of sending more clocks than reading more or writing more

kriswiner, 2016-12-14:
Tried the fix and it seems to work. I can read the MPU9250 and AK8963C. The fusion rate I measure is 20 kHz (compare to 60 kHz with STM32L4) but the fusion is not behaving properly because of the lack of floating point sqrt and sin/tan functions I think. Anyway, this I2C issue ios solved.

There is one curious problem I am having. I was polling the data ready register bit of the AK8963C to test whether bnew data was available. Now that I am reading the next byte, which is the first data byte, it is clearing the data registers so I have had to abandong the data ready test to read the data. Oh well, can't have everything!

I have reposited working ESP32 Arduino sketches here: https://github.com/kriswiner/ESP32

I have sketches for the MPU9250+MS5637, MS5637, VEML6040 and will be adding more as I can test other sensors.

me-no-dev, 2016-12-14:
awesome! BTW you should not need to read two bytes anymore :) I found the reason why one byte was not working and I hope I fixed it. So you are welcome to try reading one byte again.

kriswiner, 2016-12-14:
I tested the BNO055 and it works with the extra read byte also. I will download the latest and try without it next.

kriswiner, 2016-12-14:
Tried the latest with reading no extra byte and this does not work. Reading the extra byte does work.

hcs-svn, 2016-12-14:
With the latest version:
SSD1306 works
LM75 works
BMP180 works
BME280 works only with reading an extra byte

"works" means that not only recognition but also communication works.

Great, it gets better and better. Until this version BMP180 did not work.

That looks good:
```
BMP180: 1028 hPa  21.90 C
BME280: 1027 hPa  21.60 C  56 rH
LM75: 19.00 C
BMP180: 1028 hPa  21.90 C
BME280: 1027 hPa  21.60 C  56 rH
LM75: 19.00 C
BMP180: 1028 hPa  21.90 C
BME280: 1027 hPa  21.60 C  56 rH
LM75: 19.00 C
```

me-no-dev, 2016-12-14:
I'm reading the documentation for the chips and I'm looking at the decoded data by my logic analyzer and all looks exactly as it should, so I do not get why the extra byte would be needed here and not on other platforms

kriswiner, 2016-12-14:
I verfied the BME280 works without the extra byte. Not sure why the BNO055 and MPOU9250 might need it. I will check the MPU9250 again.

me-no-dev, 2016-12-14:
MPU working fine with one byte here
[image]

kriswiner, 2016-12-14:
OK, MPU9250 working fine with one byte reads. BNO055 is the outlier, but maybe it is me so I will try again.

hcs-svn, 2016-12-14:
Oops. kriswiner is right. BME280 works without extra byte.
Sorry, perhaps I lost the oveview which test I had activated.

me-no-dev, 2016-12-14:
now if we can get that BNO thing to work :)

kriswiner, 2016-12-14:
OK, verified the BNO055 now works with no extra byte, not sure what I was doing before.

closed 2016-12-14 by kriswiner

me-no-dev, 2016-12-14:
Thank you all for the help :)

hcs-svn, 2016-12-14:
Thank you for the arduino core

andrei-ivanov, 2016-12-14:
Tested my BME280 and works :)
Now if I could only find out why the temperature is about 3 degrees C above the real temp :-/

kriswiner, 2016-12-14:
If you have your BME280 anywhere near the ESP32 chip (or some other companion sensor) it will get warmer. Measure the current used in your circuit. If it is abobe 10 mA and the BME280 is close it will be warmer. Als, if you have the sensor nead the laptop fan exhaust it will see a warmer temperature. Lot's of reasons why it might show a higher temp.

andrei-ivanov, 2016-12-14:
@kriswiner 
Hmm, now that I've learnt how to plug the ESP32 and BME280 into a breadboard and how to connect them, any suggestions on guides about how I would use them in a "production" setup?

I assume that at least would imply a good separation of the sensor from the board to avoid such problems. Maybe other things like battery power, low battery detection, some sort of boxing...

kriswiner, 2016-12-15:
ESP32 forum is a better place for this discussion...

ifrew, 2017-08-25:
I am trying to use DMP functionality of the mpu6050 with the nodemcu esp32s. It initializes fine but I have found that when the interrupt occurs to tell me to read the data available in the FIFO for the device, when using the i2cdevlib library to read the bytes I get 0 returned. I enabled the debug statements in the i2cdevlib library and was able to identify the issue as being the  twowire call to read the bytes which returns 0. I am assuming that there is some other issue with the mpu6050 and the i2c bus as I have read on this and other forums. Hopefully me-no-dev might see this and throw some light on the issue. I am attaching the output of the debug statements so you can see what happens. The Wire.available call is returning 0
[getfifobytesdmp.txt](https://github.com/espressif/arduino-esp32/files/1252850/getfifobytesdmp.txt)

kriswiner, 2017-08-25:
The problem is with the i2cdevlib library, not the ESP32 i2c API.
<https://tools.usps.com/go/TrackConfirmAction.action?tLabels=EZ023693985US>

ifrew, 2017-08-25:
Your link is to a ups delivery kris :-) I guess you pasted the wrong link? I'm not so sure its the devlib library as the call that is causing the issue is Wire.available(); It returns 0 when there actually is data reported by the device. I think it is a MPU6050/i2c bus on esp32 issue as Me-No-Dev had found some issues with this in the past. The same code runs fine on ESP8266.
[testmpu6050esp32.txt](https://github.com/espressif/arduino-esp32/files/1252883/testmpu6050esp32.txt)

kriswiner, 2017-08-25:
Yes, sorry about that, didn;t intend to link to anything.

I found Jeff's DMP code to be fussy, let's say, which is why I haven't run
it in more than three years.The DMP can only do 6 DoF fusion anyway (I know
you are using the MPU6050, but the same applies to the MPU9250, a superior
device in a lot of ways). The ESP32 can do either 6 or 9 DoF fusion on the
host with high rates, so this would be preferred for a lot of applications,
and required fo 9DoF. There is code to do so here:

https://github.com/kriswiner/ESP32

muktillc, 2017-10-17:
I don't even see the I2C clock coming from pin 22. Am I missing something here.. I just see 3.3V on the oscilloscope

kriswiner, 2017-10-17:
Do you have pullups on the I2C lines? What board are you using?

muktillc, 2017-10-17:
Yes. I have 10k resistors attached. I am using DOIT ESP32 DevKit V1 board

kriswiner, 2017-10-17:
pin 22 is default SCL and 21 is default SDA.

Do you have this in your arduino sketch?

Wire.begin(21, 22, 400000); //(SDA, SCL) (21,22) are default on ESP32, 400
kHz I2C clock

muktillc, 2017-10-17:
Yes. I do. I can send you the code in a few minutes. It is a simple code. I
want to see something wiggle. But seeing nothing.

kriswiner, 2017-10-17:
Don't send me any code, better ask at the ESP32 arduino forum.

muktillc, 2017-10-17:
#include "Wire.h"

void setup() {
  Wire.begin(21,22);
  Serial.begin(115200);
  Wire.setClock(400000); // choose 400 kHz I2C rate
  Serial.println("Start i2c-test"); 
}

void loop() {
  byte error;

  // 0x20 is the address of a pcf8574 GPIO expander
  Serial.println("Try to contact 0x20"); 
  uint8_t data = 0;
//  Wire.beginTransmission(0x3A);
//  error = Wire.endTransmission();
    Wire.beginTransmission(0x3B);  // Address of the accelerator
    Wire.write(0x0F);              // Put WHO_AM_I address in Tx buffer
    Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
    Wire.requestFrom(0x3B, 1);     // Read two bytes from slave PROM address 
    while (Wire.available()) {
    data = Wire.read(); }               // Put read results in the Rx buffer
  Serial.print("Error code is:");
  Serial.println(data);

  delay(1000);
}

muktillc, 2017-10-17:
I am just trying to use the above code which is a clip from your code. I don't seem to be having the clock working.

kriswiner, 2017-10-17:
Yup, that should work, sure you have the I2C addresses correct? What
happens if you just run an I2C scan?

Maybe your dev board is busted...

muktillc, 2017-10-17:
the blink program and the internet connectivity works. I am using 10k pull up resistors. Do you recommend me using any other pins for I2C? If I want to use any other pins, I just need to change the number in the wire.begin() to the other pin number, right?

Do I have to change anything else? I was also wondering that how does the ESP32 know that pins are being used for I2C. I don't see that being mentioned anywhere.

muktillc, 2017-10-17:
what is the I2C_Scan() function doing? Do I have first include the function in my code first?

kriswiner, 2017-10-17:
You tell it via wire.begin(sda/scl);

You need to add a i2c scan function.

muktillc, 2017-10-18:
the board is not busted. I just wrote a code to output 1kHz at pin 22 and it works. I am getting 1kHz output

kriswiner, 2017-10-18:
Sounds like a problem for the ESP32 forum...

kriswiner, 2017-10-18:
Are you using the latest Arduino core?

muktillc, 2017-10-18:
I am getting the above clock. the saw shape clock does not make sense to me.

kriswiner, 2017-10-18:
Is the frequency 40 Hz?

muktillc, 2017-10-18:
no. it is 400kHz. the X axis is set to 20us/div

[14 more replies]