espressif/arduino-esp32 #93: Add option to compiler.cpp.flags

sorscode, 2016-12-12:
Add the following option tag in the compiler.cpp.flags, located in platform.txt file.

-fno-threadsafe-statics

This will allow the ArduinoJson library to compile.

copercini, 2016-12-12:
Same issue here

me-no-dev, 2016-12-13:
Maybe the author of the library should come along and look at the issue. We are running in a multithreaded multicore environment and there is a reason for some errors to show. Would not like to just dismiss them like that ;)
We are rewriting all Arduino libs to be thread-safe, so maybe others should do the same ;)

igrr, 2016-12-13:
Ah, this is because static initialization guards are not implemented. I'll should get this done.

me-no-dev, 2017-02-08:
seems to be fine now :) closing this

closed 2017-02-08 by me-no-dev