espressif/arduino-esp32 #97: Stopping serial monitor

eroniki, 2016-12-13:
When I use the `make flash monitor` to observe the serial communication, I cannot stop the process in regular (ctrl+c) fashion. Is there a way to stop the process without unplugging the USB cable from the PC?

me-no-dev, 2016-12-13:
mine says:
```
--- Miniterm on /dev/tty.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```
so answer is Ctrl+]

closed 2016-12-13 by eroniki

exocode, 2019-05-23:
Erm.. why is that not `ctrl+c` like any other program out there?

I have a European keyboard and have to press three keys (`ctrl+alt+5`) to exit which is a bit annoying.
It would be very user-friendly when new users don't have to search on the internet and can use the "traditional" 99% working `ctrl+c` approach.

AP-Frank, 2019-06-09:
What works for me (German keyboard):

Use ``ctrl++``. Explanation: ``+`` on a keyboard with German layout is where ``]`` would be on a keyboard using the standard English/American layout.

So for all other layouts it might be worth to try ``ctrl+<key where ] would be on english keyboard>``

bonmotwang, 2019-06-14:
ctrl+] worked on me. I am using CMAKE. Just tested it.

arnolde, 2019-12-05:
On my German keyboard CTRL+AltGr+9 works as expected. I'd still prefer ctrl-c to work, at least optionally.

zhivko, 2019-12-05:
CTRL+C would be super standard solultion - What is the reason not using CTRL+C? I could not know... It would be great if authors could let us know why they did not use CTRL+C? thx

mchobby, 2020-07-10:
On **belgian Keyboard** the "]" is produced with ALTGR+]
So to exit the monitor, press CTRL+ALT+ALTGR+] to compose the needed key combination
Cheers

lbernstone, 2020-07-10:
The reason CTRL+C is not used is because it is used to send a CTRL+C to the remote system rather than the local.

sennert, 2020-09-18:
Macbook Pro with german keyboard: control + option + 6