espressif/arduino-esp32 #100: esp32-hal-i2c.c fifo problems?

espuser, 2016-12-14:
Hi, on my scope I can see that the fifo does not accept input for a while, so in your write routine you write several times until fifo count increases. You do not check after transmitting the address byte
 and especially after a possible second address byte (10 bit addressing). I assume this byte will sometimes be lost because there is very little code between the two store opcodes.

me-no-dev, 2016-12-14:
I do not write until it increments, I wait until it increments and by many tests I concluded that address bytes are not included in the fifo_count when writing to it (as they are not when reading from it)

me-no-dev, 2016-12-15:
closing this now :) if you have more questions please come to gitter :)

closed 2016-12-15 by me-no-dev