espressif/arduino-esp32 #102: Conflict Between Wifi & analogRead()?

kherNI, 2016-12-15:
Is anyone seeing a conflict with analogRead() values when **_#include <WiFi.h>_** is included in a sketch?  analogRead works great without WiFi.h included; however, if WiFi.h is included then the analogRead values returned are all maxed out (4096 values). Note I've tested on pins 12, 14, 26 & 27.

palomarcoder, 2017-01-02:
I found a note here that may be relevant:  https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf 

Pin1, Pin43 and Pin46 are the analog power supply pins. Pin3 and Pin4 are the power supply pins for the power amplifiers. It should be noted that the sudden increase in current draw, when ESP32 is in transmission mode, may cause a power rail collapse. Therefore, it is highly recommended to add another 0603 10 uF capacitor to the power trace, which can work in conjunction with the 0402 0.1 uF capacitor.

JeroenVanOort, 2017-01-08:
I've been having trouble with this as well. After a lot of trial and error I have found out that analogRead will work on some pins while using WiFi, 32 and 36 are examples of those pins. I hope this helps when trying to find out why this happens.

nissimzur, 2017-04-19:
Hi
It seems the ESP32 pin 32 is been used as SDO to flash. Are you sure you can use is as analogRead input?
Please call my Skype nissim.test

igrr, 2017-04-19:
This happens because ADC2 pins can not be used when WiFi is used.

On the other hand, ADC1 pins can be used even when WiFi is enabled.

hkotkanen, 2017-05-13:
@JeroenVanOort thank you kindly, switched to a pin you indicated and got mine working.

@igrr thank you also for the information! Your statement is so casual and matter-of-fact, but this thread seems to be then only place that documents this ...feature. Was I supposed to have found it out from another source?

me-no-dev, 2017-09-22:
There is a fix coming for ADC2 and WiFi soon :)

pieman64, 2017-10-16:
@me-no-dev any news on the ADC2 fix?

JulienHai, 2017-12-06:
any news? when will adc2 be fixed?

palomarcoder, 2017-12-27:
Any estimated time?  I was really hoping to use this in a project over the holiday.

AirCycler, 2018-01-03:
Hi,
I am unable to read the analogRead value for GPIO 33 pin on ESP-WROOM-32. I am using Arduino IDE. Any help with a sample code?

Thanks,
Anusha

beegee-tokyo, 2018-01-04:
GPIO33 should be ok to use, it is connected to ADC1. Check [Wiki](https://github.com/beegee-tokyo/arduino-esp32/wiki/ADC2-channel-cannot-be-used-when-WiFi-is-in-use).

Extract of my code to read an analog value from a LDR:
```cpp
int ldrPin = 33;
void setup() {
  pinMode(ldrPin,INPUT);
  adcAttachPin(ldrPin);
  analogReadResolution(11);
  analogSetAttenuation(ADC_6db);
}
void loop() {
  Serial.println(analogRead(ldrPin));
  delay(2000);
}
```

AirCycler, 2018-01-05:
Thank you @beegee-tokyo. I will try that. I was not using "adcAttachPin(ldrPin);", probably that is why I was unable to read the correct values.

AirCycler, 2018-01-05:
@beegee-tokyo A quick question. Do I have to use any libraries or header files to include analogSetAttenuation(ADC_6db); and adcAttachPin(ldrPin);?

Thanks,
Anusha

beegee-tokyo, 2018-01-05:
You might need to include esp32-hal-adc.h

AirCycler, 2018-01-05:
@beegee-tokyo I am not getting stable values. They change and they don't make sense. Please check my code,
#include "esp32-hal-adc.h"                          // to read analog values

int pin =33;

void setup() {
 Serial.begin(115200);
  pinMode(19, OUTPUT);
   pinMode(pin, INPUT);
   digitalWrite(19,1);
  adcAttachPin(pin);
  analogReadResolution(11);
  analogSetAttenuation(ADC_6db);
}

void loop() {
 digitalWrite(19,0);
  for(int i = 0;i<10;i++){
  delay(2000);
  Serial.print("INA0_OUT:");                                 // print it
  Serial.println(analogRead(pin));
  }
delay(5000);
 digitalWrite(19,1);
  for(int i = 0;i<10;i++){
  delay(2000);
  Serial.print("INA1_OUT:");         
  Serial.println(analogRead(pin));
  }
  delay(3000);

}
 and The output is: 
INA0_OUT:492
INA0_OUT:481
INA0_OUT:488
INA0_OUT:511
INA0_OUT:568
INA0_OUT:784
INA0_OUT:552
INA0_OUT:549
INA0_OUT:562
INA0_OUT:573
INA1_OUT:2047
INA1_OUT:2047
INA1_OUT:2047
INA1_OUT:2047
INA1_OUT:1925
INA1_OUT:2047

Am I doing something wrong?

beegee-tokyo, 2018-01-06:
I tried your code and found no problems. Output:
```log
Testing with out = 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0
INA1_OUT1 ADC5: 0

Testing with out = 1
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
INA0_OUT ADC5: 2047
```
Did you measure the voltage on GPIO19? Is it changing from 3.3V to 0V?
Did you measure the voltage on ADC5/GPIO33? Is it changing from 3.3V to 0V?
Either your module is somehow busted or the connection between the output 19 and analog input 33 is bad.
Can you try another output pin and another analog input?

twanek, 2018-06-17:
@me-no-dev:

"There is a fix coming for ADC2 and WiFi soon :)"

any progress on this? we would need more than 6 analog inputs in our project. there is any workaround for this?

thanks!

cgonzaloch, 2018-11-05:
Hi all!

Any update on this? is there any fix already for ADC2 and WiFi support? 

Thanks in advance!

markingle, 2018-11-24:
Any progress on this issue?  I am trying to use the pulse counter function along with Bluetooth but the counter hangs when BT starts up.  I suspect the use of the same power rail is causing my problems too.

intellica, 2019-01-23:
@me-no-dev any news on the ADC2 and Wifi fix?

danielcolchete, 2019-02-23:
For the record, resetting registers SENS_SAR_START_FORCE_REG and SENS_SAR_READ_CTRL2_REG to the old values they had before WiFi was turned seem to be a work around for this issue. And will allow you to use ADC2 again after you stop and deinitialize WiFI.

An algorithm here would be something like:
old1 <- SENS_SAR_START_FORCE_REG
old2 <- SENS_SAR_READ_CTRL2_REG
InitWifi
StartWifiAndConnect
(Do your wifi thing here)
StopWifiAndDisconnect
DeinitWifi
SENS_SAR_START_FORCE_REG <- old1
SENS_SAR_READ_CTRL2_REG <- old1

danielcolchete, 2019-02-23:
I forgot to include the SENS_SAR_MEAS_START2_REG register. This one is important too.

memory125, 2019-02-25:
@me-no-dev  How about the progress on fixing this issue?

paccerdk, 2019-03-05:
@me-no-dev
> There is a fix coming for ADC2 and WiFi soon :)

This was written on Sep 22, **2017**

"Soon" should have passed by now, 
Has the "fix" been integrated? is there any news on the matter?
Are there any workarounds we can use for now, like "pausing" radio for a split second while measuring and then re-enabling the WiFi (without having to completely re-auth etc)?

yichengsun, 2019-03-22:
Similarly following up. One of the purported benefits of the ESP32 is having all these GPIOs. If we can't access them on wifi, it defeats the purpose.

hervema, 2019-03-31:
Hello. I get the same problem, and agree with yichengsun. For me it's "stupid" if we cannot have GPIOs and Wifi enable at the same time ! What is this case the purpose of a such module ?!

mfalkvidd, 2019-03-31:
@hervema you might not be aware, but the Arduino support for esp is a community effort. Nobody is paying, and nobody gets paid. Stupid or not, the problem won't be solved until someone with enough free time and knowledge decides it would be fun to try to figure out a solution.

hervema, 2019-03-31:
Can be useful. Works for me. 

https://desire.giesecke.tk/index.php/2018/01/30/adc2-channel-cannot-be-used-when-wifi-is-in-use/

hervema, 2019-03-31:
No sure I get you and I didn’t want offense someone. If this the case, sorry and I’m the stupid guy.

I’m talking about the hardware module based on ESP-WROOM-32. With my current knowledge I understand that conceptually the hardware has limitation and ADC2 is out with WIFI is on, for that I think is abnormal and decrease the interest of the module.

Regards

Envoyé le :dimanche 31 mars 2019 20:08

steve-cr, 2019-04-08:
You are talking about problem of ADC2 & WiFi but the real BIG problem is ADC.
Are you never tried to read SAME RESISTANCE or read SAME TENSION with different ESP32?
No?
Try to put simple program for analogRead on some analog GPIO. 
Read the value, write it and change the board...
Are the same value? 
NO !!!

NuclearManD, 2019-05-04:
@danielcolchete I tried your fix by resetting the registers.  I can confirm that it works.  Any idea why this works though?  Does the sudden voltage drop from a WiFi packet broadcast mess with the contents of the register somehow?  I have no idea why the code would modify them when using WiFi, so that seems most plausable to me.

If the registers are modified when WiFi packets are broadcasted, then maybe the registers can be reset after each transmit, so WiFi and the ADCs can be used at the same time.

roboticafacil, 2019-05-24:
@danielcolchete  or @NuclearManD Can you paste the code you use to fix the problem? I am using a board that maps some analog pins to ADC2 and I would like to simultaneously use it the WiFi module. Thanks!

NuclearManD, 2019-05-27:
@roboticafacil You'll have to work it into your code, as these fragments I'm about to provide were part of a much larger program.  The project now uses ADC1 so I don't use this method now, but here you go:

You'll need to declare these variables, probably globally:
```uint64_t reg_a;
uint64_t reg_b;
uint64_t reg_c;```
The name doesn't matter, as long as the name makes sense and the variables are uint64_t.

Put this right before you turn on WiFi anywhere
```reg_a = READ_PERI_REG(SENS_SAR_START_FORCE_REG);
reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
reg_c = READ_PERI_REG(SENS_SAR_MEAS_START2_REG);```
This will save the register contents so you can restore them.

After turning WiFi off, to bring the ADCs back up you'll need to have this:
```WRITE_PERI_REG(SENS_SAR_START_FORCE_REG, reg_a);  // fix ADC registers
WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
WRITE_PERI_REG(SENS_SAR_MEAS_START2_REG, reg_c);```

You will not be able to use the ADCs at all while WiFi is on, just making sure that's clear.  How to turn WiFi on or off depends on what you're using to program, weather its ESP-IDF or Arduino.  There's plenty of documentation for both.

leoNavarro95, 2019-06-26:
> @roboticafacil You'll have to work it into your code, as these fragments I'm about to provide were part of a much larger program. The project now uses ADC1 so I don't use this method now, but here [...]

I have to read battery voltage on my Heltec-esp32-wifi-LoRa V.2 while have wifi connection, and the pin attached to the batery is on ADC2. I hope you code with the reset tegisters works for me. Thaks

leoNavarro95, 2019-06-26:
@me-no-dev any advance on this issue? Its will be great to pause the wifi to read adc2 pins without do all the reconnection.

NuclearManD, 2019-06-26:
@leoNavarro95 I looked into this a bit, and sadly I have found no way to do it.  I think it might be possible if a custom WiFi driver is written, but I think that may involve reverse-engineering the ESP32.  AFAIK the ESP32 WiFi driver is closed source (if I am wrong then someone PLEASE let me know, I want that WiFi driver!).

I think that, in theory, packets may be able to be sent less frequently or something, to prevent such sharp voltage drops.  Sometimes these drops even crash the ESP32!!!  It's absolutely absurd.  The ESP32 is still amazing of course, but still.

My theory will be wrong if a single sent packet will drop the voltage too much.  Hopefully it's multiple packets in succession that drop the voltage too much.

leoNavarro95, 2019-06-27:
@NuclearManD and @danielcolchete the code for restore the adc registers works fine for my proyect. 
You forget to add this:
-> #include <soc/sens_reg.h>
without this the registers can't be access
Thank you guys

NuclearManD, 2019-06-27:
@leoNavarro95 Nice catch!  Sorry I missed that detail.  Whoops!

davidvilla, 2019-11-28:
In my case just save READ_CTRL2 register before connecting WiFi:
```
void setup() {
    reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
    WiFi.begin(SSID, PASSWORD)
    [...]
}
```
and restore it just before analogRead:
```
int read_sensor() {
  WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, read_ctl2);
  return analogRead(A0);
}
```
It is not required to disconnect WiFi at any time.

Tested in WEMOS-D1-R32 (ESP32).

Thank you

WayneKeenan, 2019-11-29:
@davidvilla I've tried this and found that the ADC readings are inverted.
Have you confirmed with Espressif that this doesn't have unexpected conciquences on WiFi? 

I've also found this: https://github.com/espressif/esp-idf/blob/4048db35b6a1ac4216150480dcacdc4230f0baa2/components/driver/test/test_adc2.c#L43

davidvilla, 2019-12-03:
You are right, the readings become inverted... :-(

WayneKeenan, 2019-12-08:
Hi @me-no-dev is there any news on the ADC2 and Wifi fix?

edstoica, 2019-12-13:
@davidvilla : Thank you all so much for the code snippets.
This is working perfectly for me on a ESP32 with pin 36 (adc2).

Once again in a more beginner form:

> // ADC2 Pin Reset: 
> // This is a Workaround to use ADC2 Pins on ESP32 when Wifi is on. 
> // (usuall only ADC1 Pins are usable for analogRead() when Wifi is on.)
> // -- We save the Pin registers before WifiBegin()
> // -- and reset the resisters before every analogRead() cycle. 
> // -- No need to turn Wifi on or off. 
> 
> #include "esp32-hal-adc.h" // needed for adc pin reset
> #include "soc/sens_reg.h"  // needed for adc pin reset
> uint64_t reg_b;            // Used to store Pin registers
> 
> #define PIN 36             // Set ADC2 Pin here
> int value;
> 
> void setup() {
>     // Save Pin Registers : Do this before begin Wifi
>    reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
>
>    Wifi.Begin(); // or similar wifi init function
> }
> 
> void loop() {
>   // ADC Pin Reset: Do this before every analogRead()
>   WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
>
>   value = analogRead(PIN); 
> 
>   delay(1);
> }

roboticafacil, 2019-12-13:
Cool! I will check that solution. Thanks!

leoNavarro95, 2019-12-13:
@edstoica That's great, so with that is not necessary to lost wifi connection? I mean to reset all the connection

atanisoft, 2019-12-13:
> This is working perfectly for me on a ESP32(M5stickC) with pin 36 (adc2).

@edstoica GPIO36 is ADC1 and not ADC2...

leoNavarro95, 2019-12-13:
mmm, so **bad news** :(

edstoica, 2019-12-13:
> 
> 
> > This is working perfectly for me on a ESP32(M5stickC) with pin 36 (adc2).
> 
> @edstoica GPIO36 is ADC1 and not ADC2...

@atanisoft :
Thank you for you input. 
I see on the official esp32 Pin layouts that you are right.

In my case with the "M5StickC", things seem to be different?!?
- pin 26 (officially ADC2) works together with wifi with no problems.
- pin 36 always returns 4096 when wifi is on.
--> for pin36, the "reset registers" solution mentioned before worked perfect.

So sorry that I added even more confusion here ;)

**But for anyone who has a pin which returns always 4096 when wifi is on: just give the "reset registers" solution a try. maybe it works for your case.**

DavidSCFA, 2020-01-26:
I found this useful:
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

phophermi, 2020-02-27:
**ADC2 working with WiFi on esp32-cam via PIN 14**

This workaround conflated so nicely by edstoica seems to function perfectly on a
 **esp32-cam** board, wifi connected and at the same time using GPIO14 for ADC 
with analogRead().
Without it, output is continuously 4095 (since it is used by ADC2).

Alteregoxxx, 2020-03-02:
> In my case just save READ_CTRL2 register before connecting WiFi:
> 
> ```
> void setup() {
>     reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
>     WiFi.begin(SSID, PASSWORD)
>     [...]
> }
> ```
> 
> and restore it just before each analogRead:
> 
> ```
> int read_sensor() {
>   WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, read_ctl2);
>   return analogRead(A0);
> }
> ```
> 
> It is not required to disconnect WiFi at any time.
> 
> Tested in WEMOS-D1-R32 (ESP32).
> 
> Thank you

Hello,
I just want to help with my 2cents:

1)The ADC2 becomes unavailable also when using Bluetooth, not only WiFi
2)The approach proposed by @davidvilla (the one I quoted) works but, as himself and @WayneKeenan already wrote, the readings one obtain from the ADC2 are inverted.
3)I discovered the way to solve the "inversion problem", find it below:

> ```
>// ADC2 restoring procedure
>// This is a Workaround to use ADC2 Pins on ESP32 when Wifi or Bluetooth is on.
>// (usually only ADC1 Pins are usable for analogRead() when Wifi or Bluetooth is on.)
>// -- We save the ADC2 control register before WifiBegin() or BluetoothBegin()
>// -- then restore its original value, then set a specific bit of the ADC2 control register   
>//to avoid inverted readings: we do the latter two before every analogRead() cycle.
>// -- This achieves ADC2 usability even when Wifi/Bluetooth are turned on!
>
>#include "soc/sens_reg.h" // needed for manipulating ADC2 control register
>uint64_t reg_b; // Used to store ADC2 control register
>#define PIN xx // Substitute xx with your ADC2 Pin number
>int value;
>
>void setup() {
>// Save ADC2 control register value : Do this before begin Wifi/Bluetooth
>reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
>Wifi.Begin(); // or similar wifi init function or Bluetooth begin()
>}
>
>void loop() {
>// ADC2 control register restoring
>WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
>//VERY IMPORTANT: DO THIS TO NOT HAVE INVERTED VALUES!
>SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
>//We have to do the 2 previous instructions BEFORE EVERY analogRead() calling!
>value = analogRead(PIN);
>delay(1);
>}
> ```

It seems now we have a fully working ADC2 even when WiFi/Bluetooth are active!!
Please test and confirm if it works!

CarlosGS, 2020-04-04:
Thanks for this, I hope it gets integrated into the main code soon.

I still remember the "There is a fix coming for ADC2 and WiFi soon :)" back in 2017!

pierreh, 2020-04-04:
Unfortunately this didn't work for me on a ESP32-CAM pin GPIO14

lbernstone, 2020-04-04:
arduino-esp32 now uses the upstream functions for adc, so this discussion would have to happen in https://github.com/espressif/esp-idf/issues/461 .

bijilbaji, 2020-04-09:
i have tested code and fund there is slights difference in measured vales in with WiFi and without WiFi. Thanks for finding a solution. "There is a fix coming for ADC2 and WiFi soon :)" waited for so long. Again Thanks.

int adc2[]={25,26,27,14,12,13,15,2,4};
void setup() {
reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
Serial.begin(115200);
wificlient();
}

void loop() {
  for(int i=0;i<8;i++)
  {
WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
value = analogRead(adc2[i]);
Serial.print(adc2[i]);
Serial.print("=");
Serial.print(value);
Serial.print(",");
  }
Serial.println("");
}

ProfJayHam, 2020-05-07:
Using Deep sleep can cause problems with reg_b after resetting from DeepSleep
Here is some code that demos the solution

// Test ADC1 and ADC2 on ESP32 with WiFi and deep sleep using control reg
// Tested on Huzzah32 and Lolin D32 
// J Ham, Colorado State Univ.
// https://github.com/espressif/arduino-esp32/issues/102
// https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#_CPPv418esp_reset_reason_t
//

#include <WiFi.h>
#include <soc/sens_reg.h>

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP 5        /* Time ESP32 will go to sleep (in seconds) */

byte adc1[]={34,39,36};   // A2,A3,A4
byte adc2[]={26,25,14};   // A0,A1,A6 dont use pin A5/4 

static RTC_NOINIT_ATTR int reg_b;  // place in RTC slow memory so available after deepsleep

//char ssid[] = "csu-guest";   // your network SSID (name) 
//char pass[] = "";   // your network password
char ssid[] = "HOME-D289-2.4";   // your network SSID (name) 
char pass[] = "393T4UC43YA44CU3";   // your network password

void setup() {
  Serial.begin(115200);
  
  for(int i=0;i<3;i++)
  {
   pinMode(adc1[i], INPUT);
   pinMode(adc2[i], INPUT);
  }

  // determine cause of reset
  esp_reset_reason_t reason = esp_reset_reason();
  Serial.print("reason ");Serial.println(reason);
  
  // get reg_b if reset not from deep sleep
  if ((reason != ESP_RST_DEEPSLEEP)) { 
    reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);  
    Serial.println("Reading reg b.....");
  }
  Serial.print("reg b: ");
  printf("%" PRIu64 "\n", reg_b);
  
  WiFi.mode(WIFI_STA);       // Initialize WiFi
  
  // Connect or reconnect to WiFi
    if(WiFi.status() != WL_CONNECTED){
      Serial.print("Attempting to connect to SSID: ");
      while(WiFi.status() != WL_CONNECTED){
        WiFi.begin(ssid, pass);  // Connect to WPA/WPA2 network. Change this line if using open or WEP network
        Serial.print(".");
        delay(3000);     
      } 
      Serial.println("\nConnected.");
    }
}

void loop() {

  Serial.print("ADC1: ");    // read ADC1
  for(int i=0;i<3;i++)
  {
   // WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
   // SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);  
    int reading = analogRead(adc1[i]);
    float volt= -0.000000000000016 * pow(reading,4) + 0.000000000118171 * pow(reading,3)- 0.000000301211691 * pow(reading,2)+ 0.001109019271794 * reading + 0.034143524634089;
    Serial.print(adc1[i]);
    Serial.print("=");
    Serial.print(volt);
    Serial.print(",");
  }
  Serial.println();

  Serial.print("ADC2: ");  // read ADC2
  for(int i=0;i<3;i++)
  {
    WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
    SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
    int reading = analogRead(adc2[i]);
    float volt= -0.000000000000016 * pow(reading,4) + 0.000000000118171 * pow(reading,3)- 0.000000301211691 * pow(reading,2)+ 0.001109019271794 * reading + 0.034143524634089;
    Serial.print(adc2[i]);
    Serial.print("=");
    Serial.print(volt);
    Serial.print(",");
  }
  Serial.println();
  Serial.println();

  ////sleep
  // esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_ON); 
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_deep_sleep_start();
  
}

michox, 2020-05-19:
Hi, thanks for the fix, it works so far (and for some reason my values are not inverted? I am at GIPO4 (ADC2 CH0)) However, after applying the fix, I can only measure up to 1V. I am on a devkit and before I was able to measure 3.3V, is there any way around this?

sakurai-youhei, 2020-09-01:
Result report for my case: GPIO4 (ADC2 CH0) of ESP32 DEVKIT V1 with WiFi ON

https://github.com/espressif/arduino-esp32/issues/102#issuecomment-593650746 by @Alteregoxxx seems to be insuficcient for my case but https://github.com/espressif/arduino-esp32/issues/102#issuecomment-565287359 by @edstoica does work perfectly!!!

tablatronix, 2020-09-19:
Somehow I just now found this out!, you think this would have been in giant red letters everywhere..

MikeMaynard14, 2020-10-07:
Wow, this took me 4 hours to fix... I can confirm that @edstoica code works on pin 14 for ESP32 CAM. Thank you so much, you saved my bacon! I bought 24 esp32 cams for a course I am giving thinking... yeah esp8266 worked well enough... lets go for the 32 Cam... safe to say wont be using Esp32 again in future. To much hassle bleh!

[33 more replies]

closed 2021-07-16 by me-no-dev