espressif/arduino-esp32 #103: Wifi Scan example fails to compile

kriswiner, 2016-12-17:
Just tried to comoile the WiFi scan example in the repository and it stumbles on

    WiFi.mode(WIFI_STA);

claiming 'class WiFiClass' has no member named 'mode'.

Am I missing some magic incantation to get it to work? I would like to give the wifi sector of the ESP32 a try.

kherNI, 2016-12-17:
I had that issue as well and found that a different WiFi.h was being called in the compiler and not the one dedicated for ESP32.  I saw this by turning on the verbose output for compilation and saw the conflict in WiFi classes.

kriswiner, 2016-12-17:
Excellent! Thank you. It works now as does Sparkfun's example.

Might I suggest that WiFi.h be renamed ESP32WiFi.h to avoid this kind of confusion similar to ESP8266WiFi.h?

closed 2016-12-17 by kriswiner

MrJustPaul, 2017-07-18:
Help!  I'm getting error "class WiFiClass' has no member named 'firmwareVersion"  I have no idea how to FIX it!  Any guidance would be appreciated.

kriswiner, 2017-07-18:
Probably have more than one wifi.h library...

MrJustPaul, 2017-07-18:
I DO!  One here - C:\Program Files (x86)\Arduino\libraries\WiFi\src
another here - C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\WiFi\src
and more! - C:\Users\justp\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src

Not sure why but...  what should I try?  Delete? Rename?  Thanks

tobozo, 2017-07-18:
delete:
     C:\Program Files (x86)\Arduino\hardware\espressif\esp32
keep:
    C:\Users\justp\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src
restart Arduino IDE, try to compile, if it fails, delete:
    C:\Program Files (x86)\Arduino\libraries\WiFi

MrJustPaul, 2017-07-19:
Thank you @kriswiner  and @tobozo and ALL who contribute their knowledge!  
Here's where I am:
Uninstalled (thru Control Panel):  Arduino IDE, Git, and Python (trying to clean up a bit)
Followed the install docs (which keep changing? )
Now I don't see the new boards on the IDE list.  My ESP32 doesn't show up.  Compiles are working but I can't upload.
I'm sure it's something simple - in fact, I recall one video that addressed this but now I can't find it.
Thanks in advance!
Paul

kriswiner, 2017-07-19:
Did you re-enter the .json for espressif boards in the Arduino IDE
preferences?

kriswiner, 2017-07-19:
> Actually this is not mentioned in the documentation (OP probably assumed
> Arduino IDE is already setup to manage ESP8266 boards) so the information
> to add in the Arduino prefs is:
>
> Additional Boards Manager URL : http://arduino.esp8266.com/
> versions/2.3.0/package_esp8266com_index.json
>
> Maybe this is worth a pull request?

Maybe there is no .json for the ESP32?

MrJustPaul, 2017-07-25:
Thanks all -  for ALL your help!  
I've been traveling and just got back to this.  Seems that I fix one thing and it breaks something else!
I now have all of the ESP32* boards to select but now I'm back to "'class WiFiClass' has no member named 'firmwareVersion'".   I deleted an extra Wifi.h, and kept the one in -
 C:\Users\justp\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src but I'm still getting the error.
I also see some other duplicate items:
wifi_spi.h (3 versions):
C:\Program Files (x86)\Arduino\hardware\arduino\avr\firmwares\wifishield\wifiHD\src
C:\Program Files (x86)\Arduino\libraries\WiFi\extras\wifiHD\src
C:\Program Files (x86)\Arduino\libraries\WiFi\src\utility 
...and some others - WifiClient.h (2 versions), wifiHD.hex (3 versions), WifiServer.h (2 versions), maybe others~~~another problem??  What info would help?  Thanks in advance!

dimkir, 2018-07-24:
Ok, I was bitten by the same thing and here's what happened and how found a fix:

- I installed fresh Arduino IDE 
- I installed esp32 stuff into `~/Arduino/hardware/espressif/esp32` folder (as per instructions)
- I used a "wifi example" and it produced errors as above (`firmwareVersion...`)

As it is discussed above, it seemed that the problem was due to the conflict of the `WiFi.h` files.  However it seemed a bit odd, that Arduino IDE, which supports a number of boards would choke on such as simple step of duplicated `WiFi.h` file. 

I tried moving `~/Arudino/hardware/espressif/esp32` into `<ArduinoIDEFolder>/hardware/espressif/esp32` but it didn't really help. 

#### Problem caused by not attentive using of Arduino IDE

It turns out that `Examples > Examples for any board > Wifi > ConnectWithWPA`  DOES NOT work with ESP32 (despite section name suggesting "Example for any board"). 

The problem was caused by:
 a) I didn't select appropriate board from the `Tools > Board` 
 b) UI of Arduino IDE has a feature: it only shows examples for Active board

So in my case some _other_ board was selected (let's say it was Yun). So when I was glancing though examples I noticed: 
 -  `Built in examples`  (no Wifi There)
 -  `Examples for any board` (logically sounds it would be there)
 -  `Arduino Yun examples` (I ignored this one when glancing). 

So out of the 3 given options I went for the "logical one"   `Examples > Examples for any board > Wifi > ConnectWithWPA` which seems (at least in my case) not compatible with ESP32. 

####  What worked for me

 - Ensure that right board is selected in `Tools > Board`  (my board is from ebay, so I chose `ESP32 Dev Module`) 
 - When the right board is selected inside of examples you will have section `Examples for ESP32 Dev Module`
 - Within those examples pick  `Examples for ESP32 Dev Module > Wifi > WifiClient` 
 - Set up your SSID and KEY and Compile. Everything worked for me

ps. I am using Ubuntu 16.04 but I have a feeling that Windows should produce exactly same results (based on OP problem)

ivan77800, 2021-04-15:
'class WiFiClass' has no member named 'forceSleepBegin'  i need to used for esp32 for deep sleep method i get this message can anyone explain?