espressif/arduino-esp32 #104: floating point functions work?

kriswiner, 2016-12-18:
I inserted this function into a sketch that compiles and runs with no problem:

 float test = 145366.45f*(1.0f - powf((VBAT/1013.25f), 0.190284f));

where VBAT is a float measured using analogRead.

The sketch does not compile and returns this error:

```
C:\Users\kris\Documents\Arduino\hardware\espressif\ESP32/tools/sdk/lib\libm.a(lib_a-ef_pow.o):(.literal+0xbc): undefined reference to `__ieee754_sqrtf'

C:\Users\kris\Documents\Arduino\hardware\espressif\ESP32/tools/sdk/lib\libm.a(lib_a-ef_pow.o): In function `__ieee754_powf':

/home/ivan/freertos/smpemu/newlib/newlib_xtensa-2.2.0/xtensa-esp108-elf/newlib/libm/math/../../../.././newlib/libm/math/ef_pow.c:115: undefined reference to `__ieee754_sqrtf'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.

```

Am I missing a library or something?

issalig, 2016-12-20:
Undefined reference normally points to a missing #include <math.h>
Also you should link to math library with flag -lm

Hope this helps.

me-no-dev, 2016-12-20:
the undefined reference above means that the methods mentioned are not implemented and not tha they are not defined :)
the flag -lm is already added. Seems that there are some methods that should be implemented in stdlib @igrr ?

me-no-dev, 2017-02-08:
compiles fine now :) closing this issue

closed 2017-02-08 by me-no-dev